mbed-ce/mbed-os #18: Make json files json5

multiplemonomials, 2022-06-05:
JSON5 introduces some really nice quality of life improvements for JSON files, including trailing commas (yes!) and the ability to have comments (yaaaaaasssss!).  We should change the mbed-tools python package (would probably need to make a fork) to parse JSON files (mbed_app.json, mbed_lib.json, targets.json) with a json5 parser.

multiplemonomials, 2023-01-13:
That could be pretty sweet too.  The only issue I could think of is that if we converted targets.json to YAML, it would make it a pain to merge changes from upstream that modify the json version.

ladislas, 2023-01-13:
we can keep targets.json in json and have the other in yaml

at the end of the day, all files are just parsed, it doesn't change much if data is from yaml or json

multiplemonomials, 2023-10-26:
Resolved by #194