mbed-ce/mbed-os #35: Unable to determine what CMake generator to use...No usable generator found.

maiorfi, 2022-09-02:
Hi.

Following your guide for VSCode here: https://github.com/mbed-ce/mbed-os/wiki/Project-Setup:-VS-Code I have an error while running "CMake: Set Build Target" step (as well as when I try to build project anyway).

Error is:
```
[main] Unable to determine what CMake generator to use. Please install or configure a preferred generator, or update settings.json, your Kit configuration or PATH variable. Error: No usable generator found.
```

I have installed all required components (toolchain, vscode extensions, tools, etc).

Kit scanning detected my GCC 10.3.1 arm-none-eabi toolchain correctly.

cmake-variants.yaml I am using is following:

```yaml
buildType:
  default: Develop
  choices:
    Develop:
      short: Develop
      long: Emit debug information but also optimize
      buildType: Develop
    Debug:
      short: Debug
      long: Emit debug information and don't optimize
      buildType: Debug
    Release:
      short: Release
      long: Optimize generated code
      buildType: Release
board:
  default: DISCO_L475VG_IOT01A
  choices:
    DISCO_L475VG_IOT01A:
      short: DISCO_L475VG_IOT01A
      settings:
        MBED_TARGET: DISCO_L475VG_IOT01A
        UPLOAD_METHOD: J-Link
```

Any hint?

Thanks!

JohnK1987, 2022-09-05:
Hello,

I also followed these instructions and I am able to Build and Flash. 
The `cmake-variants.yaml` file seems to be OK. Do you have also installed the **Ninja Build** (as mentioned in point 2)? Because when I move the `ninja.exe` out of `C:\Program Files\CMake\bin\` folder, then I got similar error about missing generator.

BR, Jan

JohnK1987, 2022-09-06:
Oh, according the guide it have to be **JLINK** and not **J-LINK**

Cmake is trying find the installation of J-link here https://github.com/mbed-ce/mbed-os/blob/535abc7515df67c8f4ab06b956ff50e3aa1c4862/tools/cmake/upload_methods/FindJLINK.cmake#L21

This is the output during project configuration
```
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Mbed: Loading default upload method configuration from targets/upload_method_cfg/NUCLEO_F767ZI.cmake
[cmake] -- Detected VS Code IDE, will generate VS Code debug configurations
[cmake] -- Mbed: Not building any Mbed OS tests.
[cmake] -- Board upload method set to JLINK
[cmake] -- Found JLINK: C:/Program Files/SEGGER/JLink/JLink.exe
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: C:/...
```

BR, Jan

maiorfi, 2022-09-07:
Ok, that solved latest issue, thanks.

I tried to cook a cmake-variants.yaml containing a target not explicitly supported by mbed-ce (i.e. DISCO_L475VG_IOT01A).

What does "unsupported" really mean? It seems that I succeeded in building a valid (i.e. flashable and debuggable) .elf image for DISCO_L475VG_IOT01A, despite I couldn't make it work toggling a led or printf-ing something on debug emulator serial.

closed 2022-09-07 by maiorfi

JohnK1987, 2022-09-07:
I'm not sure, this question is more for @multiplemonomials, but I suppose it is only because Jamie want mark boards where are already prepared all necessary things or something like that.
From my understanding all boards supported on MbedOs6 should work here, but for boards what are not on [this list](https://github.com/mbed-ce/mbed-os/wiki/Mbed-CE-Officially-Supported-Targets-&-Features) you need manually add/create [config files](https://github.com/mbed-ce/mbed-os/tree/master/targets/upload_method_cfg) for Upload methods.

BR, Jan

multiplemonomials, 2022-09-07:
Yes, unsupported means that the board doesn't have upload method configuration and you'll have to create it.

Incidentally, @JohnK1987 could you make an MR to submit your NUCLEO_F767ZI.cmake?

JohnK1987, 2022-09-07:
> Incidentally, @JohnK1987 could you make a PR to submit your NUCLEO_F767ZI.cmake?
I can try.

BR, Jan

maiorfi, 2022-09-11:
Maybe this issue isn't the right place anymore, but is it really so simple to add a new target (at least a new STM32 based one)? I could try to add those related to boards I own, so to be able to run some simple test on real hardware before creating a PR.

JohnK1987, 2022-09-11:
Let's not call it a new target

* A **new target** could be a target what is not supported in MbedOS and there is necessary to implement everything around it like HALs, pin map and so on.
* A **custom target** inherit from already existed target under MbedOS.

This case is not about any option above. [Upload Method Configuration](https://github.com/mbed-ce/mbed-os/tree/master/targets/upload_method_cfg#upload-method-configuration) is just upload configuration.

BR, Jan

multiplemonomials, 2022-09-11:
> I could try to add those related to boards I own, so to be able to run some simple test on real hardware before creating a PR.

Good point!  I made a new doc page that shows you how to run the Mbed tests locally with your own board.  Can you try it out for me if you have a chance?  https://github.com/mbed-ce/mbed-os/wiki/Running-the-Mbed-Greentea-Tests-Locally

Also yeah, I think I created some confusion with what I said earlier.  Currently, there are a lot of boards that are supported by Mbed OS but unsupported by the upload method system.