mbed-ce/mbed-os #44: First time Local Greentea

JohnK1987, 2022-09-11:
@multiplemonomials I probably missing something, I followed https://github.com/mbed-ce/mbed-os/wiki/Running-the-Mbed-Greentea-Tests-Locally, but all test failed because the test program can not connect to serial port for some reason. I am sure I have set correct port. I closed all app and for sure also restart my PC. Nothing change. Any hint?
<details><summary>GreemTea console output</summary>

```python
C:\Users\User\MbedCE_GreenTea\mbed-os\build_Nucleo_F767ZI>ctest . --output-on-failure
Test project C:/Users/User/MbedCE_GreenTea/mbed-os/build_Nucleo_F767ZI
      Start  1: mbed-drivers-ticker
 1/68 Test  #1: mbed-drivers-ticker ............................***Failed   66.23 sec
[1/1] Flashing mbed-drivers-ticker with STM32CubeProg...
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.11.0
      -------------------------------------------------------------------

ST-LINK SN  : 0671FF504955657867121730
ST-LINK FW  : V2J40M27
Board       : NUCLEO-F767ZI
Voltage     : 3.24V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x451
Revision ID : Rev Z
Device name : STM32F76x/STM32F77x
Flash size  : 2 MBytes
Device type : MCU
Device CPU  : Cortex-M7
BL Version  : --

Memory Programming ...
Opening and parsing file: mbed-drivers-ticker.elf
  File          : mbed-drivers-ticker.elf
  Size          : 51.59 KB
  Address       : 0x08000000

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 1]
Download in Progress:
██████████████████████████████████████████████████ 100%

File download complete
Time elapsed during download operation: 00:00:01.105

MCU Reset

Software reset is performed
[1662930764.74][HTST][INF] host test executor ver. 1.8.14
[1662930764.74][HTST][INF] copy image onto target... SKIPPED!
[1662930764.76][HTST][INF] starting host test process...
[1662930765.72][CONN][INF] starting connection process...
[1662930765.72][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1662930765.72][CONN][INF] initializing serial port listener...
[1662930765.72][HTST][INF] setting timeout to: 60 sec
[1662930765.72][SERI][INF] using specified port '3'
[1662930765.72][SERI][INF] serial(port=3, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1662930765.73][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930765.73][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930766.74][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930766.74][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930767.74][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930767.74][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930768.75][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930768.75][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930769.77][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930769.77][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930770.77][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930770.77][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930771.79][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930771.79][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930772.80][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930772.80][SERI][ERR] Retry after 1 sec until 60 seconds
[... 100 lines cut ...]
[1662930824.26][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930824.26][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930825.28][SERI][ERR] could not open port '3': FileNotFoundError(2, 'the system cannot find the specified file.', None, 2)
[1662930825.28][SERI][ERR] Retry after 1 sec until 60 seconds
[1662930826.23][HTST][INF] test suite run finished after 60.50 sec...
[1662930826.29][CONN][ERR] Failed to connect to resource
[1662930826.39][HTST][INF] CONN exited with code: 0
[1662930826.39][HTST][INF] Some events in queue
[1662930826.39][HTST][INF] stopped consuming events
[1662930826.39][HTST][INF] host test result(): None
[1662930826.39][HTST][WRN] missing __exit event from DUT
[1662930826.39][HTST][WRN] missing __exit_event_queue event from host test
[1662930826.39][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1662930826.39][HTST][INF] calling blocking teardown()
[1662930826.39][HTST][INF] teardown() finished
[1662930826.39][HTST][INF] {{result;timeout}}
CMake Error at mbed-run-greentea-mbed-drivers-ticker.cmake:11 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 5"
```
</details>

Thank you
BR, Jan

multiplemonomials, 2022-09-11:
Interesting.  What did you pass for the `-DMBED_GREENTEA_SERIAL_PORT` option?  If you connect to your board in e.g. TeraTerm, what COM port is it at?

JohnK1987, 2022-09-12:
Ok, I found the problem... I was probably tired, blind or  stupid last night, I figured it out in 10sec today morning.
I used correct port number, but in incorrect format. I type just number 3 instead of COM3...

* wrong one `cmake .. -GNinja -DMBED_TARGET=NUCLEO_F767ZI -DCMAKE_BUILD_TYPE=Develop -DMBED_BUILD_GREENTEA_TESTS=ON -DMBED_GREENTEA_SERIAL_PORT=3 -DUPLOAD_METHOD=STM32CUBE`
* correct one `cmake .. -GNinja -DMBED_TARGET=NUCLEO_F767ZI -DCMAKE_BUILD_TYPE=Develop -DMBED_BUILD_GREENTEA_TESTS=ON -DMBED_GREENTEA_SERIAL_PORT=COM3 -DUPLOAD_METHOD=STM32CUBE`

<details><summary>GreemTea console output</summary>

```python
C:\Users\User\MbedCE_GreenTea\mbed-os\build_Nucleo_F767ZI>ctest . --output-on-failure
Test project C:/Users/User/MbedCE_GreenTea/mbed-os/build_Nucleo_F767ZI
      Start  1: mbed-drivers-ticker
 1/68 Test  #1: mbed-drivers-ticker ............................   Passed   30.99 sec
      Start  2: mbed-hal-common-tickers
 2/68 Test  #2: mbed-hal-common-tickers ........................***Failed   85.18 sec
[1/1] Flashing mbed-hal-common-tickers with STM32CubeProg...
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.11.0
      -------------------------------------------------------------------

ST-LINK SN  : 0671FF504955657867121730
ST-LINK FW  : V2J40M27
Board       : NUCLEO-F767ZI
Voltage     : 3.25V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x451
Revision ID : Rev Z
Device name : STM32F76x/STM32F77x
Flash size  : 2 MBytes
Device type : MCU
Device CPU  : Cortex-M7
BL Version  : --

Memory Programming ...
Opening and parsing file: mbed-hal-common-tickers.elf
  File          : mbed-hal-common-tickers.elf
  Size          : 53.02 KB
  Address       : 0x08000000

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 1]
Download in Progress:
██████████████████████████████████████████████████ 100%

File download complete
Time elapsed during download operation: 00:00:01.110

MCU Reset

Software reset is performed
[1662956479.18][HTST][INF] host test executor ver. 1.8.14
[1662956479.18][HTST][INF] copy image onto target... SKIPPED!
[1662956479.18][HTST][INF] starting host test process...
[1662956480.13][CONN][INF] starting connection process...
[1662956480.13][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1662956480.13][CONN][INF] initializing serial port listener...
[1662956480.13][SERI][INF] using specified port 'COM3'
[1662956480.13][SERI][INF] serial(port=COM3, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1662956480.13][HTST][INF] setting timeout to: 60 sec
[1662956480.13][SERI][INF] reset device using 'default' plugin...
[1662956480.39][SERI][INF] waiting 1.00 sec after reset
[1662956481.40][SERI][INF] wait for it...
[1662956481.43][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1662956481.43][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1662956481.43][CONN][INF] sending preamble '7b065312-37d2-4f96-a3f3-7a236f11997a'
[1662956481.47][SERI][TXD] {{__sync;7b065312-37d2-4f96-a3f3-7a236f11997a}}
[1662956481.54][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1662956481.59][CONN][INF] found SYNC in stream: {{__sync;7b065312-37d2-4f96-a3f3-7a236f11997a}} it is #0 sent, queued...
[... 335 lines cut ...]

Total Test time (real) = 858.00 sec

The following tests did not run:
         10 - mbed-hal-ospi (Skipped)
         12 - mbed-hal-qspi (Skipped)
         39 - mbed-platform-crash-reporting (Skipped)
         49 - mbed-platform-stats-cpu (Skipped)
         50 - mbed-platform-stats-heap (Skipped)
         51 - mbed-platform-stats-sys (Skipped)
         52 - mbed-platform-stats-thread (Skipped)
         67 - mbed-connectivity-nfc-eeprom (Skipped)

The following tests FAILED:
          2 - mbed-hal-common-tickers (Failed)
         68 - mbed-connectivity-netsocket-udp (Failed)
Errors while running CTest

C:\Users\User\MbedCE_GreenTea\mbed-os\build_Nucleo_F767ZI>

```
</details>

Sorry and thank youi
BR, Jan

multiplemonomials, 2022-09-12:
Ah, for the connectivity test, do you have the board's Ethernet port plugged into a network with Internet access?  It needs to be in order for the test to pass.  Should really document that...

Regarding the other failure, that looks like there's a problem with the LP ticker not delivering interrupts when it's supposed to, so the test case times out.  To me it looks like that's an actual Mbed bug.  When I have a chance I'll write up some instructions for you on how to debug a unit test failure.

JohnK1987, 2022-09-12:
> do you have the board's Ethernet port plugged into a network with Internet access?

Nope, but I understood it as I saw the output, however I do not have time for start it again for now, I must go to work. I will run it today later.

BR, Jan

JohnK1987, 2022-09-12:
Ok, I did it again with connected cable and the `mbed-connectivity-netsocket-udp` test passed.
```
      Start 68: mbed-connectivity-netsocket-udp
68/68 Test #68: mbed-connectivity-netsocket-udp ................   Passed   25.40 sec
```
Additional question. Is somehow possible to run only specific tests? For example, I could run only the mbed-connectivity-netsocket-udp one, instead of whole test scenario.

BR, Jan

multiplemonomials, 2022-09-18:
Doc created: https://github.com/mbed-ce/mbed-os/wiki/Debugging-Failing-Greentea-Tests
This should explain how to debug these sort of errors.

If you have time, you should be able to use that to run and debug the failing LP Ticker test.

JohnK1987, 2022-09-19:
@multiplemonomials I am probably blind. I tried follow the guide, but I can not found any line like this
>13: Executing: mbedhtrun...

<details><summary>Output of  command: ctest -V -R mbed-hal-reset-reason</summary>

```python
C:\Users\User\MbedCE_GreenTea\mbed-os\build_NUCLEO_F767ZI>ctest -V -R mbed-hal-reset-reason
UpdateCTestConfiguration  from :C:/Users/User/MbedCE_GreenTea/mbed-os/build_NUCLEO_F767ZI/DartConfiguration.tcl
Parse Config file:C:/Users/User/MbedCE_GreenTea/mbed-os/build_NUCLEO_F767ZI/DartConfiguration.tcl
UpdateCTestConfiguration  from :C:/Users/User/MbedCE_GreenTea/mbed-os/build_NUCLEO_F767ZI/DartConfiguration.tcl
Parse Config file:C:/Users/User/MbedCE_GreenTea/mbed-os/build_NUCLEO_F767ZI/DartConfiguration.tcl
Test project C:/Users/User/MbedCE_GreenTea/mbed-os/build_NUCLEO_F767ZI
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 13
    Start 13: mbed-hal-reset-reason

13: Test command: "C:\Program Files\CMake\bin\cmake.exe" "-P" "mbed-run-greentea-mbed-hal-reset-reason.cmake"
13: Working Directory: C:/Users/User/MbedCE_GreenTea/mbed-os/build_NUCLEO_F767ZI/hal/tests/TESTS/mbed_hal/reset_reason
13: Test timeout computed to be: 1500
13: [1/1] Flashing mbed-hal-reset-reason with STM32CubeProg...
13:       -------------------------------------------------------------------
13:                        STM32CubeProgrammer v2.11.0
13:       -------------------------------------------------------------------
13:
13: ST-LINK SN  : 0671FF504955657867121730
13: ST-LINK FW  : V2J40M27
13: Board       : NUCLEO-F767ZI
13: Voltage     : 3.24V
13: SWD freq    : 4000 KHz
13: Connect mode: Under Reset
13: Reset mode  : Hardware reset
13: Device ID   : 0x451
13: Revision ID : Rev Z
13: Device name : STM32F76x/STM32F77x
13: Flash size  : 2 MBytes
13: Device type : MCU
13: Device CPU  : Cortex-M7
13: BL Version  : --
13:
13:
13:
13: Memory Programming ...
13: Opening and parsing file: mbed-hal-reset-reason.elf
13:   File          : mbed-hal-reset-reason.elf
13:   Size          : 45.46 KB
13:   Address       : 0x08000000
13:
13:
13: Erasing memory corresponding to segment 0:
13: Erasing internal memory sectors [0 1]
13: Download in Progress:
██████████████████████████████████████████████████ 100%0%
13:
13: File download complete
13: Time elapsed during download operation: 00:00:01.020
13:
13: MCU Reset
13:
13: Software reset is performed
13: [1663605567.62][HTST][INF] host test executor ver. 1.8.14
13: [1663605567.62][HTST][INF] copy image onto target... SKIPPED!
13: [1663605567.64][HTST][INF] starting host test process...
13: [1663605568.64][CONN][INF] starting connection process...
13: [1663605568.64][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
13: [1663605568.64][CONN][INF] initializing serial port listener...
[... 143 lines cut ...]
13:   execute_process failed command indexes:
13:
13:     1: "Child return code: 1"
13:
13:
13:
1/1 Test #13: mbed-hal-reset-reason ............***Failed    2.59 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   3.07 sec

The following tests FAILED:
         13 - mbed-hal-reset-reason (Failed)
Errors while running CTest
Output from these tests are in: C:/Users/User/MbedCE_GreenTea/mbed-os/build_NUCLEO_F767ZI/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
```
</details>

I hope I understood correctly that is the reason why we need to collect the previous one and then add the command `--skip-reset`

Can you give me a hint please?

BR, Jan

multiplemonomials, 2022-09-19:
Oh sorry, forgot to mention!  To get that printout, you need to use my #51 PR, I just added it.

If you think the #51 change, approve the PR so I can merge it!  We set it now so an approving review is required.

JohnK1987, 2022-09-20:
That sounds logic :)
I use `gh pr checkout 51` for local test.
I will let you know with another progress today later. Thank you.

BR, Jan

JohnK1987, 2022-09-20:
Ok, I did a progress. I was able to run it in Debug mode

<details><summary>Debug output</summary>

```python
Breakpoint 1, ticker_interrupt_test () at C:/Users/User/MbedCE_GreenTea/mbed-os/hal/tests/TESTS/mbed_hal/common_tickers/main.cpp:220
220         uint32_t ticker_timeout[] = { 100, 200, 300, 400, 500, 600, 700, 800, 900, 1000, 1100, 1200 };
(gdb) n
222         const ticker_info_t *p_ticker_info = intf->get_info();
(gdb) n
224         overflow_protect();
(gdb) n
229             if (TICKS_TO_US(ticker_timeout[i], p_ticker_info->frequency) < (MAX_FUNC_EXEC_TIME_US + DELTA_FUNC_EXEC_TIME_US)) {
(gdb) n
233             run_count++;
(gdb) n
234             intFlag = 0;
(gdb) n
235             const uint32_t tick_count = intf->read();
(gdb) n
239             intf->set_interrupt(tick_count + ticker_timeout[i]);
(gdb) n
243             while (intf->read() < (tick_count + ticker_timeout[i] - TICKER_DELTA)) {
(gdb) n
245                 TEST_ASSERT_EQUAL_INT_MESSAGE(0, intFlag, "Interrupt fired too early");
(gdb) n
```
</details>

<details><summary>mbedhtrun output</summary>

```python
[1663702635.49][CONN][RXD] >>> Running case #12: 'lp ticker interrupt test'...
[1663702635.54][CONN][INF] found KV pair in stream: {{__testcase_start;lp ticker interrupt test}}, queued...
[1663702685.25][CONN][RXD] <greentea test suite>:245::FAIL: Expected 0 Was 1. Interrupt fired too early
[1663702685.30][CONN][INF] found KV pair in stream: {{__testcase_finish;lp ticker interrupt test;0;1}}, queued...
[1663702685.39][CONN][RXD] >>> 'lp ticker interrupt test': 0 passed, 1 failed with reason 'Assertion Failed'
[1663702685.42][CONN][RXD] <greentea test suite>:0::PASS
[1663702685.42][CONN][RXD]
[1663702685.49][CONN][RXD] >>> Test cases: 11 passed, 1 failed with reason 'Assertion Failed'
[1663702685.50][CONN][RXD] >>> TESTS FAILED!
[1663702685.50][CONN][RXD]
[1663702685.53][CONN][RXD] -----------------------
[1663702685.56][CONN][RXD] 0 Tests 0 Failures 0 Ignored
[1663702685.56][CONN][RXD] OK
[1663702685.59][CONN][INF] found KV pair in stream: {{__testcase_summary;11;1}}, queued...
[1663702685.61][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1663702685.61][HTST][INF] __notify_complete(False)
[1663702685.61][HTST][INF] __exit_event_queue received
[1663702685.61][HTST][INF] test suite run finished after 60.35 sec...
[1663702685.63][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1663702685.63][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1663702685.75][HTST][INF] CONN exited with code: 0
[1663702685.75][HTST][INF] Some events in queue
[1663702685.75][HTST][INF] stopped consuming events
[1663702685.75][HTST][INF] host test result() call skipped, received: False
[1663702685.75][HTST][WRN] missing __exit event from DUT
[1663702685.75][HTST][INF] calling blocking teardown()
[1663702685.75][HTST][INF] teardown() finished
[1663702685.75][HTST][INF] {{result;failure}}

C:\Users\User\MbedCE_GreenTea\mbed-os\build_NUCLEO_F767ZI>
```
</details>

I need to read something about gdb via console, before I continue.

BR, Jan

JohnK1987, 2022-09-24:
Hello @multiplemonomials, 
I have a question for you. You updated the [Toolchain Setup Guide](https://github.com/mbed-ce/mbed-os/wiki/Toolchain-Setup-Guide#on-windows) with a picture of GNU 11.3 some time ago. Do you really use GNU 11.3?
I am asking because when I tried compile a simple blinky program some time ago, I were not able to compile MbedOS for a target. But how I change GNU from version11.3 to 9.3 all tests were OK.
Same reason seems to be also for the failing LP Ticker test. I found the GreenTea was build with GNU 11, so I rebuild the GreenTea under GNU 9 and run again. All tests were passed

<details><summary>GreenTea result with GNU 9.3.1</summary>

```python
C:\Users\User\MbedCE_GreenTea\mbed-os\build_NUCLEO_F767ZI>ctest . --output-on-failure
Test project C:/Users/User/MbedCE_GreenTea/mbed-os/build_NUCLEO_F767ZI
      Start  1: mbed-drivers-ticker
 1/68 Test  #1: mbed-drivers-ticker ............................   Passed   31.17 sec
      Start  2: mbed-hal-common-tickers
 2/68 Test  #2: mbed-hal-common-tickers ........................   Passed   86.61 sec
      Start  3: mbed-hal-common-tickers-freq
 3/68 Test  #3: mbed-hal-common-tickers-freq ...................   Passed   26.65 sec
      Start  4: mbed-hal-crc
 4/68 Test  #4: mbed-hal-crc ...................................   Passed    7.37 sec
      Start  5: mbed-hal-critical-section
 5/68 Test  #5: mbed-hal-critical-section ......................   Passed    6.25 sec
      Start  6: mbed-hal-flash-functional-tests
 6/68 Test  #6: mbed-hal-flash-functional-tests ................   Passed   10.06 sec
      Start  7: mbed-hal-lp-ticker
 7/68 Test  #7: mbed-hal-lp-ticker .............................   Passed    6.58 sec
      Start  8: mbed-hal-minimum-requirements
 8/68 Test  #8: mbed-hal-minimum-requirements ..................   Passed    5.90 sec
      Start  9: mbed-hal-mpu
 9/68 Test  #9: mbed-hal-mpu ...................................   Passed    6.87 sec
      Start 10: mbed-hal-ospi
10/68 Test #10: mbed-hal-ospi ..................................***Skipped   0.02 sec
      Start 11: mbed-hal-pinmap
11/68 Test #11: mbed-hal-pinmap ................................   Passed    6.46 sec
      Start 12: mbed-hal-qspi
12/68 Test #12: mbed-hal-qspi ..................................***Skipped   0.02 sec
      Start 13: mbed-hal-reset-reason
13/68 Test #13: mbed-hal-reset-reason ..........................   Passed   10.99 sec
      Start 14: mbed-hal-rtc
14/68 Test #14: mbed-hal-rtc ...................................   Passed   49.52 sec
      Start 15: mbed-hal-rtc-reset
15/68 Test #15: mbed-hal-rtc-reset .............................   Passed   11.36 sec
      Start 16: mbed-hal-rtc-time
16/68 Test #16: mbed-hal-rtc-time ..............................   Passed   10.25 sec
      Start 17: mbed-hal-rtc-time-conv
17/68 Test #17: mbed-hal-rtc-time-conv .........................   Passed   24.61 sec
      Start 18: mbed-hal-sleep
18/68 Test #18: mbed-hal-sleep .................................   Passed   10.51 sec
      Start 19: mbed-hal-sleep-manager
19/68 Test #19: mbed-hal-sleep-manager .........................   Passed   10.13 sec
      Start 20: mbed-hal-sleep-manager-racecondition
20/68 Test #20: mbed-hal-sleep-manager-racecondition ...........   Passed   18.78 sec
      Start 21: mbed-hal-stack-size-unification
21/68 Test #21: mbed-hal-stack-size-unification ................   Passed    5.60 sec
      Start 22: mbed-hal-ticker
22/68 Test #22: mbed-hal-ticker ................................   Passed   18.34 sec
      Start 23: mbed-hal-trng
23/68 Test #23: mbed-hal-trng ..................................   Passed    7.71 sec
      Start 24: mbed-hal-us-ticker
24/68 Test #24: mbed-hal-us-ticker .............................   Passed    5.66 sec
      Start 25: mbed-hal-watchdog
25/68 Test #25: mbed-hal-watchdog ..............................   Passed   12.74 sec
      Start 26: mbed-hal-watchdog-reset
26/68 Test #26: mbed-hal-watchdog-reset ........................   Passed   16.41 sec
      Start 27: mbed-hal-watchdog-timing
27/68 Test #27: mbed-hal-watchdog-timing .......................   Passed   33.56 sec
      Start 28: mbed-hal-verify-arduino-pinmap
28/68 Test #28: mbed-hal-verify-arduino-pinmap .................   Passed   13.95 sec
      Start 29: mbed-platform-mbed-functional-callback
29/68 Test #29: mbed-platform-mbed-functional-callback .........   Passed    7.67 sec
      Start 30: mbed-platform-mbed-functional-callback-big
30/68 Test #30: mbed-platform-mbed-functional-callback-big .....   Passed    7.45 sec
[... 72 lines cut ...]
      Start 67: mbed-connectivity-nfc-eeprom
67/68 Test #67: mbed-connectivity-nfc-eeprom ...................***Skipped   0.02 sec
      Start 68: mbed-connectivity-netsocket-udp
68/68 Test #68: mbed-connectivity-netsocket-udp ................   Passed   25.43 sec

100% tests passed, 0 tests failed out of 68

Total Test time (real) = 820.10 sec

The following tests did not run:
         10 - mbed-hal-ospi (Skipped)
         12 - mbed-hal-qspi (Skipped)
         39 - mbed-platform-crash-reporting (Skipped)
         49 - mbed-platform-stats-cpu (Skipped)
         50 - mbed-platform-stats-heap (Skipped)
         51 - mbed-platform-stats-sys (Skipped)
         52 - mbed-platform-stats-thread (Skipped)
         67 - mbed-connectivity-nfc-eeprom (Skipped)

```
</details>

So what now? What is wrong? This differences could have impact to many more things.

BR, Jan

multiplemonomials, 2022-09-24:
Huh, looks like my CI slave machine uses GCC 8.3.1, and the test passes with this version.  So seems like this might be compiler optimization related? Scary

multiplemonomials, 2022-09-24:
I confirm that it fails on my desktop with GCC 11 and NUCLEO_L452RE.  Half the time the ISR fires too early, and the other half of the time it never fires.

Additionally, I noticed an issue that was causing the test failures to not show up.  Whoops. https://github.com/mbed-ce/mbed-os/pull/56

multiplemonomials, 2022-09-24:
Aha!  I managed to track down the cause of this issue, by running the test multiple times until it hung, then starting OpenOCD without resetting the device.  (the issue went away if I added prints or ran in the debugger from boot).

Turns out it's actually an integer overflow issue in the common tickers test, and it's sensitive to the *initial time on the ticker when the test starts*.  That's why it's so random to reproduce.

multiplemonomials, 2022-09-25:
Fixed in #57

closed 2022-09-25 by multiplemonomials