mbed-ce/mbed-os #48: Create guide + API for custom targets with source files

multiplemonomials, 2022-09-17:
Currently, custom target support in Mbed CE is incomplete and we need to finish it out.

There are basically three types of custom targets:
1. Custom targets with configuration settings only (in custom_targets.json)
2. Custom targets which add additional source files
3. Custom targets which add source files _and override existing Mbed sources_

Currently, the first type is supported properly, and the second type should work (by defining your own CMake target matching the custom target name), but has not been tested yet.  However, the third type is not currently possible.

First of all, we need to create a new CMake function, something like
```cmake
mbed_remove_source_files_from_target()
```
This will enable the third type of target to be created.

Additionally, we need to create a comprehensive wiki page that explains how to create custom targets.  This should cover both how to write the custom_targets.json (this has been a pain point for me in the past!), and how to set up the CMake stuff so that the custom target works.

multiplemonomials, 2023-02-14:
Done: https://github.com/mbed-ce/mbed-ce-custom-targets

closed 2023-02-14 by multiplemonomials

JohnK1987, 2023-02-14:
I added the link of readme of Custom target example into the [Wiki Home page](https://github.com/mbed-ce/mbed-os/wiki#documentation-and-information)

BR, Jan