mbed-ce/mbed-os #50: Make version different from Mbed OS

multiplemonomials, 2022-09-18:
The Mbed version info is stored in this file: https://github.com/mbed-ce/mbed-os/blob/master/platform/include/platform/mbed_version.h

We should probably change it so that it's different from mainline Mbed.  Since there are some breaking changes in Mbed CE (and others yet to be made, like #13), should we change it to 7.0.0?  Or should we change it to like 6.99 until the breaking changes are mostly over with, and then change it to 7.0.0?

Also, we should fix the "99 is default value for development version" comments in that header file, they are not correct.

JohnK1987, 2022-09-18:
You reworked Mbed build system (I do not know if this is completely done or not, because I don't know your roadmap), so that seems to be a MAJOR change. 
However, it depends in which state you want to release the first release under Mbed CE brand.

Yeah, I never saw 99, from my point of view the master branch = develop version without any marking in `mbed_version.h`.

BR, Jan

multiplemonomials, 2022-09-18:
FWIW, I'm pretty much done actively making changes to the build system.  Just need to flesh out the things I already added, like adding more upload method configs and finishing custom target support.