mbed-ce/mbed-os #53: STMicro devices do not pass USB endpoint halt test

multiplemonomials, 2022-09-22:
This causes the usb device basic test to fail in CI.  See old (unfixed) Mbed issue here: https://github.com/ARMmbed/mbed-os/issues/13215

JohnK1987, 2022-09-26:
Ok, if I understand it correctly...
Current function [endpoint_abort](https://github.com/mbed-ce/mbed-os/blob/60a3a4e5241b232bd6e89a2a5207b6539422a592/targets/TARGET_STM/USBPhy_STM32.cpp#L634) for STM32F7 is exactly same as [endpoint_remove](https://github.com/mbed-ce/mbed-os/blob/60a3a4e5241b232bd6e89a2a5207b6539422a592/targets/TARGET_STM/USBPhy_STM32.cpp#L579) and that is probably the reason why the test is not working, but the root cause probably is, there is not any HAL_PCD_EP_abort function for that.

However ST already released new version of STM32F7 HAL drivers where requested function already is - [Release v1.17.0](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Drivers/STM32F7xx_HAL_Driver)
* New [HAL_PCD_EP_abort](https://github.com/STMicroelectronics/STM32CubeF7/blob/f8bda023e34ce9935cb4efb9d1c299860137b6f3/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_pcd.h#L360) - currently implemented for F7,F4, L4
*  [old/current one](https://github.com/mbed-ce/mbed-os/blob/60a3a4e5241b232bd6e89a2a5207b6539422a592/targets/TARGET_STM/TARGET_STM32F7/STM32Cube_FW/STM32F7xx_HAL_Driver/stm32f7xx_hal_pcd.h#L346) does not have it.

Exactly same situation seems to be also for all STM32 families where the USB is implemented. So the solution seems to be update HAL drivers for affected families.

What do you think @multiplemonomials?

BR, Jan

multiplemonomials, 2022-09-26:
Nice find!  Make a PR and we can test it.

JohnK1987, 2022-09-27:
I tried it locally, but unfortunately it seems still not working. I will continue with investigating.

JohnK1987, 2023-02-13:
* F4 family is also affected by this, see Greentea log of Blackpill - https://github.com/mbed-ce/mbed-os/pull/123
* H7 family is also affected by this, see  Greentea logs below
<details><summary>Greentea result of mbed-usb-device-basic test for Nucleo-H743ZI2 </summary>

```python
 2/72 Test  #2: mbed-usb-device-basic ..........................***Failed   17.36 sec
[1/1] Flashing mbed-usb-device-basic with STM32CubeProg...
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.11.0
      -------------------------------------------------------------------

ST-LINK SN  : 002000243156501320323443
ST-LINK FW  : V3J10M3
Board       : NUCLEO-H743ZI
Voltage     : 3.30V
SWD freq    : 24000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x450
Revision ID : Rev V
Device name : STM32H7xx
Flash size  : 2 MBytes
Device type : MCU
Device CPU  : Cortex-M7
BL Version  : 0x90

Memory Programming ...
Opening and parsing file: mbed-usb-device-basic.elf
Warning: File corrupted. Two or more segments defines the same memory zone
  File          : mbed-usb-device-basic.elf
  Size          : 75.42 KB
  Address       : 0x08000000

Erasing memory corresponding to segment 0:
Erasing internal memory sector 0
Download in Progress:
██████████████████████████████████████████████████ 100%

File download complete
Time elapsed during download operation: 00:00:01.251

MCU Reset

Software reset is performed
Executing: mbedhtrun --skip-flashing -p COM27 -e C:/Users/USER/MbedCE/MbedCE_Testerino/mbed-os/drivers/usb/tests/TESTS/usb_device/basic/../../host_tests -m NUCLEO_H743ZI2 --baud-rate=115200
[1676317313.70][HTST][INF] host test executor ver. 1.8.14
[1676317313.71][HTST][INF] copy image onto target... SKIPPED!
[1676317313.71][HTST][INF] starting host test process...
[1676317315.98][CONN][INF] starting connection process...
[1676317315.98][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1676317315.98][CONN][INF] initializing serial port listener...
[1676317315.98][SERI][INF] using specified port 'COM27'
[1676317315.98][HTST][INF] setting timeout to: 60 sec
[1676317315.98][SERI][INF] serial(port=COM27, baudrate=115200, read_timeout=0.01, write_timeout=5)
[1676317315.98][SERI][INF] reset device using 'default' plugin...
[1676317316.24][SERI][INF] waiting 1.00 sec after reset
[1676317317.24][SERI][INF] wait for it...
[1676317317.24][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1676317317.24][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1676317317.24][CONN][INF] sending preamble 'd38e994c-223f-4794-943c-ea8669e8f112'
[1676317317.25][SERI][TXD] {{__sync;d38e994c-223f-4794-943c-ea8669e8f112}}
[1676317317.26][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1676317317.26][CONN][INF] found SYNC in stream: {{__sync;d38e994c-223f-4794-943c-ea8669e8f112}} it is #0 sent, queued...
[1676317317.26][HTST][INF] sync KV found, uuid=d38e994c-223f-4794-943c-ea8669e8f112, timestamp=1676317317.260016
[1676317317.28][CONN][RXD] >>> Running 13 test cases...
[1676317317.28][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1676317317.28][CONN][INF] found KV pair in stream: {{__timeout;180}}, queued...
[1676317317.28][CONN][INF] found KV pair in stream: {{__host_test_name;pyusb_basic}}, queued...
[1676317317.28][CONN][INF] found KV pair in stream: {{__testcase_count;13}}, queued...
[1676317317.28][CONN][INF] found KV pair in stream: {{__testcase_name;usb control basic test}}, queued...
[... 522 lines cut ...]
[1676231090.54][CONN][RXD] FAIL
[1676231090.56][CONN][INF] found KV pair in stream: {{__testcase_summary;8;1}}, queued...
[1676231090.56][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1676231090.56][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1676231090.56][HTST][INF] __exit(0)
[1676231090.56][HTST][INF] __notify_complete(False)
[1676231090.56][HTST][INF] __exit_event_queue received
[1676231090.56][HTST][INF] test suite run finished after 10.13 sec...
[1676231090.57][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1676231090.67][HTST][INF] CONN exited with code: 0
[1676231090.67][HTST][INF] Some events in queue
[1676231090.67][HTST][INF] stopped consuming events
[1676231090.67][HTST][INF] host test result() call skipped, received: False
[1676231090.67][HTST][INF] calling blocking teardown()
[1676231090.67][HTST][INF] teardown() finished
[1676231090.67][HTST][INF] {{result;failure}}
CMake Error at mbed-run-greentea-mbed-usb-device-basic.cmake:20 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

```
</details>

BR, Jan