mbed-ce/mbed-os #63: A better build system

tripathiu, 2022-10-01:
Hi @multiplemonomials,

First of all thank you for creating this project. A lot of work that I do depends on mbed-os and their sudden disappearance got a lot of us anxious. Hopefully, this project keeps it alive and even makes it better than the original! 🏆

The most annoying thing about mbed-os in my view is the painfully slow compilation process. I think this is due to two main reasons.

1. Files that are not used are still compiled. This alone increases the compilation time significantly.
2. For every static library linked with `mbed-os`, mbed-os is compiled. For instance, if I have three static libraries that use mbed-os, it is compiled three times, plus one for the top level executable.

All this is due to keeping mbed-os as an `INTERFACE` target in the CMake build system. This problem is discussed multiple times, including [here](https://github.com/ARMmbed/mbed-os/issues/15259#issuecomment-1098113761). 

This property also causes weirdnesses in the `dot` graphs that CMake generates. Look at the output of this command:

```fish
cmake -B build -S . && \
cmake --graphviz=dot/ dot build && \
dot dot/dot -Tpdf -o dot/dot.pdf && \
mupdf ./dot/dot.pdf
```

It results in a very convoluted looking graph even for minimal application, like this...

[image]

------------------------------

Here, I must ask `mbed-ce` to take an important step to fixing this most annoying problem with mbed-os, and _make it great again_.  I would be glad if I could contribute to fixing this problem. 

Thanks and cheers. 🍻

multiplemonomials, 2022-10-01:
Glad that you're getting some use out of the project!
Oh, this is a pretty easy fix.  You just need to pass the name of your target, like:
```
cmake -DMBED_TARGET=<your target name> -B build -S . && cmake --graphviz=dot/dot build && dot dot/dot -Tpdf -o dot/dot.pdf
```
See here for the command line build instructions: https://github.com/mbed-ce/mbed-os/wiki/Project-Setup:-Command-Line

However, I will say that I don't have terribly polished support for custom targets at this point (see #48).  I can try to set up some docs for you to use in the next week or two though.

multiplemonomials, 2022-10-01:
And regarding your original question, yes indeed I have almost completely gotten rid of interface targets in Mbed CE.  The core mbed code is compiled into a single library (well, two possible versions, mbed-baremetal and mbed-os).  And optional libraries are handled as STATIC libraries, not INTERFACE or OBJECT.  This means that no code should ever get compiled more than once!

tripathiu, 2022-10-01:
Thank you for your response. I guess it was my cmake configuration causing the issue. I tried it with a blank project and it works. 

> optional libraries are handled as STATIC libraries, not INTERFACE or OBJECT

That's really awesome. I would like to explore a little more `mbed-ce`. Seems like there is already a lot of work put into it.