mbed-ce/mbed-os #67: Serial corruption on STM32L4 at 115200 baud

multiplemonomials, 2022-10-04:
It seems like the STM32L4 is experiencing quite serious corruption on its UART transmissions when run at 115200 baud.  This is enough to make a few test cases out of each run fail due to missed control messages.  Need to look into what's causing this

multiplemonomials, 2022-10-23:
I think this might be an oscillator problem on my dev board.  Adding 
```
        "STM32L452xE": {
            "target.clock_source": "USE_PLL_HSI"
        }
```
to mbed_app.json seems to fix it.  Additionally, I bought a new logic analyzer that is capable of taking long-duration captures, and managed to zoom in on the bytes where an issue is happening:
[image]
[image]
The 'p' byte is supposed to be a '0', but an additional 4us have been added into the logic low in the first half of the byte.  This shifts the MSBits over by one, making it the wrong character.  I did not find anything in the code, datasheet, or errata explaining this -- it's like the UART bit clock just stopped for a few microseconds.  So, I suspect that maybe there's something up with the crystal oscillator that's causing its clock to slow down or stop for a few microseconds intermittently.

JojoS62, 2022-10-24:
have you checked the frequency of the oscillator? You can output the clock on gpio pin, check system_clock.c in the target files for DEBUG_MCO or HAL_RCC_MCOConfig.