mbed-ce/mbed-os #77: enable manual settings for launch.json for debugging

JojoS62, 2022-10-21:
I ran also into the problem #25  when starting to debug my F746 using JLink upload. After starting the debugger, it halted the cpu somewhere but did not reset. A manual reset with 'interrupt' and 'mon reset' was possible, but this not conveniant. I prefer also that a flash code is perfomed when debugging is startet (with "request": "launch").

It also looks like the behaviour with the internal servertypes is working faster and better, the servertype 'external' gets different settings. E.g., I have added my manual configuration to launch.json:

```json
{
    "configurations": [
		{
			"type": "cortex-debug",
			"name": "Connect to GDB HSDTriggerController NUCLEO_F746ZG Debug",
			"executable": "D:/Projects/Sn/LocalGit/Mbed-CE/Test-F746/build/MyProgram.elf",
			"cwd": "${workspaceRoot}",
			"gdbPath": "C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.10/bin/arm-none-eabi-gdb.exe",
			"objdumpPath": "C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.10/bin/arm-none-eabi-objdump.exe",
			"servertype": "external",
			"gdbTarget": "localhost:23331",
			"request": "attach"
		},
		{
			"type": "cortex-debug",
			"name": "debug JLink",
			"executable": "D:/Projects/Sn/LocalGit/Mbed-CE/Test-F746/build/MyProgram.elf",
			"cwd": "${workspaceRoot}",
			"servertype": "jlink",
			"device": "STM32F746ZG",
			"runToEntryPoint": "main",
			"request": "launch"
		}
	]
}
```

The difference is that the mcu is halted, a download and reset is performed and the target runs to the given symbol 'main'. 
I have found the mbed_ide_debug_cfg_generator.cmake and UploadMethodManager.cmake where the launch.json is written, but don't quite understand all of the magic that ninja brings in.

How can we handle to keep existing manual entries in launch.json? I guess some upload method 'custom' or 'manual'  could leave the contents of launch.json untouched. Modifying just the created entry looks difficult.

Another difference with the servertype jlink is also that the disconnect works without stopping the target. The symbols for stop and disconnect look swapped, but that maybe a bug in cortex-debug.

JojoS62, 2022-10-21:
the commands for issuing a reset are different, an error is shown when external gdbserver (left pane) is called:

[image]

I guess these come from cortex-debug.

JojoS62, 2022-10-21:
I have tried the same now with STM32CUBE debug setting, and got nearly the same result. The command 'monitor reset halt' is not recognized with external server. The command 'monitor halt' is working, but not with JLink. It is bad that the gdbservers behave different.
So I suggest to use the internal serverypes where the commands are handled already correctly.

multiplemonomials, 2022-10-21:
Huh, I'd argue that that's a bug in Cortex Debug, I have never heard of `monitor halt reset` being a valid command, only `monitor halt` and `monitor reset`.  But yeah, I definitely see what you mean, seems like it's probably better to make use of the Cortex Debug plugin's built-in server features,

multiplemonomials, 2022-10-21:
But that said, this is a substantial amount of work that we'll need to do to convert options in CMake into the appropriate options for cortex-debug.  And it will probably need a fair amount of VS Code-specific testing... I wonder if the better option is to just have CMake set `overrideAttachCommands` to work around this issue.

If you want to play around with adding that, you can test that easily by going [here](https://github.com/mbed-ce/mbed-os/blob/7cd19915a2628436ab71cd4a893f7a51783e26bb/tools/cmake/mbed_ide_debug_cfg_generator.cmake#L84) and adding:
```
\"overrideAttachCommands\": \"monitor reset\\nmonitor halt\"
```
or something like that.  It's just building up a JSON string, but it's annoying because it has to be escaped.

JojoS62, 2022-10-21:
yes, I will also think about. But as I remember, cmake can also use files as a template and modify them. Maybe this is easier than writing every line for launch.json in code.
I've used the different servertypes also, I think that is not a big issue. There was also a request for BMP, I like this also because the config is simple, it needs to know only the USB port. I'm missing user variables in VSC for the com port, but this could be set in a cmake var.
So, another option than creating only one debug option is to write the whole collection of servertypes into lauch.json. Then it can be selected before starting the debug session, and the list of configurations for cmake is not so long.

multiplemonomials, 2023-01-22:
@JojoS62 Give this a try: https://github.com/mbed-ce/mbed-os/pull/129

multiplemonomials, 2023-05-08:
I think we can close this now

closed 2023-05-08 by multiplemonomials