mbed-ce/mbed-os #125: wrong define for LED1 and Button in BLACKPILL_F411CE

JojoS62, 2023-01-20:
The defines in PinNames.h for the new target contain '=' and ',' and using this symbols gives compiler errors. The defines should look like this:

```c
    // Generic signals namings
#define LED1        PC_13
    // Standardized button names
#define BUTTON1     USER_BUTTON
``` 

@JohnK1987

JohnK1987, 2023-01-20:
Hi Jojo, 

Ok, I will prepare a fix.
In the future i will avoid to make any change  online.

Thank you 
BR, Jan

JojoS62, 2023-01-20:
No problem, I have tested this target when I got problems with adding my new F303.

JohnK1987, 2023-01-20:
Okay, but everything bad is good for something else.
I made corrections on my local and this time I wanted make test, but huh, there was really strange behavior about BUTTON and LED1.
I found I did not noticed an info the LED1 is negated and then it works in reverse- `1 = OFF` and` 0 = ON`. And what's more, I found out that for the Button(KEY), what is connected to PA_0 pin, is necessary to set internal Pull Up. It doesn't work without it.

I will make PR and update board page.

BR, Jan

JohnK1987, 2023-01-21:
@JojoS62  do you have something more to this or we can close it?

JojoS62, 2023-01-21:
I will test it today. 
The negated LED output is very usual, I use also defines for on and off.

JojoS62, 2023-01-24:
fixed by #127

closed 2023-01-24 by JojoS62