mbed-ce/mbed-os #131: Greentea - mbed-hal-sleep-manager test is failing on STM32H7

JohnK1987, 2023-02-13:
Greentea - mbed-hal-sleep-manager test is failing on STM32H7 with assertion - [lp ticker sleep time incorrect](https://github.com/mbed-ce/mbed-os/blob/2d7f24285093e0c5135e5518a4d8ed54ff33fd0f/hal/tests/TESTS/mbed_hal/sleep_manager/main.cpp#L183)

<details><summary>Greentea result of mbed-hal-sleep-manager test for Nucleo-H743ZI2 </summary>

```python
     Start 23: mbed-hal-sleep-manager
23/72 Test #23: mbed-hal-sleep-manager .........................***Failed    5.90 sec
[1/1] Flashing mbed-hal-sleep-manager with STM32CubeProg...
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.11.0
      -------------------------------------------------------------------

ST-LINK SN  : 002000243156501320323443
ST-LINK FW  : V3J10M3
Board       : NUCLEO-H743ZI
Voltage     : 3.30V
SWD freq    : 24000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x450
Revision ID : Rev V
Device name : STM32H7xx
Flash size  : 2 MBytes
Device type : MCU
Device CPU  : Cortex-M7
BL Version  : 0x90

Memory Programming ...
Opening and parsing file: mbed-hal-sleep-manager.elf
Warning: File corrupted. Two or more segments defines the same memory zone
  File          : mbed-hal-sleep-manager.elf
  Size          : 53.23 KB
  Address       : 0x08000000

Erasing memory corresponding to segment 0:
Erasing internal memory sector 0
Download in Progress:
██████████████████████████████████████████████████ 100%

File download complete
Time elapsed during download operation: 00:00:01.052

MCU Reset

Software reset is performed
Executing: mbedhtrun --skip-flashing -p COM27 -e C:/Users/Odiin/MbedCE/MbedCE_Testerino/mbed-os/hal/tests/TESTS/mbed_hal/sleep_manager/../../host_tests -m NUCLEO_H743ZI2 --baud-rate=115200
[1676317617.77][HTST][INF] host test executor ver. 1.8.14
[1676317617.77][HTST][INF] copy image onto target... SKIPPED!
[1676317617.77][HTST][INF] starting host test process...
[1676317619.04][CONN][INF] starting connection process...
[1676317619.04][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1676317619.04][CONN][INF] initializing serial port listener...
[1676317619.04][SERI][INF] using specified port 'COM27'
[1676317619.04][HTST][INF] setting timeout to: 60 sec
[1676317619.04][SERI][INF] serial(port=COM27, baudrate=115200, read_timeout=0.01, write_timeout=5)
[1676317619.04][SERI][INF] reset device using 'default' plugin...
[1676317619.30][SERI][INF] waiting 1.00 sec after reset
[1676317620.32][SERI][INF] wait for it...
[1676317620.32][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1676317620.32][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1676317620.32][CONN][INF] sending preamble 'f9c3654e-218d-458f-97ea-52c4d8850f5a'
[1676317620.32][SERI][TXD] {{__sync;f9c3654e-218d-458f-97ea-52c4d8850f5a}}
[1676317620.33][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1676317620.35][CONN][RXD] >>> Running 4 test cases...
[1676317620.35][CONN][INF] found SYNC in stream: {{__sync;f9c3654e-218d-458f-97ea-52c4d8850f5a}} it is #0 sent, queued...
[1676317620.35][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1676317620.35][HTST][INF] sync KV found, uuid=f9c3654e-218d-458f-97ea-52c4d8850f5a, timestamp=1676317620.347276
[1676317620.35][CONN][INF] found KV pair in stream: {{__timeout;15}}, queued...
[1676317620.35][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1676317620.35][CONN][INF] found KV pair in stream: {{__testcase_count;4}}, queued...
[... 161 lines cut ...]
[1676231391.07][CONN][RXD] FAIL
[1676231391.08][CONN][INF] found KV pair in stream: {{__testcase_summary;2;1}}, queued...
[1676231391.08][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1676231391.08][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1676231391.08][HTST][INF] __exit(0)
[1676231391.08][HTST][INF] __notify_complete(False)
[1676231391.08][HTST][INF] __exit_event_queue received
[1676231391.08][HTST][INF] test suite run finished after 0.28 sec...
[1676231391.10][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1676231391.21][HTST][INF] CONN exited with code: 0
[1676231391.21][HTST][INF] Some events in queue
[1676231391.21][HTST][INF] stopped consuming events
[1676231391.21][HTST][INF] host test result() call skipped, received: False
[1676231391.21][HTST][INF] calling blocking teardown()
[1676231391.21][HTST][INF] teardown() finished
[1676231391.21][HTST][INF] {{result;failure}}
CMake Error at mbed-run-greentea-mbed-hal-sleep-manager.cmake:20 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

```
</details>

BR, Jan

multiplemonomials, 2023-09-29:
I think this should have been fixed by #184.  Give it a shot with master branch!

JohnK1987, 2023-09-30:
`Test #23: mbed-hal-sleep-manager .........................   Passed    8.37 sec`

closed 2023-09-30 by JohnK1987