mbed-ce/mbed-os #138: Custom Target + BLE - Link step failing

ladislas, 2023-02-22:
Creating an issue for easier tracking. I've been investigating but with no luck :( 

### Discussed in https://github.com/mbed-ce/mbed-os/discussions/136

<div type='discussions-op-text'>

<sup>Originally posted by **ladislas** February 20, 2023</sup>
I'm in the process of moving from the old USCRPL/mbed-cmake to mbed-ce

I've managed to make everything compile but the link step is failing with a lot of `undefined reference to __HeapLimit'`

```
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: extern/mbed-os/connectivity/FEATURE_BLE/libraries/libmbed-ble-cordio.a(BLEInstanceBaseImpl.cpp.obj): in function `hci_mbed_os_drv_write':
/Users/ladislas/dev/leka/LekaOS/extern/mbed-os/connectivity/FEATURE_BLE/source/cordio/source/BLEInstanceBaseImpl.cpp:83: undefined reference to `ble_cordio_get_hci_driver()'
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: extern/mbed-os/connectivity/FEATURE_BLE/libraries/libmbed-ble-cordio.a(BLEInstanceBaseImpl.cpp.obj): in function `hci_mbed_os_start_reset_sequence':
/Users/ladislas/dev/leka/LekaOS/extern/mbed-os/connectivity/FEATURE_BLE/source/cordio/source/BLEInstanceBaseImpl.cpp:89: undefined reference to `ble_cordio_get_hci_driver()'
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: extern/mbed-os/connectivity/FEATURE_BLE/libraries/libmbed-ble-cordio.a(BLEInstanceBaseImpl.cpp.obj): in function `hci_mbed_os_handle_reset_sequence':
/Users/ladislas/dev/leka/LekaOS/extern/mbed-os/connectivity/FEATURE_BLE/source/cordio/source/BLEInstanceBaseImpl.cpp:103: undefined reference to `ble_cordio_get_hci_driver()'
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: extern/mbed-os/connectivity/FEATURE_BLE/libraries/libmbed-ble-cordio.a(BLEInstanceBaseImpl.cpp.obj): in function `ble::impl::BLEInstanceBase::deviceInstance()':
/Users/ladislas/dev/leka/LekaOS/extern/mbed-os/connectivity/FEATURE_BLE/source/cordio/source/BLEInstanceBaseImpl.cpp:145: undefined reference to `ble_cordio_get_hci_driver()'
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: extern/mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_retarget.cpp.obj: in function `_sbrk':
/Users/ladislas/dev/leka/LekaOS/extern/mbed-os/platform/source/mbed_retarget.cpp:1528: undefined reference to `__HeapLimit'
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: extern/mbed-os/CMakeFiles/mbed-os.dir/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c.obj: in function `software_init_hook':
/Users/ladislas/dev/leka/LekaOS/extern/mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:52: undefined reference to `__StackLimit'
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /Users/ladislas/dev/leka/LekaOS/extern/mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:52: undefined reference to `__StackTop'
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /Users/ladislas/dev/leka/LekaOS/extern/mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:52: undefined reference to `__HeapLimit'
```

it seems related to BLE/Cordio

We are using BLE with `mbed-bluenrg-ms`

Any idea what I might be doing wrong?</div>

ladislas, 2023-02-27:
@multiplemonomials I've been trying to fix the issue but with no success.

do you have any idea what could be causing that?

multiplemonomials, 2023-02-27:
Apologies, I didn't get notified of this until now.  Is this code public anywhere, so I can try to build it?

multiplemonomials, 2023-03-01:
Okay!  I spent a bit going through your code and found basically three different issues that are causing your problems.  I fixed all three and was able to build the OS!

The first issue is, extremely long compile times due to spending ages linking a bunch of .a libraries.  The reason for this is, each .a library links to `mbed-os`.  As I recently discovered, CMake isn't very smart about .a libraries that link to OBJECT libraries, and it will actually end up including ALL of the objects from mbed-os into EACH .a library.  Hence the long link times.

Luckily, there's an easy fix.  Each of the libraries in your project needs to link to `mbed-core-flags` (and `mbed-rtos-flags` for libraries that use the RTOS) instead of `mbed-os`.  That will fix your long build times; I tried it out and it made a big difference (can push up a branch if you want).

The second issue is the 
```
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /Users/ladislas/dev/leka/LekaOS/extern/mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:52: undefined reference to `__StackTop'
/opt/homebrew/Cellar/arm-gcc-bin@10/10.3-2021.10_1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /Users/ladislas/dev/leka/LekaOS/extern/mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:52: undefined reference to `__HeapLimit'
```
This is because you need to add a call to
```cmake
mbed_set_post_build(LekaOS)
```
at the end of app/os/CMakeLists.txt.  Without this call, the linker script never gets attached to the target so linking will fail with undefined references and stuff.

The third issue, however, is a bona fide bug with Mbed OS.  The function `ble_cordio_get_hci_driver()` is defined by `libmbed-bluenrg-ms.a` but used by `libmbed-ble.a`.  This means that `libmbed-ble.a` needs to appear first on the link line, but CMake won't do this because it doesn't know about the dependency between these two libraries.

To fix this, a new link dependency needs to get added in the CMake build files.  I'll make a PR for that forthwith.

ladislas, 2023-03-01:
Amazing! Thank you so much @multiplemonomials!

I'll implement the fixes you suggested and will report here.

PR #145 (2023-03-01): Fix BlueNRG undefined reference to ble_cordio_get_hci_driver()
### Summary of changes 

Fixes #138 

#### Impact of changes 

With this change, if the BlueNRG components are used, mbed-ble can now be only used with the RTOS, not with mbed-baremetal.  That's what I had tried to avoid by making BlueNRG a different library...

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2023-03-01 by multiplemonomials