mbed-ce/mbed-os #174: Add "capability flag" for 0 length I2C transfers

multiplemonomials, 2023-07-23:
Even though 0-length I2C transfers (where the MCU just sends an address, waits for an ACK, and then stops) are useful (e.g. for scanning the I2C bus for devices), they are a bit of a corner case in the I2C protocol.  Some MCUs simply do not support them:
- STM32 v2 I2C peripheral allows transactional transfers with 0 length, but not single-byte transfers
- RP2040 does not allow 0-length transfers at all in any shape or form

We need a way for MCUs to have "I2C capabilities" that they can share with the application and the test suite, and there should be capabilities for single-byte 0 length transfers and transactional 0 length transfers.

JohnK1987, 2023-07-24:
What is it good for other than finding an address? From my experiences one dummy byte is working solution.
```
error = i2c.write(address<<1,0x00,1);
```

multiplemonomials, 2023-07-25:
Yeah but, sometimes (though rarely) writing that byte might change the behavior of the device.