mbed-ce/mbed-os #178: Flash after build on custom board

JessevGool, 2023-08-21:
I added a custom board to the custom_targets.json and gave it a custom upload config in upload_method_cfg.
I can see that the config gets detected, however the code doesn't get flashed when I press build.

custom_targets.json
```json
{
  "STM32F407ZET": {
    "inherits": ["MCU_STM32F407xE"],
    "device_name": "STM32F407ZET"
  }
}

```
cmake-variants.yml:
```cmake
board: 
  default: STM32F407ZET
  choices:
    STM32F407ZET:
      short: STM32F407ZET
      settings:
        MBED_TARGET: STM32F407ZET
        UPLOAD_METHOD: STLINK
```
The upload_method_cfg is basically a copy of another F4 board with some names changed to the correct board
What would I need to add to make it flash automatically?

JohnK1987, 2023-08-21:
Hello and welcome,

have you tried it as described [here](https://github.com/mbed-ce/mbed-os/wiki/Project-Setup:-VS-Code#building-and-flashing-code)? It means flash-xxxx (xxxx comes from top lvl cmake) is selected as current target.

Or what error do you see?

BR, Jan

JessevGool, 2023-08-21:
Hey @JohnK1987 I completely forgot about that,  I'll try that tomorrow when I have access to the hardware again

JessevGool, 2023-08-22:
That did it, thank you

closed 2023-08-22 by JessevGool

JohnK1987, 2023-08-22:
You are welcome and good luck with your project.
Feel free to ask anything in [ttps://github.com/mbed-ce/mbed-os/discussions](https://github.com/mbed-ce/mbed-os/discussions).

BR, Jan