mbed-ce/mbed-os #196: STM32U5 DMA SPI not working yet

multiplemonomials, 2023-11-03:
DMA SPI is currently not functional on this device family.  Need to test things out some more...

wdx04, 2024-04-27:
I have used the DMA SPI to drive an ILI9341 240x320 SPI display on a custom STM32U575RI board without a problem.
It takes 35ms to refresh the whole display in RGB565 format, compared to 103ms when DMA is turned off.

wdx04, 2024-05-18:
@multiplemonomials
Hi Jamie,
I found a bug(typo?) in stm_dma_utils.c, line 691 and line 709, when data alignment is 4, DestDataWidth should be set to  DMA_DEST_DATAWIDTH_WORD instead of DMA_DEST_DATAWIDTH_BYTE:

https://github.com/mbed-ce/mbed-os/blob/87bc6de65a3943d2f1cdba22da70af6b592e2ec7/targets/TARGET_STM/stm_dma_utils.c#L689

This bug only affects STM32U5 with 32bit data transfer.

wdx04, 2024-05-18:
Also, I noticed the meaning of "Length" parameter of some DMA APIs (e.g. HAL_TIM_PWM_Start_DMA) has been changed in STM32U5. In STM32 series before STM32U5, "Length" means the number of data items to be processed, but in STM32U5, "Length" means the number of data bytes (number of data items * data width in bytes) to be processed.

multiplemonomials, 2025-01-18:
This should be fixed now as of the PR above and others, I believe.

closed 2025-01-18 by multiplemonomials