mbed-ce/mbed-os #200: Issue with events under STM32L432KC

JohnK1987, 2023-12-11:
Hello, I was asked for some help with call I2C::transfer() from Interrupt. Of course I recomanded EventQueue (later just with bool flag).
However the person told me he has some issues with that. So he shared very basic example and i am facing same issue that i do not understand and I think it should work.

Code
```
#include "mbed.h"

DigitalOut led(LED1);
InterruptIn btn(PB_1);

// create an event queue
EventQueue queue(32 * EVENTS_EVENT_SIZE);

void do_something() {
  // this now runs in the context of eventThread, instead of in the ISR
  led = !led;
  printf("Toggle LED!\r\n");
}

int main() {
     printf("Start!\r\n");
  // create a thread that'll run the event queue's dispatch function
  Thread eventThread;
  eventThread.start(callback(&queue, &EventQueue::dispatch_forever));

  btn.mode(PullUp);
  // wrap calls in queue.event to automatically defer to the queue's thread
  btn.fall(queue.event(&do_something));

  while (1) {
    ThisThread::sleep_for(1s);
    // printf("Wait!\r\n");
  } 
}
```
mbed_app
```
{
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 115200,
            "platform.stdio-buffered-serial": 1,
            "target.c_lib": "std",
            "target.printf_lib": "std",
            "platform.callback-nontrivial": true

        }
    }
}
```

Console output>
```
Start!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Togg

++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: id
Location: 0x8005283
File: event.h+207
Error Value: 0x0
Current Thread: application_unnamed_thread Id: 0x20002C10 Entry: 0x8006151 StackSize: 0x1000 StackMem: 0x20003A20 SP: 0x2000FEDC 
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=NUCLEO_L432KC
-- MbedOS Error Info --
```
Also very strange is when you comented out the printf from called evet then the crash report is changed to different one. 

```
++ MbedOS Error Info ++
Error Status: 0x80020126 Code: 294 Module: 2
Error Message: CMSIS-RTOS error: ISR Queue overflow
Location: 0x8007365
Error Value: 0x2
 Current Thread: application_unnamed_thread Id: 0x200013CC Entry: 0x8003B01 StackSize: 0x1000 StackMem: 0x20002960 SP: 0x2000FE84 
For more info, visit: https://mbed.com/s/error?error=0x80020126&tgt=NUCLEO_L432KC
-- MbedOS Error Info --
```
Same result with ARM MbedOS 6.17 and 5.15. I understand the L432KC do not have so much RAM but anyway...
@multiplemonomials @JojoS62 any ideas?

BR, Jan

JojoS62, 2023-12-11:
The linker script looks wrong: the L432 has 64 kB  RAM, but not consecutive. The linker script is assuming that:
https://github.com/mbed-ce/mbed-os/blob/d9676cccca35cfba5ed60c1661cdde5998827876/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L432xC/TOOLCHAIN_GCC_ARM/stm32l432xc.ld#L41

Removing +RAM1_Size should fix it. The other part of Ram could be used by sections in code and custom linker script.

hope it helps.

JohnK1987, 2023-12-11:
I saw that too and I already tested cut out RAM1 from the calculation but still same.

JojoS62, 2023-12-11:
Ok, the datasheet says the SRAM2 is also mapped to 0x2000 C000, so it is 64 k continuous. 
difficult, I don’t have this MCU.

multiplemonomials, 2023-12-13:
I tested it on my L452RE board
```
Start!
Start!
Start!
Start!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
Toggle LED!
```
No issues...

multiplemonomials, 2023-12-13:
I wonder, could this be a debouncing related issue?  If we're seeing errors like "ISR queue overflow", could that be related to electrical noise making the ISR trigger many times in succession?

JohnK1987, 2023-12-13:
You hit the nail. I connected a rotary encoder with implemented debouncing circuit and it is working like a charm. Only when I rotate very very fast in both direction then the crash comes.

closed 2023-12-13 by JohnK1987