mbed-ce/mbed-os #207: Error trying to build for Teeny 4 boards

KurtE, 2024-01-13:
Since Paul reminded me yesterday about this port of the MBED to support Teensy 4 boards, I thought I would give it a try again.
I am currently trying to build this on Windows 11.  I tried it earlier which failed, so I deleted the hello world git clone and created a fresh copy this morning. 

More details up on the PJRC Forum thread:
https://forum.pjrc.com/index.php?threads/testers-needed-mbed-os-port-for-teensy-4-0.72107/page-5#post-336802

This time it made it through the CMake, but ninja failed with an error:
```
D:\github\mbed-ce-hello-world\build>ninja
[1/251] Generating ../mbed-teensy-40.link_script.ld
Preprocess linker script: MIMXRT1052xxxxx.ld -> mbed-teensy-40.link_script.ld
[8/251] Building C object mbed-os/CMakeFiles/mbed-os.dir/t...sso_MCUS/TARGET_MIMXRT105x/TARGET_TEENSY_4X/bootdata.c.obj
FAILED: mbed-os/CMakeFiles/mbed-os.dir/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MIMXRT105x/TARGET_TEENSY_4X/bootdata.c.obj
D:\ccache-4.7.4-windows-i686\ccache.exe "D:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\13.2 Rel1\bin\arm-none-eabi-gcc.exe" @mbed-os\CMakeFiles\mbed-os.dir\targets\TARGET_NXP\TARGET_MCUXpresso_MCUS\TARGET_MIMXRT105x\TARGET_TEENSY_4X\bootdata.c.obj.rsp -MD -MT mbed-os/CMakeFiles/mbed-os.dir/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MIMXRT105x/TARGET_TEENSY_4X/bootdata.c.obj -MF mbed-os\CMakeFiles\mbed-os.dir\targets\TARGET_NXP\TARGET_MCUXpresso_MCUS\TARGET_MIMXRT105x\TARGET_TEENSY_4X\bootdata.c.obj.d -o mbed-os/CMakeFiles/mbed-os.dir/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MIMXRT105x/TARGET_TEENSY_4X/bootdata.c.obj -c D:/github/mbed-ce-hello-world/mbed-os/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MIMXRT105x/TARGET_TEENSY_4X/bootdata.c
D:/github/mbed-ce-hello-world/mbed-os/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MIMXRT105x/TARGET_TEENSY_4X/bootdata.c:43:10: fatal error: mimxrt_flash_info.h: No such file or directory
   43 | #include <mimxrt_flash_info.h>
      |          ^~~~~~~~~~~~~~~~~~~~~

```

It appears like that line was changed or added 4 months ago as per the blame listing.

[image]

Wondering if maybe that include should be: mimxrt_memory_info.h
Which does exist in the TEENSY_40 directory as well as the TEENSY_41

Thanks

Kurt

KurtE, 2024-01-13:
Quick update: I edited the file to change tine include file and now NINJA completes.
```
D:\github\mbed-ce-hello-world\build>ninja
[223/223] Linking CXX executable HelloWorld.elf
D:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/13.2 Rel1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: HelloWorld.elf has a LOAD segment with RWX permissions
-- built: D:/github/mbed-ce-hello-world/build/HelloWorld.bin
-- built: D:/github/mbed-ce-hello-world/build/HelloWorld.hex
| Module                    |         .text |       .data |        .bss |
|---------------------------|---------------|-------------|-------------|
| CMakeFiles\HelloWorld.dir |       50(+50) |       0(+0) |       0(+0) |
| [fill]                    |     148(+148) |       8(+8) |     48(+48) |
| [lib]\c.a                 | 16586(+16586) | 1708(+1708) |   373(+373) |
| [lib]\gcc.a               |     752(+752) |       0(+0) |       0(+0) |
| [lib]\mbed-usb.a          | 10302(+10302) |       0(+0) |       2(+2) |
| [lib]\misc                |     264(+264) |       4(+4) |     25(+25) |
| [misc]                    |         0(+0) |       0(+0) |       0(+0) |
| mbed-os\CMakeFiles        | 46181(+46181) |   440(+440) | 8408(+8408) |
| Subtotals                 | 74283(+74283) | 2160(+2160) | 8856(+8856) |
Total Static RAM memory (data + bss): 11016(+11016) bytes
Total Flash memory (text + data): 76443(+76443) bytes
```

JohnK1987, 2024-01-15:
Hello, if your are sure you can make a PR.
I believe @multiplemonomials will check this.

BR, Jan

multiplemonomials, 2024-01-15:
Ah yes I refactored the MIMXRT105x flash code as it was quite badly hardcoded to support MIMXRT1050 EVK only -- now it supports MIMXRT1060 EVK as well and could be updated to work on Teensy as well!  Would just need to update [this header](https://github.com/mbed-ce/mbed-os/blob/master/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MIMXRT105x/TARGET_EVK/flash_defines.h) with the correct flash sequences.  But yeah, looks like I missed updating that include, whoops! 

Would be great if you could make a PR with this fix!

closed 2024-04-27 by multiplemonomials