mbed-ce/mbed-os #212: STM32_gen_peripheralPins.py chrashing after generating pinmaps

lefebvresam, 2024-01-18:
### Description of defect

When generating pinmap files the script is crashing:

```
mbed-os$ python targets/TARGET_STM/tools/STM32_gen_PeripheralPins.py -m "STM32U575RGTx.xml"

Script version 1.20.5

Checking STM32_open_pin_data repo...
        Already up to date.

STM32_open_pin_data DB version STM32CubeMX-DB.6.0.90

 * Output directory: /home/samlef/git/hmipanel/mbed-os/TARGET_CUSTOM/TARGET_STM/TARGET_STM32U5/TARGET_STM32U575xG/TARGET_STM32U575RGT
 * Generating PeripheralPins.c and PinNames.h with 'STM32_open_pin_data/mcu/STM32U575RGTx.xml'
 * GPIO file: STM32_open_pin_data/mcu/IP/GPIO-STM32U5x_gpio_v1_0_Modes.xml
Traceback (most recent call last):
  File "targets/TARGET_STM/tools/STM32_gen_PeripheralPins.py", line 1933, in <module>
    parse_pins()
  File "targets/TARGET_STM/tools/STM32_gen_PeripheralPins.py", line 1399, in parse_pins
    if "Variant" in s.attributes:
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 522, in __getitem__
    return self._attrs[attname_or_tuple]
KeyError: 0
```

#### Target(s) affected by this defect ?

Custom target

#### Toolchain(s) (name and version) displaying this defect ?

NA

#### What version of Mbed-os are you using (tag or sha) ?

Mbed 7.59.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

CLI2

#### How is this defect reproduced ? 

Execute command above

lefebvresam, 2024-01-18:
I could use it succesfully on another Linux laptop.

JohnK1987, 2024-01-18:
Can not be this caused by python version?
I tried that on my windows 10 machine and seems to be OK.
```
D:\Mbed\MbedCE\MbedCE_STM32H5xxxx\mbed-os\targets\TARGET_STM\tools>STM32_gen_PeripheralPins.py -m "STM32U575RGTx.xml"

Script version 1.20.5

Checking STM32_open_pin_data repo...
*** git clone https://github.com/STMicroelectronics/STM32_open_pin_data.git ***
*** git clone done

STM32_open_pin_data DB version STM32CubeMX-DB.6.0.90

 * Output directory: D:\Mbed\MbedCE\MbedCE_STM32H5xxxx\mbed-os\targets\TARGET_STM\tools\TARGET_CUSTOM\TARGET_STM\TARGET_STM32U5\TARGET_STM32U575xG\TARGET_STM32U575RGT
 * Generating PeripheralPins.c and PinNames.h with 'STM32_open_pin_data\mcu\STM32U575RGTx.xml'
 * GPIO file: STM32_open_pin_data\mcu\IP\GPIO-STM32U5x_gpio_v1_0_Modes.xml
 * I/O pins found: 120 connected: 1

D:\Mbed\MbedCE\MbedCE_STM32H5xxxx\mbed-os\targets\TARGET_STM\tools>
```

lefebvresam, 2024-01-18:
Difficult to reproduce, because it was one in WSL2 on a Windows bootcamp, but after a fatal crash with a non bootcamp supported tool in Windows I had to reinstall my whole partition and lost half of a week and a lot of docs on my desktop. But however, maybe it's an idea to catch that error in the code and create a message for it.

multiplemonomials, 2024-01-19:
I think I'd like to close this one for now unless we can find a way to reproduce the issue.  Any objections?

JohnK1987, 2024-01-19:
`"/usr/lib/python2.7/xml/dom/minidom.py"`
I believe it is caused by using python 2.7 instead of 3+

multiplemonomials, 2024-01-19:
ah yeah, that will do it

lefebvresam, 2024-01-19:
Maybe in the script doing a verifiaction for Python version before continuing?

multiplemonomials, 2024-01-19:
Could you make a PR for that?

lefebvresam, 2024-01-19:
Let me first bringup my new environment.

JohnK1987, 2024-01-19:
So we can add something like this at the beginning of the script maybe?

```
import sys
if sys.version_info[0] < 3:
    raise Exception("Must be using Python 3")
```

I can test it and make the PR later. You can focus other things.

JohnK1987, 2024-01-19:
Ok, we have an info about Python version requirement directly in the .py script now.

closed 2024-01-19 by JohnK1987