mbed-ce/mbed-os #214: Putting sleep_for in a loop with a printf not ending with \n will not execute that printf

lefebvresam, 2024-01-19:
Purpose is to write the time and overwrite it at the same line:

```
    while(1) {
        time_t seconds = time(NULL);
        char buffer[32];
        strftime(buffer, 32, "%I:%M:%S %p", localtime(&seconds));
        printf("%s             \r", buffer);
        int i = 1000000;
        while(i--); // keep busy (sleep gives programming issues and printf cannot flush)
        ThisThread::sleep_for(5000);
    }
```

But when the ThisThread::sleep_for(5000); is inserted, the printf will never be executed.
If you change \r to \n it works, but then you get a new line on each printf.

multiplemonomials, 2024-01-19:
Hmm, this has got to be something related to buffering, but I am unsure whether it's in the C library or in BufferedSerial.  

https://os.mbed.com/forum/bugs-suggestions/topic/1849/

^ This suggests it's in the C library, it's an old thread but it seems to still be valid.  You might try `fflush(stdout)` and see if that prints the missing data.

lefebvresam, 2024-01-19:
Because of the change to other compiler I cannot compile my original project anymore with mbed arm.

```
sam@elitebook-845:~/git/hmc20_test$ sh build.sh -f
Command: mbed-tools compile -b develop -m HMC20 -t GCC_ARM
Configuring project and generating build system...
-- Configuring done (0.3s)
-- Generating done (0.2s)
-- Build files have been written to: /home/sam/git/hmc20_test/cmake_build/HMC20/develop/GCC_ARM
Building Mbed project...
[1/3] Linking CXX static library _deps/greentea-client-build/libclient.a
FAILED: _deps/greentea-client-build/libclient.a 
: && /usr/bin/cmake -E rm -f _deps/greentea-client-build/libclient.a && /usr/bin/arm-none-eabi-ar qc _deps/greentea-client-build/libclient.a  _deps/greentea-client-build/CMakeFiles/client.dir/source/greentea_test_env.cpp.obj _deps/greentea-client-build/CMakeFiles/client.dir/source/greentea_test_io.c.obj && /usr/bin/arm-none-eabi-ranlib _deps/greentea-client-build/libclient.a && :
/bin/sh: 1: /usr/bin/arm-none-eabi-ar: not found
[2/3] Linking CXX static library _deps/greentea-client-build/libclient_userio.a
FAILED: _deps/greentea-client-build/libclient_userio.a 
: && /usr/bin/cmake -E rm -f _deps/greentea-client-build/libclient_userio.a && /usr/bin/arm-none-eabi-ar qc _deps/greentea-client-build/libclient_userio.a  _deps/greentea-client-build/CMakeFiles/client_userio.dir/source/greentea_test_env.cpp.obj && /usr/bin/arm-none-eabi-ranlib _deps/greentea-client-build/libclient_userio.a && :
/bin/sh: 1: /usr/bin/arm-none-eabi-ar: not found
[3/3] Linking CXX executable hmc20_test.elf
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/libc.a(libc_a-signalr.o): in function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): warning: _getpid is not implemented and will always fail
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/libc.a(libc_a-signalr.o): in function `_kill_r':
signalr.c:(.text._kill_r+0x12): warning: _kill is not implemented and will always fail
-- built: /home/sam/git/hmc20_test/cmake_build/HMC20/develop/GCC_ARM/hmc20_test.bin
-- built: /home/sam/git/hmc20_test/cmake_build/HMC20/develop/GCC_ARM/hmc20_test.hex
```

lefebvresam, 2024-01-19:
Tested in new environment with latest ce version of mbed, this is working but I have to do flush.

```
	while(true) {
        time_t seconds = time(NULL);
        char buffer[32];
        strftime(buffer, 32, "%I:%M:%S %p", localtime(&seconds));
        printf("%s\r", buffer);
        fflush(stdout);
        ThisThread::sleep_for(1000);
	}
```

multiplemonomials, 2024-01-30:
Are we good to close this?

lefebvresam, 2024-01-30:
I don't think this is fully solved. Why do you need to flush the buffer before sleep and why this is not done automatically as a safety measure?

multiplemonomials, 2024-01-30:
So if you don't disable sleep, the code sample you posted doesn't work?  Yes, that does seem suspiciously like a bug.

lefebvresam, 2024-01-31:
Exactly