mbed-ce/mbed-os #215: https://github.com/mbed-ce/mbed-os/wiki/Toolchain-Setup-Guide

lefebvresam, 2024-01-19:
I also had to install

```
sudo apt-get install pyhton3-dev
pip3 install PrettyTable
pip3 install future
pip3 install intelhex
pip3 install markupsafe==2.0.1
pip3 install click==7.1
pip3 install --upgrade six>=1.13.0

# install latest version of cmake
sudo apt remove --purge --auto-remove cmake
wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | sudo tee /usr/share/keyrings/kitware-archive-keyring.gpg >/dev/null
echo 'deb [signed-by=/usr/share/keyrings/kitware-archive-keyring.gpg] https://apt.kitware.com/ubuntu/ jammy main' | sudo tee /etc/apt/sources.list.d/kitware.list >/dev/null
sudo apt-get update
sudo apt-get install kitware-archive-keyring
sudo apt-get install cmake
cmake --version

Ensure key stays up to date (changes every year):
sudo rm /usr/share/keyrings/kitware-archive-keyring.gpg : remove the current key
sudo apt-get remove --purge --auto-remove kitware-archive-keyring : ensure there is no previous install 
sudo apt-get install kitware-archive-keyring : this overwrites the key, and when apt-get upgrade rotates key when there is an update
```

And conversion of old libs from Mercurial to Git:

```bash
pip install mercurial
cd ~/git
git clone https://github.com/frej/fast-export.git
cd ~
hg clone https://os.mbed.com/users/WiredHome/code/RA8875/ ~/git/RA8875
cd ~/git/RA8875
hg log | grep user: | sort | uniq | sed 's/user: *//' > ../authors : create authors file
sudo nano ~/git/authors : repare the file in the form "SamLefebvre" = "Sam Lefebvre <sam.lefebvre@saleconix.be>"
mkdir ~/git/RA8875_exp
cd ~/git/RA8875_exp
git init .
~/git/fast-export/hg-fast-export.sh -r ~/git/RA8875 -A ~/git/authors
# See (1) if you have unnamed heads
git checkout
# Do first now: create new repo on bitbucket named RA8875
git remote add origin https://SamLefebvre@bitbucket.org/saleconix/ra8875.git
git push origin --all

# (1) Unnamed heads:
# Git can only have one head, hg can have multiple heads in one branch.
# We will convert the unnamed heads into named branches.
# You can also ignore the head with option --ignore-unnamed-heads but you end up with commits being lost
hg update -r 174
hg branch r174-fix-git-conversion
hg commit -m "Fix git conversion (unnamed head r174)"
hg update default
# repeat for each unnamed head
```

multiplemonomials, 2024-01-19:
ooh this is a good guide, I have just been copying the mercurial repos and losing all their history.  I've been thinking, maybe we need a `mbed-ce-converted-libraries` github organization for old Mbed libraries that have been converted to run with Mbed CE...

Regarding the python dependencies, you *shouldn't* have to install those manually, the mbed requirements.txt should take care of that.  Mainline Mbed has a problem with requiring very specific versions in requirements.txt that are not achieveable on some systems/python versions, so for now I've removed all of the version requirements from requirements.txt.  If and when someone reports a problem with an old or new version of a python package, I'll add them back.

lefebvresam, 2024-01-19:
That's a good idea. I now have the RA8875 lib in a private git repo with all my changes from 2020 merged. The purpose is to make it public again so that the original authors can also come back to all my additions.

JohnK1987, 2024-01-19:
> ooh this is a good guide, I have just been copying the mercurial repos and losing all their history. I've been thinking, maybe we need a `mbed-ce-converted-libraries` github organization for old Mbed libraries that have been converted to run with Mbed CE...

The word **converted** is not necessary I think. I look forward...

BTW Github has good tool for import project from different platforms. It is not 100% working but i was able to migrate whole project from Mbed Mercurial repo into my Github repo also with history of commits.
[image]

lefebvresam, 2024-01-19:
Since long time I was using Bitbucket instead of Github. Long time ago there was a story about public/private reppo's but seems not to be true anymore. I thing Bitbucket was a bit more intuitive and had better graphics. So in my search how to streamline my software in a more uniform way I came out on fast-export. If there are better ways I'm open for it. But like always ... time is money ;-)

JohnK1987, 2024-01-19:
This is not link just a picture. You have to go for create new repo and on that page you will see the link.
Like here https://github.com/organizations/mbed-ce/repositories/new

lefebvresam, 2024-01-20:
> Support for importing Mercurial, Subversion and Team Foundation Version Control (TFVC) repositories will end on April 12, 2024. For more details, see the [changelog](https://gh.io/github-importer-non-git-eol).

We need to be fast.

multiplemonomials, 2024-03-18:
btw, I have updated Mbed to now automatically install the Python dependencies.  So I think we can probably close this now?

lefebvresam, 2024-03-18:
If documentation is fine ok for me.

closed 2024-03-19 by multiplemonomials