mbed-ce/mbed-os #221: Unknown CMake command

lefebvresam, 2024-01-21:
During build I get:

```
sam@elitebook-845:~/git/hmipanel$ sh build.sh 
Build the code for target HMC20 with profile develop
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Checking for Python package intelhex -- found
-- Checking for Python package prettytable -- found
-- Checking for Python package future -- found
-- Checking for Python package jinja2 -- found
-- Mbed: First CMake run detected, generating configs...
mbed_config.cmake has been generated and written to '/home/sam/git/hmipanel/build/develop/mbed_config.cmake'
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/gcc-arm/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/gcc-arm/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/gcc-arm/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Mbed: Target does not have any upload method configuration.  'make flash-' commands will not be available unless configured by the upper-level project.
CMake Error at mbed-os/tools/cmake/mbed_target_functions.cmake:169 (mbed_generate_upload_target):
  Unknown CMake command "mbed_generate_upload_target".
Call Stack (most recent call first):
  source/CMakeLists.txt:35 (mbed_set_post_build)
```

JohnK1987, 2024-01-21:
> Target does not have any upload method configuration.

you need to create your own settings similar to [this one ](https://github.com/mbed-ce/mbed-os/blob/master/targets/upload_method_cfg/NUCLEO_U575ZI_Q.cmake)

But in context of custom target si more logical to place it directly to top level CmakeLists of your project like is written [here](https://github.com/mbed-ce/mbed-os/wiki/Upload-Methods#what-is-an-upload-method)

Then the Cmake file can looks like below and how you can see it contains same lines (Upload method settings) what is used in the cmake config above.

```
cmake_minimum_required(VERSION 3.19)
cmake_policy(VERSION 3.19)

# Initialize Mbed OS build system. 
# Note: This block must be before the include of app.cmake
set(MBED_APP_JSON_PATH mbed_app.json5)
set(CUSTOM_TARGETS_JSON_PATH custom_targets.json5) # This activates the custom targets in custom_targets.json5

# Upload method settings
set(UPLOAD_METHOD "STM32CUBE") 
set(STM32CUBE_UPLOAD_ENABLED TRUE)
set(STM32CUBE_CONNECT_COMMAND -c port=SWD reset=HWrst)
set(STM32CUBE_GDBSERVER_ARGS --swd)

include(mbed-os/tools/cmake/app.cmake)

# Include folder with the custom targets.
# This needs to be done before add_subdirectory(mbed-os).
add_subdirectory(custom_targets) 
add_subdirectory(mbed-os)

project(MbedCE_test)

add_subdirectory(app)

mbed_finalize_build()
```

lefebvresam, 2024-01-21:
The purpose is to keep code in a separate directory 'source'. So my main is in that folder. To let cmake work I splitup the head CMakeLists.txt file into two parts. The first part is in the top dir, and the second parts is in the source dir.

First part in top dir (must see the submodules):
```
cmake_minimum_required(VERSION 3.19)
cmake_policy(VERSION 3.19)

# Initialize Mbed OS build system. 
set(MBED_APP_JSON_PATH mbed_app.json5)
set(APP_TARGET hmc20)

include(mbed-os/tools/cmake/app.cmake)
project(${APP_TARGET})

add_subdirectory(HMC20)
add_subdirectory(source)
add_subdirectory(RA8875)
add_subdirectory(mbed-os)
```

Second part in source dir:
```
set(APP_SOURCE_PATH ${CMAKE_CURRENT_SOURCE_DIR} CACHE INTERNAL "")

add_subdirectory(dmaflash)
add_subdirectory(widgets)
add_subdirectory(interpreter)
add_subdirectory(demoapps)
add_subdirectory(commandinterface)

add_executable(${APP_TARGET}
    "${CMAKE_CURRENT_SOURCE_DIR}/main.cpp"
)

target_sources(${APP_TARGET} PRIVATE
    "${CMAKE_CURRENT_SOURCE_DIR}/display.cpp"
)

target_include_directories(${APP_TARGET} PRIVATE 
    "${APP_SOURCE_PATH}/fontrom"
)

target_link_libraries(${APP_TARGET} PRIVATE
    mbed-os
    mbed-events
    RA8875
    dmaflash
    widgets
    interpreter
    apprunlight
    appbutton
    commandinterface
)

mbed_set_post_build(${APP_TARGET})
mbed_finalize_build()

```

JohnK1987, 2024-01-21:
Ok, so try to place it between target settings and app.cmake
```
set(APP_TARGET hmc20)

# Upload method settings
set(UPLOAD_METHOD "STM32CUBE") 
set(STM32CUBE_UPLOAD_ENABLED TRUE)
set(STM32CUBE_CONNECT_COMMAND -c port=SWD reset=HWrst)
set(STM32CUBE_GDBSERVER_ARGS --swd)

include(mbed-os/tools/cmake/app.cmake)
```

lefebvresam, 2024-01-21:
So you don't need to mbed_set_post_build(hmc20) in that case?
But how the rest is done then? (creating bin etc.)
Ans why I don't get this error when everything is in the top file like in my test:

```
cmake_minimum_required(VERSION 3.19)
cmake_policy(VERSION 3.19)

# Initialize Mbed OS build system. 
# Note: This block must be before the include of app.cmake
set(MBED_APP_JSON_PATH mbed_app.json5)

include(mbed-os/tools/cmake/app.cmake)
add_subdirectory(HMC20)
add_subdirectory(mbed-os)

project(hmc20)

# add subdirectories and build targets here
add_executable(hmc20 main.cpp)
target_link_libraries(hmc20 mbed-os)
mbed_set_post_build(hmc20)

mbed_finalize_build()
```

JohnK1987, 2024-01-21:
The code above was just an example about how to solve upload methods because i saw that in your log above
You probably need `mbed_set_post_build(${APP_TARGET})`.

lefebvresam, 2024-01-21:
If It put it in the first CMakeLists.txt I get:

```
sam@elitebook-845:~/git/hmipanel$ sh build.sh 
Generate build folder for target 'HMC20' with profile 'develop'
-- Mbed: Target does not have any upload method configuration.  'make flash-' commands will not be available unless configured by the upper-level project.
-- Mbed: Not building any Mbed OS tests.
-- Found stlink: /usr/local/bin/st-util (found version "1.7.0") 
-- Mbed: Code upload and debugging enabled via upload method STLINK
-- Mbed: No IDE detected, will generate configurations for command-line debugging (e.g. ninja gdbserver, then ninja debug-SomeProgram)
CMake Error at mbed-os/tools/cmake/mbed_target_functions.cmake:30 (add_custom_command):
  TARGET 'hmc20' was not created in this directory.
Call Stack (most recent call first):
  mbed-os/tools/cmake/mbed_target_functions.cmake:159 (mbed_generate_bin_hex)
  CMakeLists.txt:29 (mbed_set_post_build)

CMake Error at mbed-os/tools/cmake/mbed_target_functions.cmake:52 (add_custom_command):
  TARGET 'hmc20' was not created in this directory.
Call Stack (most recent call first):
  mbed-os/tools/cmake/mbed_target_functions.cmake:166 (mbed_generate_map_file)
  CMakeLists.txt:29 (mbed_set_post_build)
```

lefebvresam, 2024-01-22:
The point is that you must keep mbed_set_post_build in the same CMakeLists.txt file. To enable a splitup this is working in my testproject:

CMakeLists.txt in top folder:

```
cmake_minimum_required(VERSION 3.19)
cmake_policy(VERSION 3.19)

# Initialize Mbed OS build system. 
# Note: This block must be before the include of app.cmake
set(MBED_APP_JSON_PATH mbed_app.json5)
set(APP_TARGET hmc20)
include(mbed-os/tools/cmake/app.cmake)

project(${APP_TARGET})

# Upload method settings
set(UPLOAD_METHOD "STLINK") 
set(STLINK_UPLOAD_ENABLED TRUE)
set(STLINK_LOAD_ADDRESS 0x8000000)
set(STLINK_ARGS --freq=4M --reset)

add_subdirectory(HMC20)
add_subdirectory(mbed-os)
add_subdirectory(source)

mbed_finalize_build()
```

\source\CMakeLists.txt  in source folder where main is residing;

```

# add subdirectories and build targets here
add_executable(${APP_TARGET} 
    main.cpp
)

target_link_libraries(${APP_TARGET} 
    mbed-os
)

mbed_set_post_build(${APP_TARGET})
```
I'm still wondering if you can create your own file for upload method configuration instead of putting the config in the top CMakeLists.txt file and where to put this file. I tried to put it in the private target folder /HMC20 but it is not seen there.

lefebvresam, 2024-01-22:
set(CUSTOM_TARGETS_JSON_PATH custom_targets.json5) -> I don't think this is mandatory. If you ommit it still works... Why?

lefebvresam, 2024-01-22:
Even this is not necessary:

`set(MBED_APP_JSON_PATH mbed_app.json5)`

JohnK1987, 2024-01-30:
Do you have mbed_app.json5 in your project and use it? If not, then you are ignoring, or solved by different way, Custom target structure and then it is not necessary to have these commands.

multiplemonomials, 2024-01-30:
hmm... the target JSON paths are only used on the first CMake run I believe.  Maybe it's working because the paths are still stored in the CMake cache, but it would break if you deleted cache and reconfigured.

lefebvresam, 2024-01-30:
No there would be something as a 'default' location. Both lines are not in my CmakeLists and and I use both json5 files. I already deleted a lot the build folder during compilation, that's the reason I provided a -d option in my own build script and it is still working.

JohnK1987, 2024-07-18:
The project was not created according to MbedCE instructions, so it is hard to remotely debug.

closed 2024-07-18 by JohnK1987

lefebvresam, 2024-07-19:
Ok layer on, when I provide some open source software I will re-use this structure and make it public for testing.