mbed-ce/mbed-os #223: cannot find -lmbed-events: No such file or directory

lefebvresam, 2024-01-21:
At the end of the linking:

```
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: cannot find -lmbed-events: No such file or directory
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: cannot find -lmbed-events: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Compiling failed
```

lefebvresam, 2024-01-21:
If I remove mbed-events in Target_link_libraries I get the same message but once instead of twice.

```
target_link_libraries(${APP_TARGET} PRIVATE
    mbed-os
    #mbed-events   
    RA8875
    dmaflash
    widgets
    interpreter
    apprunlight
    appbutton
    commandinterface
)
```

```
ibmbed-storage-sd.a  mbed-os/storage/blockdevice/COMPONENT_SPIF/libmbed-storage-spif.a  mbed-os/storage/blockdevice/libmbed-storage-blockdevice.a  mbed-os/connectivity/mbedtls/libmbed-mbedtls.a  -lmbed-events && :
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: cannot find -lmbed-events: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Building failed
```

lefebvresam, 2024-01-21:
At several places in the code, I use 
`EventQueue* _queue;
`
To make this possible, I have to add
```

target_link_libraries(${APP_TARGET} PRIVATE
    mbed-os
    mbed-events
..
)
```
In the CMakeLists.txt file.
And this is not recognized anymore.

lefebvresam, 2024-01-21:
If I ommit mbed-events in my CMakeLists.txt files, I get:

```
...
wcrtomb                                           /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/libc.a(libc_a-wcrtomb.o)
wcsnrtombs                                        /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/libc.a(libc_a-wcsnrtombs.o)
wcsrtombs                                         /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/libc.a(libc_a-wcsrtombs.o)
write                                             mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_retarget.cpp.obj
                                                  mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_board.c.obj
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Building failed
```

lefebvresam, 2024-01-22:
It looks like this is not necessary anymore:

```
target_link_libraries(${APP_TARGET} PRIVATE
    mbed-events
```

JohnK1987, 2024-01-22:
I tested few sec ago and yes. The `mbed-events` is part of `mbed-os` and will be necessary only with `mbed-baremetal` not with `mbed-os`.

lefebvresam, 2024-01-22:
So in fact if you don't use the baremetal everything is already linked with mbed and you don't need to link other modules again? Is there any overview of all mbed modules that can be linked and when to link them?

JohnK1987, 2024-01-22:
That is very good question but I do not have exact answer for you :D The .json configuration was more clear for me.
This case is probably because other parts of Mbed OS also need `mbed-events` for their functionality. So `mbed-events` have to be already included or part of `mbed-os` (if i remember same state was in original Mbed with pure .json system). 
On another hand, for example `mbed-usb` have to be added manually when you need it and it is not part of `mbed-os`.

I create a wiki page where we can place useful info about [MbedOS configuration](https://github.com/mbed-ce/mbed-os/wiki/MbedOS-configuration)

lefebvresam, 2024-01-22:
I still find it a weird error. Cannout find should not happen.

lefebvresam, 2024-01-27:
There are other libs that you **have** to link before you can use them, like:

```
target_link_libraries(dmaflash PUBLIC
    mbed-os
    RA8875
    mbed-storage
    mbed-mbedtls
)
```

multiplemonomials, 2024-01-30:
Ah yeah, sorry about that, because of annoying build system reasons I had to make mbed-events part of the main Mbed library.  It is not available as a standalone library in Mbed CE under any configurations.  @JohnK1987 could you update the wiki page if you have a chance?

JohnK1987, 2024-01-30:
@multiplemonomials  as I wrote above I already create the [wiki page](https://github.com/mbed-ce/mbed-os/wiki/MbedOS-configuration). It is not final version but it will be slowly updated.

lefebvresam, 2024-01-30:
Maybe the error 'cannot find' should change in 'does not exists'. This is more clear.

multiplemonomials, 2024-01-31:
I have updated that wiki page to fix the mention of mbed-events and add more libraries (gotta start somewhere!).  I think we're good to close this one now right?

lefebvresam, 2024-01-31:
Ok for me

closed 2024-01-31 by lefebvresam