mbed-ce/mbed-os #224: Left align and minimum size width printf not working

lefebvresam, 2024-01-22:
When you are doing:
```
    char* si = "7"; 
    char* sj = "2547"; 
    printf("-------------\n");  
	printf("%6s ci-test\n", si);
	printf("%6s cj-test\n", sj);
    printf("-------------\n");  
```
You expect
7&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;ci-test
2547&nbsp;&nbsp;cj-test

But you get:
7 ci-test
2547 cj-test

When you are doing:
```
    int di = 7; 
    int dj = 2547; 
    printf("-------------\n");  
	printf("%-6d di-test\n", di);
	printf("%-6d dj-test\n", dj);
    printf("-------------\n"); 
```
You expect
7&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;di-test
2547&nbsp;&nbsp;dj-test

You get:
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;7 di-test
&nbsp;2547 dj-test

lefebvresam, 2024-01-22:
<html>
<body>

flags | description
-- | --
- | Left-justify within the given field width; Right justification is the default (see width sub-specifier).

</body>
</html>

and:

<html>
<body>

width | description
-- | --
(number) | Minimum number of characters to be printed. If the value to be printed is shorter than this number, the result is padded with blank spaces. The value is not truncated even if the result is larger.

</body>
</html>

JohnK1987, 2024-01-22:
Ok, this is because in default Mbed uses its own implementation of  `printf`  and it seems some flags are ignored -https://github.com/mbed-ce/mbed-os/tree/master/platform/source/minimal-printf#readme

lefebvresam, 2024-01-22:
Ok, solved when using "target.printf_lib": "std" in target_overrides.

closed 2024-01-22 by lefebvresam