mbed-ce/mbed-os #227: printf not waiting before leaving function

lefebvresam, 2024-01-28:
I have a function that is called in a loop:

```
void DMAFlash::printiindex(const imginfo_s&  iindex, uint16_t  id) {
    FINE("image index %u {", id);
    FINE(" addr: %llu", iindex.addr);
    FINE(" height: %u", iindex.height);
    FINE(" width: %u", iindex.width);
    // printf("[FINE    %-11s %4d] }\n", MODULE, __LINE__);
    // printf("[FINE    %-11s  241] }\n", MODULE);
    printf("[FINE    DMA_FLASH    241] }\n");
    // FINE("}");
}
```

I tried several posibilities but the last line at the end of the loop is not printed.

```
...
[FINE    DMA_FLASH    239]  height: 272
[FINE    DMA_FLASH    240]  width: 480
[FINE    DMA_FLASH    241] }
[FINE    DMA_FLASH    237] image index 36 {
[FINE    DMA_FLASH    238]  addr: 532480
[FINE    DMA_FLASH    239]  height: 252
[FINE    DMA_FLASH    240]  width: 4
[FINE    DMA_FLASH    241] }
[FINE    DMA_FLASH    237] image index 37 {
[FINE    DMA_FLASH    238]  addr: 536576
[FINE    DMA_FLASH    239]  height: 70
[
 [FINE    DMA_FLASH    220] -------------------------

```
It looks like the memory reservation of the string is away when the loop terminates and the printf is still sending data.

lefebvresam, 2024-01-28:
If I put this in the for loop it is solved. It looks like a buffering issue. No idea how to increase the stdio buffer.

```
        if (i==_findex.details.imgindexsize/2) { // buffer overflow when sending too fast
            ThisThread::sleep_for(100);
        }
```

JohnK1987, 2024-01-28:
I think, this is caused because default console use BufferedSerial. Try to set the default console to UnbufferedSerial.
Thy to set this parametr to 0 in your project - https://github.com/mbed-ce/mbed-ce-hello-world/blob/0e92b9b4fa503657820e30b167a378ebb69d20f6/mbed_app.json#L5

lefebvresam, 2024-01-28:
If I move from this:

```
            "platform.stdio-baud-rate": 460800,
            "platform.stdio-buffered-serial": 1,
```

to this:

```
            "platform.stdio-baud-rate": 115200,
            "platform.stdio-buffered-serial": 0,
```

it is also solved. I have to change both the speed and the mode. The idea was that if you use a very high serial speed the buffer is flushed faster if you use higher clockrates, but this is clearly not the case. I'm not so familiar with what is happening with the serial port behind the scene.

lefebvresam, 2024-01-29:
This looks weird because if you do bulk transfer you should be able to use higher speeds. Maybe there is something in the uart clock config?

JojoS62, 2024-02-10:
have you tried a fflush(stdio) after the print?

lefebvresam, 2024-02-12:
I don't think in this case but I remember another case if you are printing and doing a sleep immediately after the printf it will only work if you do a flush first. It means that sleeping blocks the serial transfer while it's still in the buffer. Here there is no sleep, it's a timing/speed issue. I experienced that the faster you send data, the more chance you have that the serial communication is blocking at a certain point with bulk transfer.

lefebvresam, 2024-07-19:
No, but I use the restrictions mentioned above and have no issues anymore, despite the fact that I could potentially use higher serial speeds. To investigate this further I should create a striped down test project for it and dive into the depth of the mbed serial stuff. I'm currently limited in time to do this.

JohnK1987, 2024-07-21:
I made new issue for better tracking. Closed

closed 2024-07-21 by JohnK1987