mbed-ce/mbed-os #229: Custom targets

timIdeaTech, 2024-01-29:
### Discussed in https://github.com/mbed-ce/mbed-os/discussions/112

<div type='discussions-op-text'>

<sup>Originally posted by **JohnK1987** December  4, 2022</sup>
Hello,

what is a Custom target for MbedCE?
For example Bluephill or Blackpill are real custom targets? Technically they are unofficial dev boards what can be bought.
From my point of view the Custom board can be a board what I will made for myself or for my own product. But when I make a dev board based on a popular MCU and will be available in a shop, for example like Blackpill... what is Custom about it? In the context of ARM Mbed it is clear - no official maintainer or Mbed partner for the dev board = it is up to community = Custom board.

@multiplemonomials 

BR, Jan</div>

timIdeaTech, 2024-01-29:
I have a custom design that is based on the B-L072Z-LRWAN1 and using VSCode.

My development setup is as documented on the mbed-ce repo and I am using STMcubeIDE for loading and debugging. Everything configures, compiles and loads properly using an STlinkv3 using (presumably) the cmake-variants.yaml file:

```
buildType:
  default: Develop
  choices:
    Develop:
      short: Develop
      long: Emit debug information but also optimize
      buildType: Develop
    Debug:
      short: Debug
      long: Emit debug information and don't optimize
      buildType: Debug
    Release:
      short: Release
      long: Optimize generated code
      buildType: Release
board:
  default: LATIUM_HVC_ABZ
  choices:
    LATIUM_HVC_ABZ:
      short: LATIUM_HVC_ABZ
      settings:
        MBED_TARGET: LATIUM_HVC_ABZ
        UPLOAD_METHOD: STM32CUBE
        STM32CUBE_UPLOAD_ENABLED: TRUE
        STM32CUBE_CONNECT_COMMAND: [-c, port=swd, reset=HWrst]
        STM32CUBE_GDBSERVER_ARGS: [--swd]

```

My problem is I am not getting a configuration loaded into .vscode/launch.json. 
```
// Auto-generated by Mbed CE.  Edits will be erased when CMake is rerun.
{
    "configurations": [
    ]
}

```
I can add a configuration ({} Cortex Debug: ST-LINK) and change the link to the proper ./build/…/.elf (it doesn’t figure out the alternate dynamic path per the VS Code setup step 6. for .vscode/settings.json file):

```
// Auto-generated by Mbed CE.  Edits will be erased when CMake is rerun.
{
    "configurations": [
        {
            "cwd": "${workspaceFolder}",
            "executable": **"./bin/executable.elf",**
            "name": "Debug with ST-Link",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "showDevDebugOutput": "none",
            "servertype": "stlink"
        }
    ]
}

```
```
{
  "cmake.buildDirectory": "${workspaceFolder}/build/${variant:board}-${variant:buildType}",
...
}
```
 This works to run the debugger, but of course it gets rewritten on rebuilds. I do get a cannot find loader message as described on the mbed-ce github repo discussion for custom targets though.

I’m pretty sure I have followed everything in the custom target example. Is there a step needed that isn’t documented or implemented yet? I’m guessing it might have something to do with the above discussion and creating a custom .cmake file and somewhere getting CMake to read it not from the mbed-os directory. I tried copying and renaming the DISCO_L072CZ_LRWAN1.cmake to LATIUM_HVC_ABZ.cmake and that gets rid of the:

`[cmake] -- Mbed: Target does not have any upload method configuration.  'make flash-' commands will not be available unless configured by the upper-level project.
`

`[cmake] -- Mbed: Loading default upload method configuration from targets/upload_method_cfg/LATIUM_HVC_ABZ.cmake
`
This had no effect.

JohnK1987, 2024-01-29:
Ok, tried to reproduce it and found same thing

This is how looks top lvl CmakeLists.txt in [mbed-ce-hello-world example](https://github.com/mbed-ce/mbed-ce-hello-world/blob/master/CMakeLists.txt)
And this is how looks top lvl CmakeLists.txt in [mbed-ce-custom-targets](https://github.com/mbed-ce/mbed-ce-custom-targets/blob/master/CMakeLists.txt)

but when I fill missing steps from mbed-ce-hello-world then all is OK.

```
cmake_minimum_required(VERSION 3.19)
cmake_policy(VERSION 3.19)

# Initialize Mbed OS build system. 
# Note: This block must be before the include of app.cmake
set(MBED_APP_JSON_PATH mbed_app.json)
set(CUSTOM_TARGETS_JSON_PATH custom_targets.json) # This activates the custom targets in custom_targets.json

include(mbed-os/tools/cmake/app.cmake)

# Include folder with the custom targets.
# This needs to be done before add_subdirectory(mbed-os).
add_subdirectory(custom_targets)

add_subdirectory(mbed-os)

project(mbed-ce-custom-targets)

# add subdirectories and build targets here
add_executable(main main.cpp)

target_link_libraries(main  mbed-os) # Can also link to mbed-baremetal here

mbed_set_post_build(main)

mbed_finalize_build()
```

But i am not so experienced Cmake user so @multiplemonomials has to bring some light to this.

timIdeaTech, 2024-01-30:
I found the problem. The mbed_finalize_build() call somehow ended up not being the last line in the top level CMakeLists.txt file. Weird that this didn't seem to cause any other issues.

closed 2024-01-30 by timIdeaTech