mbed-ce/mbed-os #234: How can you share a pin as output and IRQ input

lefebvresam, 2024-01-31:
I'm configuring a pin as output:

```
m_int = new DigitalOut(irq);

m_int->write(1);
wait_us(2000);
m_reset->write(1);  
m_int->write(0);
wait_us(6000);
```

But after the procedure to setup the IIC address I need the same pin as input for capturing IRQ:

m_irq = new `InterruptIn(irq);`

lefebvresam, 2024-02-02:
You can use the pin to define an output, use it and thereafter re-use the same pin to define an interrupt input like:

  ```
m_reset = new DigitalOut(capreset);

// GT911 Reset cycle 
m_int = new DigitalOut(irq);
m_int->write(0);
wait_us(2000); // >100µs
m_reset->write(1);

// Interrupt
m_irq = new InterruptIn(irq);
m_irq->mode(PullUp);
```

closed 2024-02-02 by lefebvresam

multiplemonomials, 2024-02-02:
Hmm, just a note, I might consider using std::variant<DigitalOut, InterruptIn> for this.  That way you can switch the pin from digital out to digital in, and you can do it without any memory allocations!  Like:
```cpp
std::variant<DigitalOut,InterruptIn> irqPin(std::in_place_type<DigitalOut>(), irq);
std::get<DigitalOut>(irqPin).write(0);
// ...
// switch to InterruptIn
irqPin.emplace(std::in_place_type<InterruptIn>(), irq);
std::get<InterruptIn>(irqPin).mode(PullUp);
```

lefebvresam, 2024-02-06:
I tried this but I'm struggeling with variant:

```
std::variant<DigitalOut,InterruptIn> * irqPin; // in header to have access on other places

irqPin= new std::variant<DigitalOut,InterruptIn>(std::in_place_type<DigitalOut>, irq);
std::get<DigitalOut>(*irqPin).write(0);

irqPin->emplace(std::in_place_type<InterruptIn>, irq);
```

/home/sam/git/hmipanel/ra8875/RA8875.cpp:321:20: error: no matching function for call to 'std::variant<mbed::DigitalOut, mbed::InterruptIn>::emplace(const std::in_place_type_t<mbed::InterruptIn>&, PinName&)'
  321 |     irqPin->emplace(std::in_place_type<InterruptIn>, irq);

lefebvresam, 2024-02-06:
I think in_place_type can only be used in the constructor to select the 'default' configuration, this looks better:

```
    std::variant<DigitalOut,InterruptIn>irqPin(std::in_place_type<DigitalOut>, irq);
    std::get<DigitalOut>(irqPin).write(0);

    irqPin.emplace<InterruptIn>(irq);
    std::get<InterruptIn>(irqPin).mode(PullUp);
```

multiplemonomials, 2024-02-06:
Ah yes sorry, that's the right way to do it!

lefebvresam, 2024-02-06:
And what is now the advantage of this method in comparison with instatiating/deinstatiating objects? Because you also need to do a context switch and you can't share the use in both cases. I can imagine that you create memory leaks if you do m_int = NULL; So not in this case?

multiplemonomials, 2024-02-07:
Two advantages:
- No dynamic memory allocation when switching between the two.  It depends on your application, but some types of application have restrictions on dynamic allocation to prevent heap fragmentation issues over time.
- Having both objects instantiated at the same time on the same pin is *technically* undefined behavior in Mbed (even if you might get away with it).  Using the variant removes the possibility that this situation could happen.

lefebvresam, 2024-02-07:
How it is working then? Can I see it as a 'switchable pointer', that allocates memory once when you do an emplace and keeps it for the lifetime of the object, so that you do not create memory leaks? Ans when you emplace the interrupt again it re-uses the memory allocated at construction?

multiplemonomials, 2024-02-07:
Pretty much, yeah.  (technically I think the memory is statically allocated inside the object but that doesn't really make a difference)

lefebvresam, 2024-02-07:
It's like it reserves memory for both forms at construction and then fills in concrete details after doing an emplace.