mbed-ce/mbed-os #242: Alternative for PwmOut

lefebvresam, 2024-02-07:
The output where I want PWM in line with my HW is not listed in the pinmap 'PWM'. Probably because it is not one of the timer outputs. If I configure PwmOut with that pin there is no warning or error, and the PWM output is not working as expected. So two questions:

1/ Why there is no warning/error
2/ Is there an easy alternative? (By overwriting PwmOut by an own function that puts a pin high and low?)

JohnK1987, 2024-02-07:
1. I believe [this part should trigger an exception](https://github.com/mbed-ce/mbed-os/blob/031d274c62cc71b4c5a24fd66d3a9e164bb07f1c/targets/TARGET_STM/pwmout_api.c#L87-L95) when you use a pin which is not in the PWM pinmap. But this happen only during call constructor at runtime. 
But this depends on settings of the pinmap, when you do a mistake in the pinmap then it will not work anyway. In general you are responsible for your own design and you can not expect everything will be covered by some automat.
2. You can switch any pin via DigitalOut but this will be not so accurate i think and will take more CPU time ofc.

lefebvresam, 2024-02-07:
To make it more flexible I do the following in the initialization step:

```
    uint32_t peripheral = pinmap_find_peripheral(_buzzerpin, PinMap_PWM);

    if (peripheral == (uint32_t)NC) { // no mapping available
        WARNING("Haptic pin not assigned to a PWM output");   
    } else {
        INFO("Setup buzzer for PWM peripheral at address 0x%X", peripheral);
        _buzzer = new PwmOut(_buzzerpin);
    }
```

And later test if _buzzer == nullptr to perform the necessary actions.

JohnK1987, 2024-02-07:
Ok, I do not know how, but place bad link, [this is correct one](https://github.com/mbed-ce/mbed-os/blob/031d274c62cc71b4c5a24fd66d3a9e164bb07f1c/targets/TARGET_STM/pwmout_api.c#L209-L217) and this call https://github.com/mbed-ce/mbed-os/blob/031d274c62cc71b4c5a24fd66d3a9e164bb07f1c/hal/source/mbed_pinmap_common.c#L68-L80 

So when you put a pin what is not in the pinmap then you should see "pinmap not found for peripheral" your serial terminal.

closed 2024-02-07 by lefebvresam