mbed-ce/mbed-os #243: Resolving names in VSC

lefebvresam, 2024-02-07:
In Visual Studio Code there is a name resolver running in the background. But because of the fragmentation of the mbed code and the generation of some code parts during the build, the name resolver is not able to resolve everything. This can sometime be annoying if your are dealing with objects like Thread.

```
#include "Thread.h"
using namespace rtos;

    private members:
        Thread _thread;
        Thread* _buzzerthread;
```

Then Thread is no longer underlined in red, but is pointing to the test classes, not the real ones.

If you initialize and want tor recognize the proirities:

```
#include "cmsis/CMSIS_5/CMSIS/RTOS2/Include/cmsis_os2.h"

In constructor
_thread(osPriorityBelowNormal, 1024, nullptr, "display")
```

JojoS62, 2024-02-08:
do you have the setting "configurationProvider": "ms-vscode.cmake-tools" in c_cpp_properties.json?
And also the cmake-tools extension installed of course. Then the include path are used that the compiler generates and Intellisense works much better than with other settings.

lefebvresam, 2024-02-09:
Do you mean this. Has no effect.
[image]

JojoS62, 2024-02-09:
Yes, and do have the CMake Extension installed?

lefebvresam, 2024-02-09:
[image]

JojoS62, 2024-02-10:
that looks fine. I have also this setting in c_cpp_properties.json:
"intelliSenseMode": "${default}",

lefebvresam, 2024-02-12:
I don't find one of these files in my test project. Where are they located and when are they created?

JojoS62, 2024-02-12:
compile_commands.json is in my ./build/target folder and the export of this file the default setting in the cmake-extension.
To get the subidr target created, check step 6 in:
https://github.com/mbed-ce/mbed-os/wiki/Project-Setup:-VS-Code

The c_cpp_prorperties.json is located in the .vscode directory.

lefebvresam, 2024-02-12:
I have non of these files even after I generated the build folder again with the line you mentioned in settings.json:

[image]

JojoS62, 2024-02-12:
strange, I have checked again, the compile_commands.json is created when I select the configuration in the CMake extension and the Mbed configuration has finished.
Also the cmake.buildDirectory is not created as in the setting, it does not have the target name. So it looks like the extension is not running properly or is missing something.

[image]

lefebvresam, 2024-02-13:
https://github.com/microsoft/vscode/issues/129466 looks more like a bug

JojoS62, 2024-02-13:
I found this issue also, but it is old. Usually things get fixed soon, so in this case it may happen under some special circumstances. 

[image]

When this CMake View is selected, you see the configuration. When the configuration is changed, it runs the cmake configure with the menu from cmake-variants.yaml.

lefebvresam, 2024-02-14:
[image]

lefebvresam, 2024-02-14:
I did the while procedure and after run and debug nothing happens:

[image]

But I created a build shell script for it instead of using the baked in functionality for cmake.
I was also wondering why compiling in home folder should be faster in case of WSL? My experiënce is that the best solution is to connect to a fast Linux build laptop next to you on the table. Then you unload your desk pc during compilation.

JohnK1987, 2024-07-18:
Project was probably not created according to MbedCE instructions, that is the reason why there was different behaviour.

closed 2024-07-18 by JohnK1987

lefebvresam, 2024-07-19:
In my case there are two ways to compile and program:

1/ The standard way F7, using cmake. In that case I have a build folder with a compile_commands.json file being generated automatically.

2/ Using my own build script _'sh build.sh -f'_  (-f to flash also). In that case the compile_commands.json is not generated.
See [build.sh.txt](https://github.com/user-attachments/files/16312742/build.sh.txt)

I have currently no additional files in  .vscode/settings.json. No settings.json and no c_cpp_properties.json. This is however not practical because this file will not be checked in with source control and this is lost after each installation.

I think the setting c_cpp_properties.json is overwritten in:
[image]
And this survives re-installs of development pc because it's probably a local setting that remains persistent. 

In both cases I do not longer face issues with intellisense, but this is maybe due to the fact that everything is already cached locally and I encountered this issue the first time I used my shell script without compiling with cmake. If you compile with cmake the first time maybe from that point everything is cached.