mbed-ce/mbed-os #244: Starting therad in another thread

lefebvresam, 2024-02-07:
I have the following situation:

Two threads and two flags are defined in a class:

```
        Thread _thread;
        Thread* _buzzerthread;
        atomic_bool _running;
        atomic_bool _buzzerrunning;
```
The first thread is initialized in the constructor:

`_thread(osPriorityBelowNormal, 1024, nullptr, "display")`

When the object is started with a start method, I start the first thread:

```
    _running = true;
    _thread.start(callback(this, &Display::touchProcessor));
```

The first thread runs as long as running is true and it is calling a function when specific events are occuring:

```
void Display::touchProcessor() {
    point_t tp;
    while(_running) {
        _touchUpdate = false;
        _queue->event(this, &Display::touchPanelGet).post(&tp);
        while (!_touchUpdate); // hammer time
        ThisThread::sleep_for(1ms); 
        
        if (_activepage) {
            switch (_tc) {
                case touch: {           ///< touch is detected
                    touchPressHandler(tp);
                    break;
                }
                case no_touch: break;   ///< no touch is detected
                case held: break;       ///< held after touch
                case release: {         ///< release is detected
                    touchReleaseHandler(tp);
                    break;
                }
                case no_cal: break;     ///< no calibration matrix is available
                default:
                    break;
            };
        }
    }
    _thread.terminate();
}
```

When calling touchPressHandler a software buzzer is started and stopped:

```
void Display::touchPressHandler(point_t tp) {
    bool found = false;
    static Timeout t;
    widget_map::reverse_iterator it; // reverse order for layering by modifing alphabetic order of object names
    for (it  = _activepage->getWidgets().rbegin(); it != _activepage->getWidgets().rend(); ++it) {
        if (it->second->isVisibleType() && it->second->checkTouchArea(tp)) {
            uint8_t r = it->second->eventTouchPress();
            if (r == 2) continue;
            if (r == 0 && BUZZER_ON) {
                startBuzzer();
                wait_us(100000);
                stopBuzzer();
                // t.attach(callback(this, &Display::stopBuzzer), BUZZER_DURATION);
            }   
            found = true;
            break;                       
        }
    }
    if (!found) {
        FINE("touch press event for [%s]", _activepage->getObjName());
    }
}
```

I first did a test with Timeout object but get the same issue with a wait.

In the initialization method of the class I prepared the second thread:

```
        _buzzerthread = new Thread(osPriorityRealtime, 1024, nullptr, "buzzer");
        _buzz = new DigitalOut(_buzzerpin);
```

When startbuzzer is called, I should start the second thread:

```
    if (_buzzerthread != nullptr) { // Buzzer thread
        _buzzerrunning = true;
        _buzzerthread->start(callback(this, &Display::buzzBuzzer));
    }
```

Stopbuzzer should stop the thread:

`        _buzzerrunning = false;`

But as soon as the thread started to run, the software hangs forever:

```
void Display::buzzBuzzer() {
    while(_buzzerrunning) {
            wait_us(1000);
    }
    _buzzerthread->terminate();
}
```

lefebvresam, 2024-02-07:
I did a trial where I let the thread close itself:

```
void Display::buzzBuzzer() {
    for (int i = 0; i < 20; i++) {
        _buzz->write(1);
        wait_us(100);
        _buzz->write(0);
        wait_us(100);
    }
    _buzzerthread->terminate();
}
```

If I do startBuzzer:

`_buzzerthread->start(callback(this, &Display::buzzBuzzer));`

It runs only once and stops.
Running again is not possible, until I reallocate the object:

```
        _buzzerthread = new Thread(osPriorityRealtime, 1024, nullptr, "buzzer");
        _buzzerthread->start(callback(this, &Display::buzzBuzzer));
```

But this seems dirty. What I'm missing here?

lefebvresam, 2024-02-07:
Another method is not using a separate thread and post it on a infinite queue running in the main, but of course because theads are shared the SW PWM is not smooth and interrupted by other calls. Better is let the SW just hang for 100ms. And even in that case the PWM is interrupted by interrupts... I think I really need a HW PWM and have to patch my PCB.

JojoS62, 2024-02-10:
> startBuzzer();
>                 wait_us(100000);
>                 stopBuzzer();

this does not look good to me. wait_us is blocking and waisting cpu cycles, it should be used with short delays only.
Sequences can be scheduled by using the EventQueue:

```cpp
	queue.call(startBuzzer);
	queue.call_in(10s, stopBuzzer");
```

Waiting for touch events can be done by using the EventFlags, they are built-in in the Thread class, check the Mbed6 API documentation for the Thread class, there is an example how to wait for flags and set_flags). This is more efficient because it avoids context switching for nothing.

BTW, these discussions are better done in the forum, here it is hard to find it again.

lefebvresam, 2024-02-12:
First I did:

```
static Timeout t;
startBuzzer();
t.attach(callback(this, &Display::stopBuzzer), BUZZER_DURATION);
```

Which is less or more the same.
The point is that starting thread in a thread looks like erroneous.
But more fundamental I left the idea by creating a buzzer with a IO pin in a thread.
It is however interruptred by other threads and sounds not smooth.
I accepted to do hardware patches to reroute a PWM capable output pin to the PWM driver fet.

I will check your suggestion about the flags.

lefebvresam, 2024-02-13:
I did a test with:

```
                _queue->call(this, &Display::startBuzzer);
                _queue->call_in(BUZZER_DURATION, this, &Display::stopBuzzer);
```

```
void Display::stopBuzzer() {
    if (_buzzer != nullptr) {
        _buzzer->suspend();
    }
}

// bug reported at 24/8/20: after resume object should restore settings
void Display::startBuzzer() {
    FINE("Start buzzer");
    if (_buzzer != nullptr) {
        _buzzer->resume();
        _buzzer->period_us(BUZZER_PERIOD_US);
        _buzzer->write(BUZZER_DUTY_CYCLE);
    } else {
        ERROR("Cannot start buzzer");
    }
}
```

Ans this is working also. Note that I did not test again starting a thread in this context. I use PWM module now with hardware patch.

I do not really understand the difference between:

```
                // _queue->event(this, &Display::startBuzzer).post();
                _queue->call(this, &Display::startBuzzer);
```

JojoS62, 2024-02-13:
You can add a thread with higher priority that is just executing thread.start(queue.dispatch _forever());

lefebvresam, 2024-02-13:
Indeed with a separate eventqueue there must be a method to dispatch it forever. If I do one dispatch each time the sofware hangs. I could make it as a class member of Display en do the dispatch forever in a separate thread that may be be hang on that (like main is doing) as you suggested. But there is a simpeler workaround:

```
 startBuzzer(); // doing the start in the current thread to execute it immediately (because here is the source of the timing issue) 
// use the injected main queue, and after buzzer duration the main thread is less occupied
_queue->call_in(BUZZER_DURATION, this, &Display::stopBuzzer);
```

lefebvresam, 2024-02-13:
No this implementation is crashing sometimes. With Timeout I nover have a crash. So I will revert it.

JohnK1987, 2024-07-18:
This seems to be dead so just for an end.
I am not sure if this is good practice - call a second thread from a first thread which is called from main thread. Maybe multiple separated threads with EventFlags or ConditionVariable could help with this.

closed 2024-07-18 by JohnK1987

lefebvresam, 2024-07-19:
My final solution looks like:

```
PwmOut* _buzzer;
Timeout _timeout;
...
void Display::stopBuzzer() {
    if (_buzzer != nullptr) {
        _buzzer->suspend();
    }
}

// bug reported at 24/8/20: after resume object should restore settings
void Display::startBuzzer(int us, float value) {
    FINE("Start buzzer");
    if (_buzzer != nullptr) {
        _buzzer->resume();
        _buzzer->period_us(us);
        _buzzer->write(value);
    } else {
        ERROR("Cannot start buzzer");
    }
}
...

if (r == 0 && widget->second->getSound() && BUZZER_ON) {
    startBuzzer(BUZZER_TOUCH_PERIOD_US, BUZZER_TOUCH_DUTY_CYCLE);
    _timeout.attach(callback(this, &Display::stopBuzzer), BUZZER_TOUCH_DURATION);
}  
```