mbed-ce/mbed-os #245: Reset LSCOEN after test with 32KHz output

lefebvresam, 2024-02-08:
In system_clock.c I create a test option to output the 32kHz signal to PA_2 pin:

```
#if DEBUG_LSCO == 1
    HAL_RCCEx_EnableLSCO(RCC_LSCOSOURCE_LSE); // 32 kHz
#endif
```

But if you disable that option after using it, the register config survives a reboot and the 32KHz clock is still at the output pin.
It makes it impossible to use that pin for digital I/O, Interrupt or Pwmout.
I don't know why those register settings are permanent. I should expect to redo register settings after every reset.

My suggestion is to disable the LSCO before this call to ensure that the output is released when the debug option is put off:

```
// Output clock on LSCO pin(PA2) for debugging purpose
// LSE is set with the low power ticker
HAL_RCCEx_DisableLSCO();
#if DEBUG_LSCO == 1
    HAL_RCCEx_EnableLSCO(RCC_LSCOSOURCE_LSE); // 32 kHz
#endif
```

I tested this and it works as expected.
The strange thing is that HAL_RCCEx_DisableLSCO is never used for other CPU's.

multiplemonomials, 2024-02-11:
Hmm, how about making it a target option instead?  e.g. here: https://github.com/mbed-ce/mbed-os/blob/master/targets/targets.json5#L4834

That would allow people to decide if they want LSCO via a mbed_app.json setting!

JohnK1987, 2024-02-12:
It is not related to your application with the battery? The LSCO is related to LSE and can be used for external devices. So its settings/register could have same behavior like LSE when Vbat is powered, mean it can be stored in backup domain.

lefebvresam, 2024-02-12:
I think you are right. I have a backup battery which is permanently connected. I agree to make it open for everyone and put it in an option, but I was wondering if there is a 'standardized' method for this. With other words, are other devices also exposing it as an option?

JohnK1987, 2024-02-13:
> The strange thing is that HAL_RCCEx_DisableLSCO is never used for other CPU's.

Nothing strange, you do not need call disable when you never call enable. That seems be state of all clock settings of STM32 in Mbed.

So i see this like a project specific option. But i believe it could be handy to make a note about this in https://github.com/mbed-ce/mbed-os/tree/master/targets/TARGET_STM#low-power-clock

closed 2024-02-29 by multiplemonomials