mbed-ce/mbed-os #246: Interrupt is triggering once after enabling

lefebvresam, 2024-02-08:
When I use an interrupt:

```
m_irq.mode(PullUp);
eventThread.start(callback(&queue, &EventQueue::dispatch_forever));
m_irq.fall(queue.event(callback(this, &RA8875::TouchPanelISR)));
 m_irq.disable_irq();   // ensure it is disabled until the init()
```

After a while I enable it:

`m_irq.enable_irq();
`

And it is alway triggered once before the real falling edges occur.

lefebvresam, 2024-02-08:
I already tried this in gpio_irq_api.c for TARGET_STM32U5:

```
    /* Restore previous edge interrupt configuration if applicable */
    if (obj->event & IRQ_RISE) {
        LL_EXTI_EnableRisingTrig_0_31(1 << STM_PIN(obj->pin));
        LL_EXTI_ClearRisingFlag_0_31(1 << STM_PIN(obj->pin));
    }
    if (obj->event & IRQ_FALL) {
        LL_EXTI_EnableFallingTrig_0_31(1 << STM_PIN(obj->pin));
        LL_EXTI_ClearFallingFlag_0_31(1 << STM_PIN(obj->pin));
    }
```
But this is not solving the issue.

lefebvresam, 2024-02-09:
After some investigation I find out that after this call:

```
void InterruptIn::enable_irq()
{
    core_util_critical_section_enter();
    gpio_irq_enable(&gpio_irq);
    core_util_critical_section_exit();  // < this one
}
```
in InterruptIn.cpp, the interrupt is fired three times

lefebvresam, 2024-02-09:
I found it by using this method:

1/ I created a test project with interrupt in CubeIDE and saw that the interrupt was not triggered before enabling and before the first edges.

2/ I created a test project in Mbed-ce and saw the same.

3/ I put pulses on an output pin to check the exact timing between the events. 

```
   // PULSE X

    DigitalOut testout(OUT_4);
    testout.write(1);
    wait_us(1000);
    testout.write(0);

```

And I saw it is my touch controller chip that is producing falling edges.

[image]

Yellow = reset line touch controller chip
Red = interrupt line touch controller chip
Blue = test glitches
Green = Serial data produced by interrupt ISR

multiplemonomials, 2024-02-11:
Nice debugging, glad you figured it out!  Good to close this then?

closed 2024-02-12 by lefebvresam