mbed-ce/mbed-os #256: Compiler still not using C++20

lefebvresam, 2024-03-03:
When you use typical C++20 features like:

`typedef std::map<const char*, Widget*, decltype([](const char* str1, const char* str2){ return std::strcmp(str1, str2) < 0; }) > widget_level_map;`

Then you get:

`[build] /home/sam/git/hmipanel/source/widgets/widget.h:30:49: error: lambda-expression in unevaluated context only available with '-std=c++20' or '-std=gnu++20'`

JohnK1987, 2024-03-03:
Your probably need to look into profiles settings
https://github.com/mbed-ce/mbed-os/blob/2deef5a6817a9bdf2db597c5f4e61db3f15474e9/tools/profiles/debug.json#L2-L17

However, the compiler will probably report many issues because MbedOS code base is compatible with older version i think.

lefebvresam, 2024-03-06:
I think more issues will popup if I do the switch. As long as the whole project is not C++20 compliant, users are not encouraged to switch to the latstest standard because of these potential issues. I don't know if there are 'programming rules' in the CE to align programming efforts into a specific C++ revision?

JojoS62, 2024-03-10:
In Mbed-os, the C++ standard was long time only C++11 and also not always using the current gcc. It was handled conservative because changes caused a lot of testing to ensure the correct functions. And Mbed-os had to maintain compatibility for GCC, IAR and Keil compilers, and not all where always at the same standard level. Also the C++ runtime libs differ and new features are usually later available in Cortex-M than in X86/64 architectures.
With Mbed-CE it is a little bit easier as it support is limited to GCC, so I would like also to have C++20 using the current gcc version. Newer GCC were also more picky about warnings, the current Mbed code base produces a lot and some clean up is also needed.

multiplemonomials, 2024-03-20:
Hmm... I am a bit hesitant to change all of Mbed to C++20, because the Arduino IDE is using a very old GCC version that does not support it, so it would break Arduino support.

That said, you can enable C++20 for your own code, I believe, by doing something like
```
set_property(TARGET MyExecutable PROPERTY CXX_STANDARD 20)
```
That will enable C++20 support for a specific executable target (MyExecutable).  Let me know if that's not working!

lefebvresam, 2024-07-19:
I did not switched to C++20 because of all potential issues and warnings. I used the 'old way' practices. However this could be a discussion point for the whole community how and when to switch to a newer cpp version to make it feature proof. I don't know if there is already a topic created for it. I prefer using the 'out of the box' practices as much as possible.