mbed-ce/mbed-os #285: SPI defined as enum in OSPIBlockDevice.h

timIdeaTech, 2024-06-19:
The following is in OSPIFBlockDevice.h:
```
enum _mode {
    SPI,
    SOPI,
    DOPI
};

```

and is causing the following compiler error when I try to instantiate an SPI device when OSPIFBlockDevice.h has been included:
`error: reference to 'SPI' is ambiguous`

I can get around it by using mbed::SPI instead. I'm thinking those enum's should have been prefixed with OSPI_BD_ or similar. The original mbed OS repo also has this problem. If there is no objections I could do the pull request but thought I'd ask here first.

multiplemonomials, 2024-06-19:
Ooh good catch on this!

Hmm, what about changing the declaration to:
```cpp
enum mode {
    SPI,
    SOPI,
    DOPI
};
```
and moving it inside the `OSPIFBlockDevice` class?  That should make it so they aren't in the global namespace, right?

timIdeaTech, 2024-06-20:
The problem would be id the _mode value is public because OSPIFBlockDevice::chnage_mode() is public. Also, naming a global enum __mode seems like a recipe for collision also, hence my suggestion of adding a prefix to them all. Wouldn't ospif_mode and OSPIF_MODE_[SPI|SOPI|DOPI] or even something like ospif_opi_mode and OPSIF_OPI_MODE_[SPI|SOPI|DOPI] make sense?

multiplemonomials, 2024-06-20:
Strictly speaking, it's fine to move it inside the class because you could just make it public in the class, so then people would be able to access it just fine.  Also, either way, it's a breaking change -- people will have to change `DOPI` to either `OSPIFBlockDevice::DOPI` or `OSPIF_MODE_DOPI`.

That said, you're right it would have more symmetry with the other types to do it the way you said.  So I'll accept either way!

timIdeaTech, 2024-06-20:
Ok, I get where and why your going with this. Due to my only moderate experience with C++, I had to go back and reread and get the nuance of the _move it inside the ... class_ suggestion to get rid of the global collision. For this reason I don't have a good feel for the proper convention here.

That said, the convention used in this driver are global driver prefixed enum's so maybe makes sense to stick with it. I'll give this a go...

multiplemonomials, 2024-07-10:
Fixed by #291

closed 2024-07-10 by multiplemonomials