mbed-ce/mbed-os #297: Configuration of mbed os may be useful

wubaojun1982, 2024-07-05:
In my past development, there are many problems from configuration. In the zephyr or rt-thread rtos system, they used Kconfig as system configuration. So i think we can used Kconfig tools as the system configuration tools. Thanks very much.

JohnK1987, 2024-07-05:
Hello,

from my point of view this is not an issue and should be discussed in https://github.com/mbed-ce/mbed-os/discussions/categories/ideas

When i look on Zephyr's description of the kconfig then i see similar/same what do the configuration system in Mbed via .json files and cmake.
So what are differences? I suppose you did a comparison so would be good to share your findings as arguments.

BR, Jan

JohnK1987, 2024-07-18:
No more reactions plus this is not an issue but an idea.
Next time please use https://github.com/mbed-ce/mbed-os/discussions/categories/ideas but it is necessary to come with some comparison, with the expected benefits why we should choose to the kconfig. Not just a wish.

Thank you for understanding.

closed 2024-07-18 by JohnK1987