mbed-ce/mbed-os #300: Missing include files reffed from mbed.h

Gijsco, 2024-07-12:
mbed.h includes the following files that don't exist. Cant compile as a result.
#include "mbed-target-config.h"
#include <math.h>
#include "drivers/DigitalOut.h"
#include "drivers/BusOut.h"
#include "drivers/MbedCRC.h"
#include "drivers/HighResClock.h"
#include "drivers/Ticker.h"
#include "drivers/LowPowerClock.h"
#include "drivers/RealTimeClock.h"
#include "platform/mbed_thread.h"
#include "platform/ATCmdParser.h"
#include "platform/FileSystemHandle.h"
#include "platform/DeepSleepLock.h"
#include "platform/ScopedRomWriteLock.h"
#include "platform/mbed_stats.h"
#include "platform/Stream.h"

JohnK1987, 2024-07-12:
Hello,

welcome and thank you for the issue. I am sure the MbedOS is ok and I believe the issue is just in your setup. To provide assistance we need more context, so can you share little bit more information please?
Is it your first attempt with MbedCE? Did you use our [Wiki guides](https://github.com/mbed-ce/mbed-os/wiki)? For example [New Project Setup Guide](https://github.com/mbed-ce/mbed-os/wiki/New-Project-Setup-Guide).

Maybe some pictures of the project structure and the content of top lvl CMakeLists.txt file

If we found some misleading information we will improve instructions.

BR, Jan

Gijsco, 2024-07-12:
Hi Jan,

Sorry for the inconvenience. The goal of the project is to setup a STM32F103C8U6 (Blue-Pill) and use some simple CAN-bus features. I am switching from MBed-studio to VScode together with the switch to Community Edition. I followed the [Wiki Guide](https://github.com/mbed-ce/mbed-os/wiki) on a fresh windows installation which led me to:
- Creating a local project repository as [New Project Setup Guide](https://github.com/mbed-ce/mbed-os/wiki/New-Project-Setup-Guide).
- Following the [VScode guide](https://github.com/mbed-ce/mbed-os/wiki/Project-Setup:-VS-Code)
- Following up on [dependencies](https://github.com/mbed-ce/mbed-os/wiki/Toolchain-Setup-Guide) as Cmake, python, ninja and GNU toolchain.

Ending up issue above, so i realy appreciate your offer on assistance. Also could you educate me on why in a normal case an include can be made without the actual file being there without being an issue? I obviously made the (wrong) assumption it would which leaves room to learn a bit. As for the Wiki Guides, i think they are great because i would not have thought of combining all these different tools to get this going.

Best Regards, Gijs

JohnK1987, 2024-07-12:
Aha, the Bluephill is not supported and has to be add via [Custom target](https://github.com/mbed-ce/mbed-ce-custom-targets/tree/master). I have one piece of Bluephill somewhere, i think. I will look on it tomorrow.

As source we can use https://github.com/ARMmbed/stm32customtargets/tree/master

BR, Jan

JohnK1987, 2024-07-13:
Ok, look on this https://github.com/mbed-ce-libraries-examples/Custom_Bluepill and let me know.

BR, Jan

Gijsco, 2024-07-13:
I did make a custom target like i did in mbed studio. Worked fine then, is this the reason why i'm getting those compile errors?

JohnK1987, 2024-07-13:
Probably yes, because Mbed Studio did not work with CMake. In MbedCE is necessary to manage all files via CMake.

Gijsco, 2024-07-13:
Oke, thank you! What should one do after the tutorials above to get a custom target going?

JohnK1987, 2024-07-13:
1. Run Command line and reproduce steps from this [readme](https://github.com/mbed-ce-libraries-examples/Custom_Bluepill?tab=readme-ov-file#how-to-start) . For example like below. This means clone remote repository **Custom_Bluepill** to local repository **MbedCE-Bluepill** (you can fill your own name here) and update MbedOS to latest.
   - git clone --recursive https://github.com/mbed-ce-libraries-examples/Custom_Bluepill.git MbedCE-Bluepill
   - cd HerePlaceMyProjectName && git submodule update --remote mbed-os

2. Run VS Code and open the project on patch (in my case) C:\MbedCE\HerePlaceMyProjectName.
3. VS Code will ask you for GCC (the content of the list depends on what GCC versions you have installed)
[image]
 After you choose GCC and everything necessary is installed, the VS Code will auto start CMake configuration which will install many dependencies and it fail at the end. 
4. Then you need set the MBED_TARGET. Click to "Unknown" - Left bottom
[image]
Then you will see this on top
[image]
After your choice the configuration will be triggered again and should be positive now - "[cmake] -- Build files have been written to: C:/MbedCE/MbedCE-Bluepill/build"
5. When you hit the button like bellow then compilation starts.
[image]
6. Optional. On the last line of cmake-variants.yaml file set your [upload method](https://github.com/mbed-ce/mbed-os/wiki/Upload-Methods). Default is NONE = it generates just .bin file. Optimal for external loaders like DFU method and so on (I personaly prefer [STM32CUBE](https://github.com/mbed-ce/mbed-os/wiki/Upload-Methods#stm32cube) with ST-Link).

JohnK1987, 2024-07-15:
@Gijsco, any progress here?

Gijsco, 2024-07-16:
> @Gijsco, any progress here?

Yes, building works now. Thank you so much! Debugging does not work yet. It does when I create a project with STM32 CS code Extension so i'm trying to figure out how to place Mbed-CE into an existing cmake project.

JohnK1987, 2024-07-16:
Nice.

> Debugging does not work yet.

An error or ?

Gijsco, 2024-07-16:
> An error or ?

Yeah:
[proc] Executing command: gdb --version
[proc] The command: gdb --version exited with code: 1
Also comes with a popup saying: Unable to find GDB in default search path and gdb.
This does not happen with the STM generated project, I also tried copying the launch.json and tasks.json file over to get it going but no change.

JohnK1987, 2024-07-16:
I will look on this later and come back to you.

Gijsco, 2024-07-16:
Ok, i came past this now by removing the GCC and gdb packages from MinGW.
Now i'm back at:  message(FATAL_ERROR "You must define MBED_TARGET to the Mbed target you wish to build for!")
The most reproducable version of what i did was 
- clone: https://github.com/mbed-ce-libraries-examples/Custom_Bluepill
- add the mbed-os submodule as instructed on that page. 
- open VS-code and in STM32 code extension: import CMake project (and select the Custom_Bluepill folder and F103C8Tx chip)
(this generates build settings that normally work with a bluepill on a stlink-v2)
- Hit the debug button.

JohnK1987, 2024-07-16:
> I also tried copying the launch.json and tasks.json

These files are auto generated during build i think. So there is no point to modified them.
https://github.com/mbed-ce/mbed-os/blob/eb0012983a23c67b2de4a9854c69931d665df4ff/tools/cmake/mbed_ide_debug_cfg_generator.cmake#L94

-------------------------------------------------------------------------------------------------------------------------

>Now i'm back at: message(FATAL_ERROR "You must define MBED_TARGET to the Mbed target you wish to build for!")

Repeat the step 4 from above.

-------------------------------------------------------------------------------------------------------------------------

> * open VS-code and in STM32 code extension

I do not know this is not expected use case.

-------------------------------------------------------------------------------------------------------------------------

>[proc] Executing command: gdb --version
[proc] The command: gdb --version exited with code: 1
Also comes with a popup saying: Unable to find GDB in default search path and gdb.

Did you change [this](https://github.com/mbed-ce-libraries-examples/Custom_Bluepill/blob/fd49cb290df253a19a638be74cff205408142ab9/cmake-variants.yaml#L29) from NONE to `STM32CUBE` ? Of course you should have installed [STM32CubeDE or STM32CubeCLT](https://github.com/mbed-ce/mbed-os/wiki/Upload-Methods#stm32cube). Then change this
[image]
to 
[image]
Last what i do not know why happen is this timeout

[image]

wdx04, 2024-07-17:
"GDB: 127.0.0.1:23331: Connection timed out" means the GDB server did not start correctly. You can find detailed information in the GDB server terminal.
Some possible causes:
1. The ST-LINK firmware is outdated. Upgrade it using STM32CubeProgrammer.
2. STM32Cube connection parameters are not set properly. Add the following lines in the CMakeLists.txt, before "add_subdirectory(mbed-os)":
set(STM32CUBE_UPLOAD_ENABLED TRUE)
set(STM32CUBE_CONNECT_COMMAND -c port=SWD mode=UR)
set(STM32CUBE_GDBSERVER_ARGS -d -k)

JohnK1987, 2024-07-17:
@wdx04 thanks for the tips.
1. ST-link's firmware is OK.
2. I am using [this](https://github.com/mbed-ce-libraries-examples/Custom_Bluepill/blob/main/custom_targets/upload_method_cfg/BLUEPILL_F103C8.cmake) in default and I also tried different settings but unfortunately nothing change.

```
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command C:/ST/STM32CubeCLT/STLink-gdb-server/bin/ST-LINK_gdbserver.exe -d -k -cp C:/ST/STM32CubeCLT/STM32CubeProgrammer/bin -p 23331 --halt" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: & 'C:/Program Files/CMake/bin/cmake.exe' --build C:/MbedCE/HerePlaceMyProjectName/build --target main 

ninja: no work to do.
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: C:/ST/STM32CubeCLT/STLink-gdb-server/bin/ST-LINK_gdbserver.exe -d -k -cp C:/ST/STM32CubeCLT/STM32CubeProgrammer/bin -p 23331 --halt 

STMicroelectronics ST-LINK GDB server. Version 7.5.0
Copyright (c) 2023, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 31
        Listen Port Number         : 23331
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled
        InitWhile                  : Enabled

COM frequency = 4000 kHz
Target connection mode: Under reset
Reading ROM table for AP 0 @0xe00fffd0
Hardware watchpoint supported by the target
ST-LINK Firmware version : V2J45M31
Device ID: 0x0
PC: 0x1fffb1e8
ST-LINK device status: HALT_MODE
ST-LINK detects target voltage = 0.11 V

Error in initializing ST-LINK device.
Reason: ST-LINK: Could not verify ST device! Abort connection.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command C:/ST/STM32CubeCLT/STLink-gdb-server/bin/ST-LINK_gdbserver.exe -d -k -cp C:/ST/STM32CubeCLT/STM32CubeProgrammer/bin -p 23331 --halt" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

Gijsco, 2024-07-17:
So i followed your steps you posted yesterday Jan and came with the following in the debug console:
```
=thread-group-added,id="i1"
GNU gdb (GNU Tools for STM32 12.3.rel1.20240612-1315) 13.2.90.20230627-git
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Warning: Debuggee TargetArchitecture not detected, assuming x86_64.
=cmd-param-changed,param="pagination",value="off"
ERROR: Unable to start debugging. Unexpected GDB output from command "-exec-run". Don't know how to run.  Try "help target".
The program 'C:\Users\G3\Documents\VScodeProjects\Custom_Bluepill_cpy\build\YOUR_MBED_TARGET-Debug\main.elf' has exited with code 42 (0x0000002a).

I'm no cmake hero so i have no clue how to use the working CMake from the STM example to get this MBED bluepill example going.
```

Gijsco, 2024-07-17:
> @wdx04 thanks for the tips.
>
> 1. ST-link's firmware is OK.
> 2. I am using [...]

You seem to be getting somewhere, only your target voltage is very low. Probably the bluepill is not powered or connected to the ST-link.

with ' makefile Tools'  removed from VScode i used the .vscode/.....json files from the working stm example which is bringing me further. I'm getting this now:
```
STMicroelectronics ST-LINK GDB server. Version 7.8.0
Copyright (c) 2024, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 31
        Listen Port Number         : 50000
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled

COM frequency = 4000 kHz
Target connection mode: Default
Target connection failed. Try connecting under reset
Target connection failed
COM frequency = 1800 kHz
Target connection mode: Default
Reading ROM table for AP 0 @0xe00fffd0
Hardware watchpoint supported by the target 
ST-LINK Firmware version : V2J45S7
Device ID: 0x410
PC: 0x800025c
ST-LINK device status: HALT_MODE
ST-LINK detects target voltage = 3.25 V
ST-LINK device status: HALT_MODE
ST-LINK device initialization OK
Stm32Device, pollAndNotify running...
SwvSrv state change: 0 -> 1
Waiting for debugger connection...
Waiting for connection on port 50000...
Waiting for connection on port 50001...
Accepted connection on port 50000...
Debugger connected
Waiting for debugger connection...
Waiting for connection on port 50000...
GDB session thread running
GdbSessionManager, session started: 1
Enter STM32_SystemReset() function 
Enter STM32_InitAfterReset() function 
NVIC_DFSR_REG = 0x00000008
NVIC_CFGFSR_REG = 0x00000000
Stm32Device, closeDevice() entry
GDB session, device event: 5
Stm32Device, pollAndNotify stopped
Stm32Device, closeDevice() exit
 ------ Switching to STM32CubeProgrammer ----- 
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.17.0                  
      -------------------------------------------------------------------

Log output file:   C:\Users\G3\AppData\Local\Temp\STM32CubeProgrammer_a06456.log
ST-LINK SN  : 56FF6A065077525139571187
ST-LINK FW  : V2J45S7
Board       : --
Voltage     : 3.25V
Error: ST-LINK error (DEV_TARGET_CMD_ERR)
Encountered Error when opening c:\ST\STM32CubeCLT_1.16.0/STM32CubeProgrammer/bin\STM32_Programmer_CLI.exe
 ------ Switching context ----- 
Error in STM32CubeProgrammer
[2024-07-17T07:45:07.007Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

wdx04, 2024-07-17:
@Gijsco You problem seems to be caused by the mix of forward/backward slashes in the path of STM32_Programmer_CLI.exe. Please modify the CMake cache variable STM32CubeProg_PATH to use only forward slashes, and make sure that executable file exists.

Gijsco, 2024-07-17:
Well that got me somewhere.
```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from C:/ST/STM32CubeCLT_1.16.0/GNU-tools-for-STM32/bin/arm-none-eabi-objdump.exe --syms -C -h -w C:/Users/G3/Documents/VScodeProjects/Custom_Bluepill_cpy/build/YOUR_MBED_TARGET-Develop/main.elf
Reading symbols from c:/st/stm32cubeclt_1.16.0/gnu-tools-for-stm32/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users/G3/Documents/VScodeProjects/Custom_Bluepill_cpy/build/YOUR_MBED_TARGET-Develop/main.elf
Launching GDB: "C:/ST/STM32CubeCLT_1.16.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb.exe" -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Finished reading symbols from objdump: Time: 87 ms
Finished reading symbols from nm: Time: 77 ms
Output radix now set to decimal 10, hex a, octal 12.
Input radix now set to decimal 10, hex a, octal 12.
0x08000254 in uart_irq (uart_name=0) at C:/Users/G3/Documents/VScodeProjects/Custom_Bluepill_cpy/mbed-os/targets/TARGET_STM/TARGET_STM32F1/serial_device.c:55
55	    int8_t id = get_uart_index(uart_name);
Program stopped, probably due to a reset and/or halt issued by debugger
STM32 Successfully completed reset operation (System reset)
Loading section .text, size 0x6404 lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x8006404
Loading section .data, size 0x180 lma 0x800640c
Start address 0x080001fc, load size 25996
Transfer rate: 12 KB/sec, 6499 bytes/write.
Temporary breakpoint 1 at 0x8004304: file C:/Users/G3/Documents/VScodeProjects/Custom_Bluepill_cpy/main.cpp, line 5.
Note: automatically using hardware breakpoints for read-only addresses.
STM32 Successfully completed reset operation (System reset)

Temporary breakpoint 1, main () at C:/Users/G3/Documents/VScodeProjects/Custom_Bluepill_cpy/main.cpp:5
5	int main(){

Breakpoint 2, main () at C:/Users/G3/Documents/VScodeProjects/Custom_Bluepill_cpy/main.cpp:8
8	        thread_sleep_for(100);
warning: Exception condition detected on fd 452
Remote communication error.  Target disconnected.: No error.
GDB never responded to an interrupt request. Trying to end session anyways
```
Thank you @wdx04 ! Its a start.
It will stop at the breakpoint at main, after that at the breakpoint i set at thread_sleep_for(). 
After that it will not come around and hang itself up.

wdx04, 2024-07-18:
@Gijsco It seems the processor entered STOP mode when you called thread_sleep_for function.
You may try one of these two methods:
1. Prevent the processor from entering STOP mode by calling sleep_manager_lock_deep_sleep() at the beginning of your main() fucntion.
2. Manually enable debugging in STOP mode before sleeping:
__HAL_RCC_DBGMCU_CLK_ENABLE();
HAL_DBGMCU_EnableDBGStopMode();

Gijsco, 2024-07-18:
@wdx04 Thank you! i'l try that to. I just replaced the sleep function with wait_us(WAIT_TIME); and that also did the trick.
Thank you and @JohnK1987 both for helping me out.
Happy debugging :)

JohnK1987, 2024-07-18:
That is nice!

Maybe try  to remove SLEEP in mbed_app.json file, then you can still use thread_sleep_for() function, i think.
```
 "device_has_remove": ["SLEEP"],
```

@Gijsco can we lose the issue?
Just recommendation for the future. If you will have any question or face trouble try to use [Discussion](https://github.com/mbed-ce/mbed-os/discussions) firstly, then we will together decide if this is an issue or not.

@wdx04very appreciate your help!

Gijsco, 2024-07-18:
We can lose the issue. And i'l jump to the discussion page next time.

closed 2024-07-18 by JohnK1987