mbed-ce/mbed-os #301: Compiling not working

lefebvresam, 2024-07-15:
When using my build script, I get:

```
sam@saleconix:~/git/hmipanel$ sh build.sh
Generate build folder for target 'HMC20' with profile 'develop'
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Checking for Python package intelhex -- found
-- Checking for Python package prettytable -- found
-- Checking for Python package future -- found
-- Checking for Python package jinja2 -- found
-- Mbed: First CMake run detected, generating configs...
mbed_config.cmake has been generated and written to '/home/sam/git/hmipanel/build/develop/mbed_config.cmake'
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /usr/local/gcc-arm/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/local/gcc-arm/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/local/gcc-arm/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/share/cmake-3.29/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/usr/local/gcc-arm/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/home/sam/git/hmipanel/build/develop/CMakeFiles/CMakeScratch/TryCompile-y6z5Mv'

    Run Build Command(s): /usr/bin/ninja -v cmTC_3c575
    [1/2] /usr/local/gcc-arm/bin/arm-none-eabi-gcc   -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -fomit-frame-pointer -g3 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=softfp -mcpu=cortex-m33 -o CMakeFiles/cmTC_3c575.dir/testCCompiler.c.obj -c /home/sam/git/hmipanel/build/develop/CMakeFiles/CMakeScratch/TryCompile-y6z5Mv/testCCompiler.c
    FAILED: CMakeFiles/cmTC_3c575.dir/testCCompiler.c.obj
    /usr/local/gcc-arm/bin/arm-none-eabi-gcc   -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -fomit-frame-pointer -g3 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=softfp -mcpu=cortex-m33 -o CMakeFiles/cmTC_3c575.dir/testCCompiler.c.obj -c /home/sam/git/hmipanel/build/develop/CMakeFiles/CMakeScratch/TryCompile-y6z5Mv/testCCompiler.c
    /bin/sh: 1: /usr/local/gcc-arm/bin/arm-none-eabi-gcc: Exec format error
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  mbed-os/tools/cmake/app.cmake:55 (enable_language)
  CMakeLists.txt:8 (include)

-- Configuring incomplete, errors occurred!
```

JohnK1987, 2024-07-15:
Hello,

how I remember you already had working MbedCE on your machine, so what happen?
I can not verified this because i do not use Linux but first what i see is this

```
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /usr/local/gcc-arm/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/local/gcc-arm/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/local/gcc-arm/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/share/cmake-3.29/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/usr/local/gcc-arm/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/home/sam/git/hmipanel/build/develop/CMakeFiles/CMakeScratch/TryCompile-y6z5Mv'
```
This is how it looks like on Windows

```
[cmake] -- The C compiler identification is GNU 11.3.1
[cmake] -- The CXX compiler identification is GNU 11.3.1
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: C:/ST/STM32CubeCLT/GNU-tools-for-STM32/bin/arm-none-eabi-gcc.exe
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: C:/ST/STM32CubeCLT/GNU-tools-for-STM32/bin/arm-none-eabi-gcc.exe - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: C:/ST/STM32CubeCLT/GNU-tools-for-STM32/bin/arm-none-eabi-g++.exe - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
```

So maybe your GNU is not registered in environment variables? I do not know just a tip.

BR, Jan

lefebvresam, 2024-07-15:
Yes I has to re-install everything after an unwanted remove of an important Linux folder.

Check this:
[CMakeConfigureLog.yaml.txt](https://github.com/user-attachments/files/16237586/CMakeConfigureLog.yaml.txt)

lefebvresam, 2024-07-15:
I found it.
I used the wrong package:
arm-gnu-toolchain-13.3.rel1-aarch64-arm-none-eabi.tar.xz
instead of
arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi.tar.xz

JohnK1987, 2024-07-15:
OK, then solved. Good luck.
Closed.

closed 2024-07-15 by JohnK1987