mbed-ce/mbed-os #304: ATHandler::cmd_start build error

chrissnow, 2024-07-17:
We are looking at porting an existing project that uses cellular over to mbed-ce but there seems to be a compile error and I cant figure out why given the types seem to match.

https://github.com/mbed-ce/mbed-os/blob/71cb8129af5803b3268eadbe56e20e10a42183aa/connectivity/cellular/source/framework/device/ATHandler.cpp#L1193

```
[1/3] Building CXX object mbed-os/connectivity/cellular/CMakeFiles/mbed-cellular.dir/source/framework/device/ATHandler.cpp.obj
FAILED: mbed-os/connectivity/cellular/CMakeFiles/mbed-cellular.dir/source/framework/device/ATHandler.cpp.obj
"C:\Program Files\ccache-4.10.1-windows-x86_64\ccache.exe" C:\PROGRA~2\ARMGNU~1\125863~1.3RE\bin\AR10B2~1.EXE @mbed-os\connectivity\cellular\CMakeFiles\mbed-cellular.dir\source\framework\device\ATHandler.cpp.obj.rsp -MD -MT mbed-os/connectivity/cellular/CMakeFiles/mbed-cellular.dir/source/framework/device/ATHandler.cpp.obj -MF mbed-os\connectivity\cellular\CMakeFiles\mbed-cellular.dir\source\framework\device\ATHandler.cpp.obj.d -o mbed-os/connectivity/cellular/CMakeFiles/mbed-cellular.dir/source/framework/device/ATHandler.cpp.obj -c RETRACTEDcell_test/mbed-os/connectivity/cellular/source/framework/device/ATHandler.cpp
RETRACTED\cell_test/mbed-os/connectivity/cellular/source/framework/device/ATHandler.cpp: In member function 'void mbed::ATHandler::cmd_start(const char*)':
RETRACTED\cell_test/mbed-os/connectivity/cellular/source/framework/device/ATHandler.cpp:1193:38: error: no matching function for call to 'sleep_until(std::__duration_common_type<std::common_type<long long int, short unsigned int>, std::ratio<1, 1000>, std::ratio<1, 1000>, void>::type)'
 1193 |         rtos::ThisThread::sleep_until(_last_response_stop.time_since_epoch() + _at_send_delay);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from RETRACTEDcell_test/mbed-os/connectivity/cellular/source/framework/device/ATHandler.cpp:26:
RETRACTED\cell_test\mbed-os\rtos\include\rtos\thisthread.h:238:6: note: candidate: 'void rtos::ThisThread::sleep_until(uint64_t)'
  238 | void sleep_until(uint64_t millisec);
      |      ^~~~~~~~~~~
RETRACTED\cell_test\mbed-os\rtos\include\rtos\thisthread.h:238:27: note:   no known conversion for argument 1 from 'std::__duration_common_type<std::common_type<long long int, short unsigned int>, std::ratio<1, 1000>, std::ratio<1, 1000>, void>::type' {aka 'std::chrono::duration<long long int, std::ratio<1, 1000> >'} to 'uint64_t' {aka 'long long unsigned int'}
  238 | void sleep_until(uint64_t millisec);
      |                  ~~~~~~~~~^~~~~~~~
RETRACTED\cell_test\mbed-os\rtos\include\rtos\thisthread.h:248:6: note: candidate: 'void rtos::ThisThread::sleep_until(rtos::Kernel::Clock::time_point)'
  248 | void sleep_until(Kernel::Clock::time_point abs_time);
      |      ^~~~~~~~~~~
RETRACTED\cell_test\mbed-os\rtos\include\rtos\thisthread.h:248:44: note:   no known conversion for argument 1 from 'std::__duration_common_type<std::common_type<long long int, short unsigned int>, std::ratio<1, 1000>, std::ratio<1, 1000>, void>::type' {aka 'std::chrono::duration<long long int, std::ratio<1, 1000> >'} to 'rtos::Kernel::Clock::time_point' {aka 'std::chrono::time_point<mbed::internal::OsClock, std::chrono::duration<long long int, std::ratio<1, 1000> > >'}
  248 | void sleep_until(Kernel::Clock::time_point abs_time);
      |                  ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
ninja: build stopped: subcommand failed.
```

The project was originally based on 
https://github.com/ARMmbed/mbed-os-example-cellular

Anything that makes use of the ATHandler Class is likely to fail to build.

JohnK1987, 2024-07-17:
Hello,

maybe **mbed-cellular** has to be add to top lvl CMakeLists.txt

Change from this 
> target_link_libraries(main  mbed-os)

to this
> target_link_libraries(main  mbed-os  mbed-cellular)

wdx04, 2024-07-18:
@chrissnow Hello, are you using the latest Mbed CE? In ATHandler.cpp, line 1193 should be:

```C++
rtos::ThisThread::sleep_until(_last_response_stop + _at_send_delay); 
```

But the code in your compiler output shows:
```C++
rtos::ThisThread::sleep_until(_last_response_stop.time_since_epoch() + _at_send_delay);
```

chrissnow, 2024-07-18:
@wdx04 I'm on the latest commit?
[image]
https://github.com/mbed-ce/mbed-os/blob/eb0012983a23c67b2de4a9854c69931d665df4ff/connectivity/cellular/source/framework/device/ATHandler.cpp#L1193C3-L1193C77

wdx04, 2024-07-18:
> @wdx04 I'm on the latest commit? [...]

Yes, this is the latest commit. But maybe you have multiple copies of mbed-os and this is not the one you are using?
The compiler output shows you are using this file: RETRACTED/cell_test/mbed-os/connectivity/cellular/source/framework/device/ATHandler.cpp. 
You may open the file with an editor and verify that it matches the latest version, line 1193.

chrissnow, 2024-07-18:
I think I had meddled at the point I took the logs to try and get it working.

chrissnow, 2024-07-18:
back to normal and you get

mbed-os/connectivity/cellular/source/framework/device/ATHandler.cpp:1193: undefined reference to `rtos::ThisThread::sleep_until(std::chrono::time_point<mbed::internal::OsClock, std::chrono::duration<long long, std::ratio<1ll, 1000ll> > >)'

multiplemonomials, 2024-07-19:
OK I can reproduce this on my end by using any cellular APIs in an application.  To be honest, I do not believe anyone has tried to use these features of Mbed CE since the build system changeover.  OK lemme take a look...

multiplemonomials, 2024-07-19:
Ah OK I see the issue.  This ultimately comes from a rather odd decision by ARM (tho, maybe made worse by Mbed CE) to support building both baremetal Mbed and Mbed OS in the same configured project, rather than having users pick one or the other at configure time.  This leads to weirdness: libraries like `mbed-cellular` are compiled against `mbed-baremetal`, but linked against `mbed-os`.  In this case, the problem is that `rtos::Kernel::duration` changes its underlying type between baremetal and OS, so the function it expects to find no longer exists.

multiplemonomials, 2024-07-19:
As a workaround, for now, try dropping the following into your CMakeLists.txt after add_subdirectory(mbed-os):
```cmake
target_link_libraries(mbed-cellular PUBLIC mbed-rtos-flags)
```
That will cause mbed-cellular to be compiled against the RTOS, fixing the link error.  This workaround, plus #307, allowed me to link an app with mbed-cellular!

chrissnow, 2024-10-15:
@multiplemonomials  Apologies it's taken so long to test all of this, were close now!

The project links fine now and AT commands are passing correctly however NSAPI_PPP_AVAILABLE is not true despite MBED_CONF_LWIP_PPP_ENABLED being true in mbed-target-config.h

```
#define MBED_CONF_PPP_ENABLED 0
#define MBED_CONF_LWIP_PPP_ENABLED 1
#define NSAPI_PPP_AVAILABLE (MBED_CONF_PPP_ENABLED || MBED_CONF_LWIP_PPP_ENABLED)
```

This means PPP doesn't come up during connect in here.
mbed-os\connectivity\cellular\source\framework\AT\AT_CellularContext.cpp

It's like mbed-target-config.h isnt being included for the cellular stack.

multiplemonomials, 2024-10-16:
Hmm, could you elaborate on how you know that PPP doesn't come up?  If you put a #warning inside one of the 
```
#if NSAPI_PPP_AVAILABLE
```
blocks, does it print during compilation?

chrissnow, 2024-10-16:
Interesting, VSCode doesn't think it's defined but it is
[image]

Time to dig deeper as to why PPP doesn't come up after the CONNECT response

chrissnow, 2024-10-16:
Fixed it :-)
I forgot I had to patch the EC2X driver to work with the EG915.
Once I have tested it a bit further I will sort a PR for a EG915 driver.

multiplemonomials, 2025-01-18:
The workaround should no longer be needed as of #419