mbed-ce/mbed-os #305: Bootloader support

chrissnow, 2024-07-17:
I don't think there is a way to implement bootloaders? mbed-tools was never updated to include it.

https://github.com/ARMmbed/mbed-tools/issues/156
https://github.com/ARMmbed/mbed-tools/issues/315

The give away is this in the configure

> WARNING: You are attempting to override an undefined config parameter `target.header_format`.
> It is an error to override an undefined configuration parameter. Please check your target_overrides are correct.
> The parameter `target.header_format` will not be added to the configuration.
> WARNING: You are attempting to override an undefined config parameter `target.bootloader_img`.
> It is an error to override an undefined configuration parameter. Please check your target_overrides are correct.
> The parameter `target.bootloader_img` will not be added to the configuration.

How complicated to implement this would it be? I could work around it if too complex.

There was once talk in mbed of supporting mcuboot as it has a lot of nice features.
https://docs.mcuboot.com/

multiplemonomials, 2024-07-19:
I should have some updates about this within the next week or two!  Currently the CMake build system (started by ARM, improved by us) does not support this, but it can, someday.

multiplemonomials, 2024-07-19:
However, I will admit, it's probably a 1-2 years away thing, not a few months thing.

multiplemonomials, 2024-11-23:
To provide an update, since the merging of the memory banks PR, progress is being made on this.  It is now possible for Mbed targets to coeexist with a bootloader, *if* the target's linker script uses `MBED_CONFIGURED_ROM_BANK_x_START/SIZE` variables to size its flash region.  If that is the case, you can use the `memory_bank_config` section in mbed_app.json to reconfigure your target's flash bank to a different size, enabling a bootloader to be installed at the start of flash.

However, the majority of targets still don't have linker scripts that make use of these defines.

multiplemonomials, 2024-11-23:
@timIdeaTech ^

JohnK1987, 2024-12-04:
Okay, what features should the MbedCE bootloader have?
ARM Mbed had two options 
1. managed bootloader - the bootloader is always at start of ROM and application/s follow after it. Maybe a app header included, not sure.
2. unmaneged bootloader - any other variant what does not meet the managed bootloader option

From my point of view the first one is very simple and enough for most of applications.
With latest modification on STM32H7 (memory banks, linker scripts) seems to be easy to do this. 
There are probably missing only some scripts:
* script for padding of empty place in rest of reserved sectors. Probably important for second point and for possibility of checksums.
* script for merging two binaries into one. Probably necessary for first default app deploy.

timIdeaTech, 2024-12-05:
@JohnK1987 I've  only done one mbed project with bootloader support and it used the managed option, and then only using minimal managed support. Basically I just used the `target.restrict_size` in the bootloader and `target.bootloader_img` in the applications. The mbed-os build system of course created all three combinations of binaries: bootloader, app and bootloader+app. 

It is/was an STM32L433 and STM32H743 dual processor design using a common custom targeted bootloader, where each micro had it's own Littlefilesystem2 NOR Flash. The app bin file and a bootloader trigger descriptor file (bin file name pointer and checksum) would be put in the root or predefined directory for a reboot/power-on update. This design didn't get out of development and by no means was complete. It required an embedded CLI in the app binary to load the files from any of a number of CLI enabled ports.

I only recount this to give you an idea where I got to and this was by no means where I wanted it or complete. I have only scratched the surface on MCUboot which I see has been brought up a few times and looks interesting. I'm all for keeping the overall design simple, but effective for most cases, and better yet an example project along the lines of the custom targets example. Some form of checksum and size checking on the app binary would be useful and maybe a mechanism for the the application to find the bootloader revision. I wish I could give this some gray matter time but got a little busy as of late and the mbed-os to mbed-ce port of the above project got put on hold.

JohnK1987, 2024-12-06:
Ok, that mean the mcuboot will be official bootdloader for MbedCE and no other solution is necessary, right? But that also mean someone will must update mbed port with every mcuboot release, I suppose.

lefebvresam, 2024-12-06:
I'm doing an attempt to create a project with mcu-bootloader as described on [mbed-mcuboot-demo](https://github.com/AGlass0fMilk/mbed-mcuboot-demo/blob/master/README.md)

But with the new toolchain (git submodules, cmake, etc..)

After creating a project, importing the libs and producing the necessary files, I try to do  sh build.sh -d and I get

```
-- Mbed: First CMake run detected, generating configs...
ERROR: Target specifies device_name HMC20 but this device is not
listed in /home/sam/git/mcuboot/mbed-os/tools/cmake/../../targets/cmsis_mcu_descriptions.json5.  Perhaps you need to use
the 'python -m mbed_tools.cli.main cmsis-mcu-descr fetch-missing' command to download
the missing MCU description?

More information may be available by using the command line option '-vv'.
CMake Error at mbed-os/tools/cmake/mbed_generate_configuration.cmake:123 (message):
  mbedtools configure failed! Cannot build this project.  Command was cd
  /home/sam/git/mcuboot/mbed-os/tools/cmake/../python &&
  /home/sam/git/mcuboot/mbed-os/venv/bin/python3.10 -m mbed_tools.cli.main -v
  configure -t GCC_ARM -m HMC20 --mbed-os-path
  /home/sam/git/mcuboot/mbed-os/tools/cmake/../..  --output-dir
  /home/sam/git/mcuboot/build --program-path /home/sam/git/mcuboot
Call Stack (most recent call first):
  mbed-os/tools/cmake/app.cmake:24 (include)
  CMakeLists.txt:7 (include)
```

I made to project public for testing purposes: 

`bitbucket.org/saleconix/mcuboot.git`

lefebvresam, 2024-12-06:
Adding this

```
set(MBED_APP_JSON5_PATH ${CMAKE_CURRENT_SOURCE_DIR}/mbed_app.json5)
set(CUSTOM_TARGETS_JSON5_PATH ${CMAKE_CURRENT_SOURCE_DIR}/custom_targets.json5)
```

or this

```
set(MBED_APP_JSON5_PATH mbed_app.json5)
set(CUSTOM_TARGETS_JSON5_PATH custom_targets.json5)
```

is not helping

JohnK1987, 2024-12-07:
@lefebvresam this seems like a separate issue around project configuration...
* I still see the "device_name": "HMC20" in your `custom_targets.json5` and also `memory_banks` section is still still missing -> https://github.com/mbed-ce/mbed-os/pull/394.
* According to log it seems like you compile `mbed-os` and` mbed-baremetal` totegether. This is probably caused by colistion between your CMakeLists and the one in mcuboot/boot/mbed/... You can use Jamie's fork - https://github.com/multiplemonomials/mcuboot/blob/464c79661da3409ebe4abb8c4bb09f28d5ec5907/boot/mbed/CMakeLists.txt
* not sure but most of these `requires` are set via CMake I think

lefebvresam, 2024-12-07:
First point was not comitted yet. If you mean this:

```
            "target.memory_bank_config": {
                "IROM1": {
                    "size": 0x20000
                }
            },
```

There was a complaint that IROM1 was not existing.

Point 2: Where is the clash then? 

```
target_link_libraries(${LIB_BOOTUTIL}
    PUBLIC
        mbed-mcuboot
        mbed-mbedtls
)
```

versus?

```
target_link_libraries(${APP_TARGET}
    mbed-baremetal
    mbed-storage
    mbed-mcuboot
)
```

Shall I make a separate ticket for this?

JohnK1987, 2024-12-07:
> Shall I make a separate ticket for this?

Yeah, please do it and transfer this content to the new issue. I will try to help..