mbed-ce/mbed-os #320: STM32WB55 16-bit DMA

chrissnow, 2024-08-08:
There seems to be a problem when using 16-bit SPI DMA, our use case is an ILI9341 SPI TFT display.

If we use 16 bit no DMA it works, if we use 8-bit DMA and byte swap it works, but if we use 16-bit DMA something is very wrong.
Some sections get drawn and then I think the display ends up with some display config being overwritten with graphic data.

My suspicion is that the complete callback is being called too early, potentially after half the bytes are written?

tx_length documentation specifically mentions it is in bytes, which is a bit odd, I would have expected this to be the number of 8/16/32 bit values to send? 

```
//As 16 bit transfer
// _spi.format(16, 0);		// switch to 16 bit transfer for data
// [[maybe_unused]] volatile int rc = _spi.transfer((uint16_t *)px_map, pixels*2 , nullptr,  0, callback(this, &LVGLDispILI9341::flush_ready));

//As 16 bit DMA transfer
_spi.set_dma_usage(DMA_USAGE_ALWAYS);
_spi.format(16, 0);		// switch to 16 bit transfer for data
[[maybe_unused]] volatile int rc = _spi.transfer((uint16_t *)px_map, pixels*2 , nullptr,  0, callback(this, &LVGLDispILI9341::flush_ready));

//As 8 bit DMA transfer
//_spi.set_dma_usage(DMA_USAGE_ALWAYS);
//lv_draw_sw_rgb565_swap(px_map,pixels);
//[[maybe_unused]] volatile int rc = _spi.transfer(px_map, pixels*2 , nullptr,  0, callback(this, &LVGLDispILI9341::flush_ready));

```

I haven't had time to hook up a logic analyser yet but will do at some point if needed.

chrissnow, 2024-08-09:
Unrelated to this issue as we are using SPI1 but this looks to be a typo
https://github.com/mbed-ce/mbed-os/blob/c1e3c86a4da19aa580a537d399da6381d97641c5/targets/TARGET_STM/TARGET_STM32WB/stm_dma_info.h#L38
It should be SPI2?

multiplemonomials, 2024-08-25:
Could you give #332 a try?

closed 2024-08-29 by chrissnow