mbed-ce/mbed-os #321: MBED STUDIO export to MICROVISION issue

DavidElmoRoss, 2024-08-09:
I have about 500 students a year that have been using KEIL STUDIO CLOUD with the FRDM-K64 board. My course has
been running since the early days of using the original CLOUD Compiler then exporting to MicroVision for debugging.
I have successfully been able to EXPORT a few of the over 100 programs that my students use from Github using MBED STUDIO to MicroVision. 
The only issue that I have is that from the simplest HELLO WORLD program to larger programs that use a number of devices that have custom MBED libraries, each of the MicroVision conversions generate 790Mb of storage requirements.
I am pretty sure that over 90% of this is due to compiling MBED OS with its 25,000 files.
Does anybody know of a way to reduce MICROVISION code size export to something more reasonable?

David Ross
Professor, School of Electronics and Mechanical Engineering Technology
Seneca Polytechnic, 1750 Finch Ave East North York (Toronto)
Ontario, Canada
M2J-2X5
Phone: 1-416-764-9313

JohnK1987, 2024-08-13:
Hello,

not sure if this is good place for this. It is not issue related to MbedCE it is more a question for [discussion](https://github.com/mbed-ce/mbed-os/discussions). Moreover maybe visit [Keil forum]( https://community.arm.com/support-forums/f/keil-forum) for support about their IDE and its settings.

In general Mbed project will take approx 1,5 GB of HDD space - approx 800MB mbed-os library and after first build additional 500MB of build folder + rest.
I understand when you have 100 examples multiply 1,5GB... that is a lot of wasted space. 
I know about two ways:
1. use [SymLink](https://github.com/JohnK1987/SymLink-for-MbedStudio-projects), I used this before MbedStudio adopt this method
2. create one project with 100 .cpp file inside of that project
3. ofc a solution with CMake - very similar to previous point

Method one will take down size from 100x800MB to only 800MB but 100x500MB for Build folders will persist. Rest methods will be more effective (only approx 1,5GB for everything) but it will take a time to manage it.

I wish some one will come with more elegant solution.

BR, Jan

DavidElmoRoss, 2024-08-14:
Thank you so much for your suggestions. Dave Ross

JohnK1987, 2024-12-01:
Closed because MbedStudio is not related to MbedCE.

closed 2024-12-01 by JohnK1987