mbed-ce/mbed-os #368: Need help on memory_banks.json for custom targets

wdx04, 2024-10-02:
Hi,
I have several custom boards which used to work, but failed to build after the Mbed Memory Bank Information PR was merged.

For example, when building an application for my STM32F407ZG board, I got the following error:

```
[build] memap.py: error: argument -m/--memory-banks-json: C:/Develop/Mbed/mbed-ce-app/build/STM32F407ZG_M144Z-Develop/memory_banks.json does not exist in the filesystem.
```
 
It seems that a custom target should have a memory_banks.json file, unless the memory bank information of its MCU is already included in the cmsis_mcu_descriptions.json5 file.

Questions:
1. Is there an existing example of 'memory_banks.json'?
2. Where should I put the 'memory_banks.json'? It seems to be not good to put it directly into a 'build' folder.
3. Currently, startup file and linker script for STM32F407xG do exist in the Mbed targets source tree, but the memory bank information of STM32F407xG is not included in the cmsis_mcu_descriptions.json5 file. This is also true for many other STM32 targets, where there is no supported board for that MCU family. This is not very convenient for custom board users. Does it make sense to add the memory bank information for those MCU families to the cmsis_mcu_descriptions.json5 file?

Thanks.

multiplemonomials, 2024-10-03:
Ah, this should not be causing a build error like this, that is a bug and I will take a look 

You should be able to supply memory banks information directly in custom_targets.json via a `memory_banks` attribute.  There's an example of this in the [memory banks wiki page](https://github.com/mbed-ce/mbed-os/wiki/Mbed-Memory-Bank-Information) under the "Adding Flash Information to RP2040" section

wdx04, 2024-10-04:
I added a memory_banks section in the custom_targets.json5, then I got another error:

```
[cmake] -- Mbed: custom_targets.json modified, regenerating configs...
[cmake] ERROR: Target specifies device_name STM32F407ZGTx but this device is not
[cmake] listed in C:\Develop\Mbed\mbed-ce-app\mbed-os\tools\cmake\..\..\targets\cmsis_mcu_descriptions.json5.  Perhaps you need to use
[cmake] the 'python -m mbed_tools.cli.main cmsis-mcu-descr fetch-missing' command to download
[cmake] the missing MCU description?
```

multiplemonomials, 2024-10-04:
Ah, remove the "device_name" property from your custom_targets.json

wdx04, 2024-10-04:
> Ah, remove the "device_name" property from your custom_targets.json

Yeah, removing the "device_name" property fixed the problem. Thanks.

closed 2024-10-04 by wdx04

wdx04, 2024-10-07:
It would be nice to have a "memory_banks_add" attribute to add board-specific external memories (SDRAM, SRAM, PSRAM, QSPI Flash, etc.) besides the memory banks provided by the microcontroller.