mbed-ce/mbed-os #370: Relocate MCUboot enabled application

ccli8, 2024-10-08:
I am checking [Mbed Memory Bank Information](https://github.com/mbed-ce/mbed-os/wiki/Mbed-Memory-Bank-Information) for MCUboot enabled application. This needs relocating application to non-default address. In Mbed CLI 1, this can achieve via `target.mbed_app_start/size`. In Mbed CE, this is achieved by overriding the `memory_bank_config` section and updating linker script to check `MBED_CONFIGURED_ROM_BANK_IROM1_START/SIZE` instead of `MBED_APP_START/SIZE`.

Per my test, overriding `memory_bank_config` section can only be done in `targets.json` or `custom_targets.json`, and cannot in `mbed_app.json`. In most use cases, it is more friendly to relocate the application without creating one new target. Is is officially supported to override the `memory_bank_config` section in `mbed_app.json` or the like to get to this purpose?

multiplemonomials, 2024-10-08:
Hmm, I think it will work to create a `memory_bank_config` section directly in mbed_app.json.  I feel like I did check that but not 100% sure.

Though that does bring up another point which is that if people are going to configure memory banks in mbed_app.json, there should probably be a way to configure it by target.  Something like a `target_memory_bank_config` option in mbed_app.json.

ccli8, 2024-10-11:
According to tests below, `memory_bank_config` is supported only in `targets.json` or `custom_targets.json`, but not in `mbed_app.json`. It should be supported to enable change of application base address just in `mbed_app.json`

Test `memory_bank_config` on `custom_targets.json`: OK
Test `memory_bank_config` on `mbed_app.json`: Failed
```
{
    "target_overrides": {
        "SOME_TARGET": {
            "memory_bank_config": {
                "IROM1": {
                    "start": 0x0,
                    "size": 0x20000
                }
            },
        },
    },
}
```
> WARNING: You are attempting to override an undefined config parameter `app.memory_bank_config`.
It is an error to override an undefined configuration parameter. Please check your target_overrides are correct.
The parameter `app.memory_bank_config` will not be added to the configuration.

Test `target.memory_bank_config` on `mbed_app.json`: Failed
```
{
    "target_overrides": {
        "SOME_TARGET": {
            "target.memory_bank_config": {
                "IROM1": {
                    "start": 0x0,
                    "size": 0x20000
                }
            },
        },
    },
}
```
> WARNING: You are attempting to override an undefined config parameter `target.memory_bank_config`.
It is an error to override an undefined configuration parameter. Please check your target_overrides are correct.
The parameter `target.memory_bank_config` will not be added to the configuration.

multiplemonomials, 2024-10-11:
Makes sense, looks like this is a feature we need to add

PR #381 (2024-11-18): Support override of memory_bank_config in mbed_app.json
### Summary of changes 

`memory_bank_config` can be defined in targets.json or custom_targets.json per-target, but cannot by application. This enables this override by adding `target.memory_bank_config` config in `target_overrides` section of mbed_app.json as usual.

Fix #370.

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [X] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Func [... cut]
closed 2024-11-18 by multiplemonomials