mbed-ce/mbed-os #383: Debug configuration for mutilple targets adds wrong overrideLaunchCommands

JojoS62, 2024-11-18:
The generated debug configuration launch.json contains a wrong entry for overrideLaunchCommands in the 2nd target (same as for 1st target), causing to load a wrong program:

```json5
 {
    "configurations": [
	// Debug launch for target Hello.
	{
	    "type": "cortex-debug",
	    "name": "Debug Hello FK743M5_XIH6 debug",
	    "executable": "/home/jojo/projects/mbed/mbed-ce-H7/build/FK743M5_XIH6_STM32CUBE-Debug/app_hello/Hello.elf",
	    "cwd": "${workspaceRoot}",
	    "gdbPath": "/opt/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gdb",
		"objdumpPath": "/opt/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-objdump",
		"servertype": "external",
		"gdbTarget": "127.0.0.1:23331",
		"request": "launch",
		"preLaunchTask": "Build Hello and start GDB server",
		// Override the command sequences used by VS Code to be correct for this GDB server
		"overrideLaunchCommands": ["monitor reset", "load \"/home/jojo/projects/mbed/mbed-ce-H7/build/FK743M5_XIH6_STM32CUBE-Debug/app_hello/Hello.hex\"", "tbreak main", "monitor reset"],
		"overrideRestartCommands": ["monitor reset"],
	},
	// Debug launch for target test_display.
	{
	    "type": "cortex-debug",
	    "name": "Debug test_display FK743M5_XIH6 debug",
	    "executable": "/home/jojo/projects/mbed/mbed-ce-H7/build/FK743M5_XIH6_STM32CUBE-Debug/app_test_display/test_display.elf",
	    "cwd": "${workspaceRoot}",
	    "gdbPath": "/opt/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gdb",
		"objdumpPath": "/opt/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-objdump",
		"servertype": "external",
		"gdbTarget": "127.0.0.1:23331",
		"request": "launch",
		"preLaunchTask": "Build test_display and start GDB server",
		// Override the command sequences used by VS Code to be correct for this GDB server
		"overrideLaunchCommands": ["monitor reset", "load \"/home/jojo/projects/mbed/mbed-ce-H7/build/FK743M5_XIH6_STM32CUBE-Debug/app_hello/Hello.hex\"", "tbreak main", "monitor reset"],
		"overrideRestartCommands": ["monitor reset"],
	},
    ]
}
```

JojoS62, 2024-11-18:
it llooks like mbed_adjust_upload_debug_commands() tries to find 'load' in the already processed MBED_UPLOAD_LAUNCH_COMMANDS. For the next target, there need to be a new clean entry where load can be replaced.
But I don't understand the flow completely.

When I disable the validation of the list FIND, I can see the debug message:
```cmake
function(mbed_adjust_upload_debug_commands target)
    # MBED_UPLOAD_LAUNCH_COMMANDS defined?
    if(NOT DEFINED MBED_UPLOAD_LAUNCH_COMMANDS)
        return()
    endif()

    # GDB load command in MBED_UPLOAD_LAUNCH_COMMANDS?
    list(FIND MBED_UPLOAD_LAUNCH_COMMANDS "load" LOAD_INDEX)
    # if(${LOAD_INDEX} LESS "0")
    #     return()
    # endif()
    message(STATUS "load index: ${LOAD_INDEX}  ${target}  ${MBED_UPLOAD_LAUNCH_COMMANDS}")
```

Output:
```
[cmake] -- load index: 1  Hello  monitor reset;load;tbreak main;monitor reset
[cmake] -- load index: -1  test_display  monitor reset;load \"/home/jojo/projects/mbed/mbed-ce-H7/build/FK743M5_XIH6_STM32CUBE-Debug/app_hello/$<TARGET_FILE_BASE_NAME:Hello>.hex\";tbreak main;monitor reset
```

multiplemonomials, 2024-11-21:
@ccli8 this looks like it's related to your recent changes? ^

ccli8, 2024-11-21:
I will look into it.

PR #391 (2024-11-21): Upload: Fix mbed_adjust_upload_debug_commands for multiple targets
### Summary of changes 

The purpose of `mbed_adjust_upload_debug_commands()` is to adjust first version of `MBED_UPLOAD_LAUNCH_COMMANDS`. For multiple executalbe cmake targets, `MBED_UPLOAD_LAUNCH_COMMANDS` becomes not fresh and may cause error. To overcome this, one new cmake cache variable `MBED_UPLOAD_LAUNCH_COMMANDS_BAK` is introduced to save first version of `MBED_UPLOAD_LAUNCH_COMMANDS`.

Fix #383

----------------------------------------------------------------------------------------------------------------
### Pull re [... cut]
closed 2024-11-21 by multiplemonomials