mbed-ce/mbed-os #393: Adding header references to interface library

lefebvresam, 2024-11-22:
I have a folder where I gather all my specific files for my target:

[image]

The CMakeLists.txt looks like:

```
# Copyright (c) 2023 ARM Limited. All rights reserved.
# SPDX-License-Identifier: Apache-2.0

add_library(mbed-hmc20 INTERFACE)

target_sources(mbed-hmc20 INTERFACE
    PeripheralPins.c
)

target_include_directories(mbed-hmc20 INTERFACE
    .
)

target_link_libraries(mbed-hmc20 INTERFACE 
    mbed-stm32u575xg
)
```

PeripheralPins.c has:

```
#include "PeripheralPins.h"
#include "mbed_toolchain.h"
#include "PeripheralPinMaps.h"
```

In PinNames.h I define the pins and a bit further I use a MACRO:

```
#include "cmsis.h"
#include "PinNamesTypes.h"
...
    // STDIO for console print
#ifdef MBED_CONF_TARGET_STDIO_UART_TX
    CONSOLE_TX = MBED_CONF_TARGET_STDIO_UART_TX,
#else
    #ifdef STDOUT_USB
    CONSOLE_TX = PA_0,
    #else
    CONSOLE_TX = PB_6,
    #endif
#endif
#ifdef MBED_CONF_TARGET_STDIO_UART_RX
    CONSOLE_RX = MBED_CONF_TARGET_STDIO_UART_RX,
#else
    #ifdef STDOUT_USB
    CONSOLE_RX = PA_1,
    #else
    CONSOLE_RX = PB_7,
    #endif
#endif
```

But I want STDOUT_USB to define in my main.h file in my source code, where other settings are.

Normally I could do something like:

```
target_include_directories(mbed-hmc20 PRIVATE 
    "${APP_SOURCE_PATH}"
)
```

But this is not working because mbed-hmc20 is an interface library, just to let 'mbed-stm32u575xg' know that there are some extra files to compile.

If I put #include "main.h" in PinNames.h is is not found and if I put app_source_path I get:

```
CMake Error at HMC20/CMakeLists.txt:14 (target_include_directories):
  target_include_directories may only set INTERFACE properties on INTERFACE
  targets
```

and with INTERFACE I get:

```
/home/sam/git/hmipanel/HMC20/./PinNames.h:36:10: fatal error: main.h: No such file or directory
   36 | #include "main.h"
```

How can I avoid having to put my setting MACRO's in PinNames.h?

multiplemonomials, 2024-11-23:
Hmm... assuming main.h is in your project root directory, it might be easiest to just include it by relative path in PinNames.h:
```cpp
#include "../main.h"
```

Though this should also work:
```cmake
target_include_directories(mbed-hmc20 INTERFACE
    "${APP_SOURCE_PATH}"
)
```
If it's not working, perhaps the APP_SOURCE_PATH variable is not defined at this point in the build script?

Also, keep in mind that doing it this way means that every Mbed source file will include main.h, which may or not be what you want.  main.h must be includable from C, and any change to main.h will rebuild the entire project.

Last but not least, for the future, since this is not an actual issue with Mbed CE, this kind of thing would be better as a discussion than an issue.  OK for now, just remember for next time!

Hope this is helpful, let me know if that works!

multiplemonomials, 2024-11-26:
That sounds more or less right, but I'm surprised you need an empty PeripheralPins.c.  Not having that .c file should still let the build work, as long as the actual peripheral pins arrays that are supposed to go there are declared somewhere else.  If that's not it's behaving, please post your project in a discussion and we will take a look!

lefebvresam, 2024-11-28:
It's not working after a full compile I get a message for each part that is using PinNames.h:

```
In file included from /home/sam/git/hmipanel/mbed-os/targets/TARGET_STM/TARGET_STM32U5/./objects.h:22,
                 from /home/sam/git/hmipanel/mbed-os/targets/TARGET_STM/./device.h:38,
                 from /home/sam/git/hmipanel/mbed-os/hal/include/hal/ticker_api.h:25,
                 from /home/sam/git/hmipanel/mbed-os/drivers/./include/drivers/TimerEvent.h:20,
                 from /home/sam/git/hmipanel/mbed-os/platform/include/platform/internal/SysTimer.h:22,
                 from /home/sam/git/hmipanel/mbed-os/platform/include/platform/internal/mbed_os_timer.h:21,
                 from /home/sam/git/hmipanel/mbed-os/rtos/./include/rtos/Kernel.h:31,
                 from /home/sam/git/hmipanel/mbed-os/rtos/./include/rtos/Mutex.h:30,
                 from /home/sam/git/hmipanel/mbed-os/drivers/usb/include/usb/internal/AsyncOp.h:21,
                 from /home/sam/git/hmipanel/mbed-os/drivers/usb/source/OperationListBase.cpp:19:
/home/sam/git/hmipanel/HMC20/./PinNames.h:25:10: fatal error: main.h: No such file or directory
   25 | #include "main.h"
```

And it's a whole series ...

lefebvresam, 2024-11-28:
It's working now. Indeed I have to put APP_SOURCE_PATH before the add_subdirectory(HMC20), and not in the subdirectory source:

```
set(APP_SOURCE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/source CACHE INTERNAL "")

# Include folder with the custom targets.
add_subdirectory(nlohmann_json)
add_subdirectory(ra8875)
add_subdirectory(HMC20)
add_subdirectory(mbed-os)
add_subdirectory(source)

mbed_finalize_build()
```

closed 2024-11-28 by lefebvresam