mbed-ce/mbed-os #395: USBSerial does not redirect stdout correctly to USB on NUCLEO_WB55RG

ChristophKarlHeck, 2024-11-24:
Hi,
Steps to Reproduce
1. Configure mbed_app.json with the following settings:
```
{
    "target_overrides": {
        "*": {
            "platform.stdio-driver": "USBSerial",
            "platform.stdio-baud-rate": 115200,
            "platform.stdio-convert-newlines": true
        },
        "NUCLEO_WB55RG": {
            "target.device_has_add": ["USBDEVICE"]
        }
    }
}
```
2. Build and flash the program to the NUCLEO_WB55RG.
3. Connect the board's USB port (CN1) to a Raspberry Pi or PC and check for `stdout` messages.
4. Observe that output still prints on CN15 (UART), not CN1 (USB).

**Expected Behavior:**
* printf should output to the USB port (CN1) via USBSerial.
* Newline conversion should function as expected.

**Observed Behavior**
* printf continues to output on CN15.
* When manually redirecting to USBSerial using mbed_override_console, no output appears or line breaks are incorrect.

**Temporary Solution**
Explicitly override the console using the following code in main.cpp:
```
#include "USBSerial.h"

USBSerial serial;

FileHandle *mbed::mbed_override_console(int) {
    return &serial;
}
```
This ensures that stdout is routed to USBSerial. However, this should not be necessary when "platform.stdio-driver": "USBSerial" is specified in mbed_app.json.

**Additional Notes**
* Adding debugging output in `USBSerial::_putc confirmed that the method is not invoked when using `printf.
* The configuration in `mbed_config.cmake includes the expected macros, such as:
```
MBED_CONF_PLATFORM_STDIO_DRIVER=USBSerial
MBED_CONF_PLATFORM_STDIO_CONVERT_NEWLINES=1

```
* However, these do not seem to take effect. Therefore I had to ajust the `_putc` implementation in USBSerial.cpp to make sure that I have a \r\n at the end of every print statement
```
int USBSerial::_putc(int c) {
    // For debugging: send directly to another serial port if available
    // or use a GPIO-based debug mechanism
    // log_debug_char(c); // Example: Implement a lightweight logging function

    if (c == '\n') {
        uint8_t carriage_return = '\r';
        send(&carriage_return, 1);
    }
    if (send((uint8_t *)&c, 1)) {
        return c;
    } else {
        return -1;
    }
}
```
Does anybody else faces the same issue?

Cheers,
Christoph

JohnK1987, 2024-11-24:
Hello,

Where did you get the info `"platform.stdio-driver": "USBSerial",` work like this?
When we look to [Platform/mbed_lib.json]( https://github.com/mbed-ce/mbed-os/blob/master/platform/mbed_lib.json ) then there is not a parameter of this name.

As I remember you are able to configure the studio over configuration system of .json files just to different UART. So your Temporary Solution is only one correct solution for this.

ChristophKarlHeck, 2024-11-24:
Hi Jan,
Thank you for the fast help. I will have a look tomorrow morning since I am still facing two issues:
1. Hard Faults are not transmitted
2. I often get his error: https://github.com/ARMmbed/mbed-os/issues/10339 
When I add `printf` statement to my main function,
So far, I have yet to recognize the pattern behind that error.

Cheers,
Christoph

JohnK1987, 2024-11-24:
Moreover, your target already has USBDEVICE in default. That change nothing, just saying.
https://github.com/mbed-ce/mbed-os/blob/f3da3a1cbbbed16e7ca84d8b45f9f75b932ea9cb/targets/targets.json5#L5044-L5058

With USBSerial you also need manual control of DTR signal or an app which is able to automatically like putty.

Hmm, you mean just similar fault, right. This one seems to be specific.

BR, Jan

ChristophKarlHeck, 2024-11-25:
Okay,
I found a more amicable solution:
 ```
 {
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,     
            "platform.stdio-convert-tty-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "target.console-uart": true                 // use of default pints PB_&, PB_7
            
        },
        "NUCLEO_WB55RG": {
            "target.features_add": ["BLE"],              // Enable BLE features
        }
    }
}
```
I am now getting the following Hard Faults displayed via UART:
```
'++ MbedOS Fault Handler ++'
'FaultType: HardFault'
'Context:'
'R   0: 200006B8'
'R   1: 00000004'
'R   2: 00000004'
'R   3: 77726F66'
'R   4: 20002BC8'
'R   5: 00000001'
'R   6: 00000000'
'R   7: 20000C58'
'R   8: 2000042C'
'R   9: 20002F1C'
'R  10: 0000000C'
'R  11: 20000420'
'R  12: 00000000'
'SP   : 20002AB8'
'LR   : 20002E68'
'PC   : 0801055E'
'xPSR : 61070000'
'PSP  : 20002A50'
'MSP  : 2002FFC0'
'CPUID: 410FC241'
'HFSR : 40000000'
'MMFSR: 00000000'
'BFSR : 00000082'
'UFSR : 00000000'
'DFSR : 00000000'
'AFSR : 00000000'
'BFAR : 77726F72'
'Mode : Thread'
'Priv : Privileged'
'Stack: PSP'
'-- MbedOS Fault Handler --'
'++ MbedOS Error Info ++'
'Error Status: 0x80FF013D Code: 317 Module: 255'
'Error Message: Fault exception'
'Location: 0x801055E'
'Error Value: 0x20001240'
'Current Thread: main Id: 0x20001C70 Entry: 0x8009ED5 StackSize: 0x1000 StackMem: 0x20001CB8 SP: 0x20002AB8'
'For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NUCLEO_WB55RG'
'-- MbedOS Error Info --'
```
What causes this HardFault seems very strange to me. When I comment out this `print` statement, it works:
```
printf("Counter, %d\n", counter);
```
To give you the entire context:
Works perfectly:
```
[... 232 lines cut ...]
			if(CLASSIFICATION){
				
				// Convert received bytes to floats
				std::vector<float> inputs = get_analog_inputs(inputs_as_bytes, DATABITS, VREF, GAIN);
				// Execute Model with received inputs
				executor.setModelInput(method, inputs);
				executor.executeModel(method, method_name);
				classification_result = executor.getModelOutput(method);
			}

			printf("Counter, %d\n", counter);
		
			counter = counter + 1;

		}

		// Needed to avoid immediate resource exhaustion
		thread_sleep_for(DOWNSAMPLING_RATE); // ms

	}

	return 0;
}
```
Do you have any ideas about that issue?

JohnK1987, 2024-11-25:
The following settings you did is also default and it is not for USB but furt UART what is connected to onboard ST-Link which provides VCP (Virtual Com Port).
```
"target.console-uart": true 
```
The following line is also for nothing when you use UART.
```
#include "USBSerial.h" // Uncomment if needed for debugging
```

About the printf. I do not see any obvious reason. Maybe try to place a welcome printf at the beginning of main and check the behavior. Eventually you can try to debugging.

BR, Jan

ChristophKarlHeck, 2024-11-25:
Okay, I got it! 
Since I need to understand more about debugging issues of RTOS, I would appreciate to get some tips from you:
The Video shows my current procedure: https://drive.google.com/file/d/1F_vAuMUfX1EZW8KLK95YMvZC-KQznu4V/view?usp=drive_link
Do you see things I could improve or have other helpful tips?
BR, Christoph

ChristophKarlHeck, 2024-11-25:
I solved it. However, I am unsatisfied since I have yet to figure out the issue.
The Hard Fault occurred whenever I placed an additional print statement with the argument in the while loop. I was running the debugger and didn't figure out the issue since it didn't occur while debugging. I assumed a timing problem and added the `thread_sleep_for(10)` statement at the correct position, but in the end, I didn't figure out the exact reason. I think it's CPU exhaustion or something else. How would you continue to specify the cause further?
This is basically the solution so far:
```
void ModelExecutor::executeModel(
    Result<torch::executor::Method>& method,
    const char* method_name)
{
    ET_LOG(Info, "Starting the model execution...");
    thread_sleep_for(10); // Solution
    Error status = method->execute();
    if (status != Error::Ok) {
        ET_LOG(
            Info,
            "Execution of method %s failed with status 0x%" PRIx32,
            method_name,
            status);
 } else {
        ET_LOG(Info, "Model executed successfully.");
 }
```

JohnK1987, 2024-11-25:
Maybe the loop needs few attempts before it will crash. Try to press continue without stepping until crash occurs.

I do not understand what mean this 
```
// Function called in thread "reading_data_thread"
void get_input_model_values_from_adc(unsigned int* model_input_size){

	AD7124 adc(SPI_FREQUENCY);
	adc.init(true, true); // activate both channels
	adc.read_voltage_from_both_channels(DOWNSAMPLING_RATE,*model_input_size);
}

	//Start reading data from ADC Thread
	unsigned int n = 4;
	reading_data_thread.start(callback(get_input_model_values_from_adc, &n));

```

This doesn't make sense. You just call ADC constructor, init and one time read, nothing more. For that you do not need thread or this function is wrong.

ChristophKarlHeck, 2024-11-26:
I appreciate your help!
The `read_voltage_from_both_channels(DOWNSAMPLING_RATE,*model_input_size)` contains a while(true) loop that reads the values via SPI. Does that make sense for you?
The repo is now available here:
https://github.com/ChristophKarlHeck/mbed-torch-fusion-os

JohnK1987, 2024-11-26:
Ah, ok that make sense now. Thank you for explanation :)