mbed-ce/mbed-os #401: Bootloader mcuboot with custom device

lefebvresam, 2024-12-07:
I'm doing an attempt to create a project with mcu-bootloader as described on [mbed-mcuboot-demo](https://github.com/AGlass0fMilk/mbed-mcuboot-demo/blob/master/README.md)

But with the new toolchain (git submodules, cmake, etc..)

After creating a project, importing the libs and producing the necessary files, I try to do  sh build.sh -d and I get

```
-- Mbed: First CMake run detected, generating configs...
ERROR: Target specifies device_name HMC20 but this device is not
listed in /home/sam/git/mcuboot/mbed-os/tools/cmake/../../targets/cmsis_mcu_descriptions.json5.  Perhaps you need to use
the 'python -m mbed_tools.cli.main cmsis-mcu-descr fetch-missing' command to download
the missing MCU description?

More information may be available by using the command line option '-vv'.
CMake Error at mbed-os/tools/cmake/mbed_generate_configuration.cmake:123 (message):
  mbedtools configure failed! Cannot build this project.  Command was cd
  /home/sam/git/mcuboot/mbed-os/tools/cmake/../python &&
  /home/sam/git/mcuboot/mbed-os/venv/bin/python3.10 -m mbed_tools.cli.main -v
  configure -t GCC_ARM -m HMC20 --mbed-os-path
  /home/sam/git/mcuboot/mbed-os/tools/cmake/../..  --output-dir
  /home/sam/git/mcuboot/build --program-path /home/sam/git/mcuboot
Call Stack (most recent call first):
  mbed-os/tools/cmake/app.cmake:24 (include)
  CMakeLists.txt:7 (include)
```

'missing MCU description' and I have a custom_targets.json5

I made to project temporary public for testing purposes: 

`bitbucket.org/saleconix/mcuboot.git`

Adding this

```
set(MBED_APP_JSON_PATH mbed_app.json5)
set(CUSTOM_TARGETS_JSON_PATH custom_targets.json5)
```

or this

```
set(MBED_APP_JSON_PATH ${CMAKE_CURRENT_SOURCE_DIR}/mbed_app.json5)
set(CUSTOM_TARGETS_JSON_PATH ${CMAKE_CURRENT_SOURCE_DIR}/custom_targets.json5)
```

is not helping

lefebvresam, 2024-12-07:
I had to remove "device_name": "HMC20" in custom targets. Strange that my original application never complained about this.

But now I have something like a loop:

```
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:53: multiple definition of `mbed_error_in_progress'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:53: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_error_initialize':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:260: multiple definition of `mbed_error_initialize'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:260: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_get_first_error':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:267: multiple definition of `mbed_get_first_error'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:267: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_get_last_error':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:274: multiple definition of `mbed_get_last_error'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:274: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_get_error_count':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:281: multiple definition of `mbed_get_error_count'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:281: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `core_util_atomic_load_bool':
/home/sam/git/mcuboot/mbed-os/platform/include/platform/internal/mbed_atomic_impl.h:728: multiple definition of `mbed_get_error_in_progress'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/include/platform/internal/mbed_atomic_impl.h:728: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_warning':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:291: multiple definition of `mbed_warning'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:291: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_set_error_hook':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:351: multiple definition of `mbed_set_error_hook'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:351: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_reset_reboot_error_info':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:360: multiple definition of `mbed_reset_reboot_error_info'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:360: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_reset_reboot_count':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:387: multiple definition of `mbed_reset_reboot_count'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:387: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_get_reboot_error_info':
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:404: multiple definition of `mbed_get_reboot_error_info'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:404: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_error.c.obj: in function `mbed_get_first_error_info':
[... cut ...]
/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:456: multiple definition of `mbed_clear_all_errors'; mbed-os/CMakeFiles/mbed-baremetal.dir/platform/source/mbed_error.c.obj:/home/sam/git/mcuboot/mbed-os/platform/source/mbed_error.c:456: first defined here
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: mbed-os/CMakeFiles/mbed-os.dir/platform/source/mbed_mem_trace.cpp.obj: in function `SingletonPtr<rtos::Mutex>::get() const':
```

I was also wondering if I need:

`"requires": ["bare-metal", "mbedtls", "mcuboot", "flashiap-block-device", "spif-driver", "qspif", "mbed-trace", "sd"],`

lefebvresam, 2024-12-07:
https://github.com/JohnK1987

@lefebvresam this seems like a separate issue around project configuration...
* I still see the "device_name": "HMC20" in your `custom_targets.json5` and also `memory_banks` section is still missing -> https://github.com/mbed-ce/mbed-os/pull/394.
* According to log it seems like you compile `mbed-os` and` mbed-baremetal` totegether. This is probably caused by colistion between your CMakeLists and the one in mcuboot/boot/mbed/... 
  You can use Jamie's fork for an inspiration - https://github.com/multiplemonomials/mcuboot/blob/464c79661da3409ebe4abb8c4bb09f28d5ec5907/boot/mbed/CMakeLists.txt
* not sure but most of these `requires` are set via CMake I think

lefebvresam, 2024-12-07:
First point was not comitted yet. If you mean this:

```
            "target.memory_bank_config": {
                "IROM1": {
                    "size": 0x20000
                }
            },
```

There was a complaint that IROM1 was not existing.

Point 2: Where is the clash then? 

```
target_link_libraries(${LIB_BOOTUTIL}
    PUBLIC
        mbed-mcuboot
        mbed-mbedtls
)
```

versus?

```
target_link_libraries(${APP_TARGET}
    mbed-baremetal
    mbed-storage
    mbed-mcuboot
)
```

lefebvresam, 2024-12-07:
I just pushed my test changes but it is still not compiling. I always have the issue with 'first defined here'

JohnK1987, 2024-12-07:
Thank you!

1. We can say your override of IROM1 via mbed_app.json is not working because there is nothing for override.
   Memory_banks are working in default with STM32H7 targets because there is everything prepared, but rest families are not ready for this. Linker script of STM32U5 is still use old definitions like APP_START and SIZE.  So you should probably use it by old way. That mean you have to use names what will override  [these values](https://github.com/mbed-ce/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32U5/TARGET_STM32U575xG/cmsis_nvic.h) which are used in [the linker of STM32U575xG](https://github.com/mbed-ce/mbed-os/blob/c3f9a8ac07df6f92ddd98af448d4fcf9ae20604f/targets/TARGET_STM/TARGET_STM32U5/TARGET_STM32U575xG/TOOLCHAIN_GCC_ARM/stm32u575xg.ld#L21-L42).
   Or you can update it by your self (I am not sure what all exactly need to be done at this moment, for this is maybe better to ask directly Jamie.) for STM32U5 familly based on the [STM32H7 modifications](https://github.com/mbed-ce/mbed-os/blob/c3f9a8ac07df6f92ddd98af448d4fcf9ae20604f/targets/TARGET_STM/TARGET_STM32H7/linker_scripts/STM32H743_STM32H72x_FAMILIES/STM32H743_H72x.ld#L23-L42). 
 2. I have no info about progress but how Jamie wrote he already started to do something with MCUBOOT, so I believe this is good start - https://github.com/mbed-ce/mbed-mcuboot-bootloader
 
3. 
   > Point 2: Where is the clash then?
   > 
   > ```
   > target_link_libraries(${LIB_BOOTUTIL}
   >     PUBLIC
   >         mbed-mcuboot
   >         mbed-mbedtls
   > )
   > ```
   > 
   > versus?
   > 
   > ```
   > target_link_libraries(${APP_TARGET}
   >     mbed-baremetal
   >     mbed-storage
   >     mbed-mcuboot
   > )
   > ```
   I did not write about top lvl CMakeList of your project but the one of your imported mcuboot library - https://github.com/mcu-tools/mcuboot/blob/b778ad9e163da2212677d19ad6788c63127881a1/boot/mbed/CMakeLists.txt#L38-L48
  In Jamie's fork of mcuboot you can see a modification - https://github.com/multiplemonomials/mcuboot/blob/464c79661da3409ebe4abb8c4bb09f28d5ec5907/boot/mbed/CMakeLists.txt#L32 

BR, Jan

JohnK1987, 2024-12-07:
From my point of view you should have mbed-os or mbed-baremetal (never both at once) just one time per app.
However libraries which use some dependencies from mbed-os need an asaccess to mbed-os library, then you need link the library to mbed-core-flags or mbed-rtos-flags if the library needs to use RTOS features.

> However I think I best change the dependency to Jamie's version.

Agreed.

lefebvresam, 2024-12-07:
I updated the project with the new lib and the branch.

Why this IROM1 setting is needed? 
Do I have to reduce this setting to 0x20000 by overwriting it somewhere in the config?

#define MBED_ROM_SIZE  0x100000 // 1MB

A macro at custom targets?
@multiplemonomials  can you elaborate on this?

lefebvresam, 2024-12-07:
What I also do not really understand is if you follow the procedure on [AGlass0fMilk](https://github.com/AGlass0fMilk/mbed-mcuboot-demo/blob/master/README.md) you get a file with rsa_pub_key[], and in the example I see enc_priv_key[].

multiplemonomials, 2024-12-07:
Sorry I really just started on the mcuboot stuff, I haven't really done anything other than update the build system yet.  You should definitely use my fork of mcuboot to fix the link errors, but I still need to actually get a basic example working

multiplemonomials, 2024-12-07:
In order to make things work properly with the bootloader, you will need to use `memory_bank_config` to configure the ROM base address.  This also requires that your target's linker script support memory bank configuration, which a lot of them currently don't

lefebvresam, 2024-12-07:
I use a custom target inherited from MCU_STM32U575xG. Are there some examples how to change/adapt the linker script?

multiplemonomials, 2024-12-07:
Actually, yeah!  Been working on an update for the K64F here: https://github.com/mbed-ce/mbed-os/tree/dev/k64f-linker-script-refactor

lefebvresam, 2024-12-07:
Is it the purpose that this:

[image]

Is in line with this:

[image]

Or where do you find the right details specific for the controller?

multiplemonomials, 2024-12-08:
Yeah, you will find this info in the memory map section of the datasheet, or (apparently) in CubeIDE

lefebvresam, 2024-12-08:
I see it's essentially renaming stuff. In my case there is no differentiation between RAM1, 2 or 3. I'm I right that the only purpose is to let the compiler believe that the size of the flash is only 0x20000 in order to have the code fit in that part before adding different hex files to create the complete image? I have no clue what .text and .bss and others exaclty mean.

JohnK1987, 2024-12-08:
These definitions allow you to change memory settings according to your requirements via configuration without direct intervention to the linker script.

It should working probably like that
1. prepare custom_targets.json
	In case of custom target you should fill necessary information to `custom_targets_json` file as default target definition. For example your one
	<details><summary>For content click here</summary>

	```
		"HMC20": {
			"inherits": [
				"MCU_STM32U575xG"
			],
			"components_add" : ["SD", "SPIF"],
			"overrides": {
				"clock_source" : "USE_PLL_HSE_XTAL",
				"lse_available" : "1",
				"lse_drive_load_level" : "RCC_LSEDRIVE_HIGH"
			},
			"macros_add": [
				"HSE_VALUE=10000000UL"
			],
			"device_has_remove": [
				"ANALOGOUT",...
			],
			"device_name": "STM32U575VGTx"
		}
	```
	</details>

	To `device_name` we will fill almost exact MCU name what we have. In our case STM32U575VG (not sure which one you really have). Then we will see error what you had
	```
	-- Mbed: First CMake run detected, generating configs...
	ERROR: Target specifies device_name HMC20 but this device is not
	listed in /home/sam/git/mcuboot/mbed-os/tools/cmake/../../targets/cmsis_mcu_descriptions.json5.  Perhaps you need to use
	the 'python -m mbed_tools.cli.main cmsis-mcu-descr fetch-missing' command to download
	the missing MCU description?
	```
	Then we go to '...project_name\mbed-os\tools\python' and call 'python -m mbed_tools.cli.main cmsis-mcu-descr fetch-missing'. 
	If our device name is not correct then we will see something like this
	```
	RuntimeError: MCU STM32U575VG is not present in the CMSIS MCU index (C:\Users\user\AppData\Local\cmsis-pack-manager\cmsis-pack-manager\index.json).  Maybe wrong part number, or this MCU simply doesn't exist in the CMSIS index and has to be added manually?
	```
	So we check the name one more time or visit mentioned file index.json and look for best match. In our case "STM32U575VGTx". Then we update `device_name` parametr and perform 'fetch-missing' (from above) one more time. Then we update `device_name` parameter and perform 'fetch-missing' (from above) one more time. Then we see output in console like 
	<details><summary>For console output click here</summary>

	```
	PS ...project_name\mbed-os\tools\python> python -m mbed_tools.cli.main cmsis-mcu-descr fetch-missing
	INFO: Custom_targets file detected -  ...project_name\custom_targets\custom_targets.json5
	INFO: Scanning targets.json5 for used MCU names...
	INFO: Scanning custom_targets.json/json5. for used MCU names...
	INFO: Scanning cmsis_mcu_descriptions.json5 file for missing MCUs...
	INFO: CMSIS MCU description cache was last updated: a day ago
	INFO: In case of Custom target remove 'device_name' from your custom_targets.json5 file and add
	just the 'memories' section as 'memory_banks' section from content below.
	Otherwise add the whole following entries to  ...project_name\mbed-os\targets\cmsis_mcu_descriptions.json5:
	{
		"STM32U575VGTx": {
			"algorithms": [
				{
					"default": true,
					"file_name": "CMSIS/Flash/STM32U5xx_1M_0800.FLM",
					"ram_size": 32768,
					"ram_start": 536870912,
					"size": 1048576,
					"start": 134217728,
					"style": "Keil"
				},
				{
					"default": true,
					"file_name": "CMSIS/Flash/STM32U5xx_1M_0C00.FLM",
[... 209 lines cut ...]
   About names of macros I am not exactly sure but it seems like they should be generated by a python script, right @multiplemonomials ?

3. preparation for bootloder and its app
   As soon as linker will be ready then you can place  `target.memory_bank_config` into `mbed_app.json5` with specific parts what should override default target definition just for specific application.
   The mbed_app.json5 of a bootloader:
   ```
            "target.memory_bank_config": {
                "Flash": {
                    "size": 0x20000
                }
            },
   ```
   In this case we will set Flash size to just 128KB because we are sure or we want our bootloader will be not bigger than that. And according to  [RM0456](https://www.st.com/resource/en/reference_manual/rm0456-stm32u5-series-armbased-32bit-mcus-stmicroelectronics.pdf) (section 7.3.1 Flash memory organization on page 290) the size must be divisible by 8kb (from my understanding the FlashIAP is able to erase and program only by whole sectors/pages).
  This size is also important for later calculation. It will be use like offset for app start address, app flash size reduction and also like address where bootloader should jump to. 
   The mbed_app.json5 of an app:
   ```
            "target.memory_bank_config": {
                "Flash": {
                    "size": 0x80000
                    "start": 0x20000
                }
            },
   ```

This are basics how to achieve or understand a simple bootloader and also handy for start with Mcuboot I think. Only a method how to merge both binaries to one is missing.

lefebvresam, 2024-12-08:
Thank you very much. I will test this procedure tomorrow. To be honest I never worked with linker scripts and I was starting to read a bit about this topic but it requires a deeper knowledge about ARM architecture I guess. I even didn't know exactly what to change and why. What is missing or wrong in the current linker scripts? Fyi the controller I use is STM32U575RGT6.

To merge the hexes my idea is to test the procedure described on [mbed-mcuboot-demo](https://github.com/AGlass0fMilk/mbed-mcuboot-demo/blob/master/README.md). The memory size is exactly the same as in my case (1MB) and my idea is to write a piece of software to transfer a bin file from the SD card to a separate flash chip I already use to store my images in my HMI display application to provide the secondary slot.

lefebvresam, 2024-12-09:
I did the whole procedure and it compiles without errors, but the strange this is that I still see:

ROM Bank Flash: 80505(+0)/1048576 bytes used, 7.7% (+0.0%) used -> that's still 1MB

```
[410/410] Linking CXX executable mbed-mcuboot-bootloader.elf
-- built: /home/sam/git/mcuboot/build/mbed-mcuboot-bootloader.bin
-- built: /home/sam/git/mcuboot/build/mbed-mcuboot-bootloader.hex
/home/sam/git/mcuboot/mbed-os/tools/python/memap/memap.py:63: DeprecationWarning: the 'HEADER' constant is deprecated, use the 'HRuleStyle' and 'VRuleStyle' enums instead
  from prettytable import PrettyTable, HEADER
| Module                                 |         .text |     .data |          .bss |
|----------------------------------------|---------------|-----------|---------------|
| CMakeFiles/mbed-mcuboot-bootloader.dir |     516(+516) |     0(+0) |     408(+408) |
| [fill]                                 |       99(+99) |     4(+4) |       20(+20) |
| [lib]/bootutil.a                       |   7001(+7001) |     8(+8) | 10612(+10612) |
| [lib]/c_nano.a                         |   4648(+4648) |   92(+92) |     329(+329) |
| [lib]/gcc.a                            |   2748(+2748) |     0(+0) |         0(+0) |
| [lib]/mbed-mbedtls.a                   |   7978(+7978) |     0(+0) |         0(+0) |
| [lib]/mbed-mcuboot.a                   |   1644(+1644) |   16(+16) |     132(+132) |
| [lib]/mbed-storage-blockdevice.a       |   5172(+5172) |     0(+0) |         0(+0) |
| [lib]/mbed-storage-flashiap.a          |     997(+997) |     0(+0) |         0(+0) |
| [lib]/mbed-storage-spif.a              |   5373(+5373) |     0(+0) |       40(+40) |
| [lib]/misc                             |     296(+296) |   12(+12) |       25(+25) |
| [lib]/stdc++_nano.a                    |         1(+1) |     0(+0) |         0(+0) |
| mbed-os/CMakeFiles                     | 43528(+43528) | 364(+364) |   2026(+2026) |
| Subtotals                              | 80001(+80001) | 496(+496) | 13592(+13592) |
Total Static RAM memory (data + bss): 14088(+14088) bytes
Total Flash memory (text + data): 80497(+80497) bytes

RAM Bank SRAM1_2: 14112(+0)/262144 bytes used, 5.4% (+0.0%) used
RAM Bank SRAM3: 0(+0)/524288 bytes used, 0.0% (+0.0%) used
ROM Bank Flash: 80505(+0)/1048576 bytes used, 7.7% (+0.0%) used
```

lefebvresam, 2024-12-09:
This looks better, the "target.memory_bank_config" was not member of the HMC20 custom device.

```
[410/410] Linking CXX executable mbed-mcuboot-bootloader.elf
-- built: /home/sam/git/mcuboot/build/mbed-mcuboot-bootloader.bin
-- built: /home/sam/git/mcuboot/build/mbed-mcuboot-bootloader.hex
/home/sam/git/mcuboot/mbed-os/tools/python/memap/memap.py:63: DeprecationWarning: the 'HEADER' constant is deprecated, use the 'HRuleStyle' and 'VRuleStyle' enums instead
  from prettytable import PrettyTable, HEADER
| Module                                 |         .text |     .data |          .bss |
|----------------------------------------|---------------|-----------|---------------|
| CMakeFiles/mbed-mcuboot-bootloader.dir |     516(+516) |     0(+0) |     408(+408) |
| [fill]                                 |       99(+99) |     4(+4) |       20(+20) |
| [lib]/bootutil.a                       |   7001(+7001) |     8(+8) | 10612(+10612) |
| [lib]/c_nano.a                         |   4648(+4648) |   92(+92) |     329(+329) |
| [lib]/gcc.a                            |   2748(+2748) |     0(+0) |         0(+0) |
| [lib]/mbed-mbedtls.a                   |   7978(+7978) |     0(+0) |         0(+0) |
| [lib]/mbed-mcuboot.a                   |   1644(+1644) |   16(+16) |     132(+132) |
| [lib]/mbed-storage-blockdevice.a       |   5172(+5172) |     0(+0) |         0(+0) |
| [lib]/mbed-storage-flashiap.a          |     997(+997) |     0(+0) |         0(+0) |
| [lib]/mbed-storage-spif.a              |   5373(+5373) |     0(+0) |       40(+40) |
| [lib]/misc                             |     296(+296) |   12(+12) |       25(+25) |
| [lib]/stdc++_nano.a                    |         1(+1) |     0(+0) |         0(+0) |
| mbed-os/CMakeFiles                     | 43528(+43528) | 364(+364) |   2026(+2026) |
| Subtotals                              | 80001(+80001) | 496(+496) | 13592(+13592) |
Total Static RAM memory (data + bss): 14088(+14088) bytes
Total Flash memory (text + data): 80497(+80497) bytes

RAM Bank SRAM1_2: 14112(+0)/262144 bytes used, 5.4% (+0.0%) used
RAM Bank SRAM3: 0(+0)/524288 bytes used, 0.0% (+0.0%) used
ROM Bank Flash: 80505(+0)/131072 bytes used, 61.4% (+0.0%) used
```

JohnK1987, 2024-12-09:
> This looks better, the "target.memory_bank_config" was not member of the HMC20 custom device.

Just for clarification. The "target.memory_bank_config" should be just in `mbed_app.json5` file under the override section. In `custom_targets.json5` should be "memory_banks" like I wrote above. Because let say you want to set custom target settings for your HMC20 just once and then only copy&paste it to every new project. For application specific offsets like for bootloader or its apps you should use `mbed_app.json5`  file.

lefebvresam, 2024-12-10:
Is this still necessary?

`"target.restrict_size": "0x20000",`

JohnK1987, 2024-12-10:
> Is this still necessary in mbed_app.json5?
> `"target.restrict_size": "0x20000",`

Nope, this one was working in ARM Mbed, in MbedCE this is not implemented.
It was just for define that the bootloader can not be bigger than 128KB (0x20000), which we are able to do with
```
         "target.memory_bank_config": {
             "IROM1": {
                 "size": 0x20000
             }
         }
```

lefebvresam, 2024-12-10:
I see this error is invoked because of the latest commit in the branch mbed-ce-update-cmakelists, "Start on automation for creating images". Is this something I already have to deal with or do I need to stay on the previous commit?

JohnK1987, 2024-12-10:
Sorry, I don't understand your description.

lefebvresam, 2024-12-10:
The requirement to define variabele MCUBOOT_SIGNING_KEY is something thas has been added in this recent [commit ](https://github.com/multiplemonomials/mcuboot/commit/58ad44dee078570e02c11128869ab30797f85719) of the mbed-ce-update-cmakelists branch of `https://github.com/multiplemonomials/mcuboot.git`. Is this something I can already use and how, or just neglect and dtill use the previous commit?

JohnK1987, 2024-12-10:
Ok, so it is related to McuBoot and Jamie's current work. I have no idea about this, sorry.

lefebvresam, 2024-12-10:
After doing all the right configuration for the primary application it looks like the start address is not picked up in the right way:

```
// This file contains application specific settings.
{
    "macros": [
        "MBED_HEAP_STATS_ENABLED=true",
        "MBED_THREAD_STATS_ENABLED=true",
        "OS_TIMERS=0"
    ],
    "config": {
        "serial-bootloader-enable": {
            "help": "Build bootloader with serial update support",
            "value": 0
        },
        "secondary-slot-in-flash": {
            "help": "If enabled, store the secondary slot in the application flash immediately after the primary slot.",
            "value": false
        }
    },    
    "target_overrides": {
        "*": {
            "mcuboot.bootloader-build": false,
            "target.c_lib": "small",
            "mcuboot.log-level": "MCUBOOT_LOG_LEVEL_DEBUG",
            "platform.stdio-baud-rate": 67800,
            "platform.stdio-buffered-serial": true,
            "platform.memory-tracing-enabled": true,
            "platform.callback-nontrivial": true,
            "target.printf_lib": "std",
            "mbed-trace.enable": true,
            "mbed-trace.max-level": "TRACE_LEVEL_DEBUG",
            "mbed-trace.fea-ipv6": false            
        },
        "HMC20": {
            "events.shared-dispatch-from-application": true,            
            "rtos.main-thread-stack-size": 11240,
            "drivers.uart-serial-rxbuf-size": 512,
            "drivers.uart-serial-txbuf-size": 512,
            "sd.SPI_CS": "SD_CS",
            "sd.SPI_MOSI": "SD_MOSI",
            "sd.SPI_MISO": "SD_MISO",
            "sd.SPI_CLK": "SD_CLK",
            "sd.TRX_FREQUENCY": 40000000,
            "spif-driver.SPI_CS": "SPIF_CS",
            "spif-driver.SPI_MOSI": "SPIF_MOSI",
            "spif-driver.SPI_MISO": "SPIF_MISO",
            "spif-driver.SPI_CLK": "SPIF_CLK",            
            "spif-driver.SPI_FREQ": 40000000,
            "mcuboot.primary-slot-address": "0x8020000",
            "mcuboot.slot-size": "0xC0000",
            "mcuboot.scratch-address": "0x80E0000",
            "mcuboot.scratch-size": "0x20000",
            "mcuboot.max-img-sectors": "0x180",
            "mcuboot.read-granularity": 1,
            // This replaces "target.restrict_size":
            "target.memory_bank_config": {
                "Flash": {
                    "size": 0xE0000,
                    "start": 0x8020000
                }
            }
        }
    }
}
```

And I get a lot of those messages:

```
...
Warning: Symbol .text._ZN6RA887514_readColorTrioEh (at address 0x801f40c, size 84) is not inside a defined memory bank for this target.
Warning: Symbol .text._ZN6RA887530SetBackgroundTransparencyColorEt (at address 0x801f460, size 28) is not inside a defined memory bank for this target.
Warning: Symbol .text._ZN6RA887530GetBackgroundTransparencyColorEv (at address 0x801f47c, size 6) is not inside a defined memory bank for this target.
Warning: Symbol .text._ZN6RA88759fontwidthEv (at address 0x801f482, size 30) is not inside a defined memory bank for this target.
Warning: Symbol .text._ZN6RA88757columnsEv (at address 0x801f4a0, size 16) is not inside a defined memory bank for this target.
Warning: Symbol .text._ZN6RA887512GetTextWidthEPKcb (at address 0x801f4b0, size 108) is not inside a defined memory bank for this target.
...
```

Is there something I miss in the configuration?

lefebvresam, 2024-12-10:
I get same warnings when I test with:

```
            "Flash": {
                "access": {
                    "execute": true,
                    "non_secure": false,
                    "non_secure_callable": false,
                    "peripheral": false,
                    "read": true,
                    "secure": false,
                    "write": false
                },
                "default": true,
                "p_name": null,
                "size": 1048576,
//                "start": 134217728,
                "start": 0x8020000,
                "startup": true
            },
```

multiplemonomials, 2024-12-10:
What does CMake output when you do a fresh configuration of your project?

lefebvresam, 2024-12-10:
I always do "sh build.sh -d"; my build script is deleting the folder first before doing a full compile.

JohnK1987, 2024-12-10:
Look to build folder for project_name.elf.map and scroll down untill you see an address instead just zeros. The first number will be app address.

lefebvresam, 2024-12-10:
Looks like those matches are all zero:
/home/sam/git/hmipanel/build/develop/source/hmc20.elf.map

lefebvresam, 2024-12-10:
This is it:

```
LOAD /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/libnosys.a
END GROUP
LOAD /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/thumb/v8-m.main+fp/softfp/crtend.o
LOAD /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/thumb/v8-m.main+fp/softfp/crtn.o

.text           0x08000000    0x5e680
 *(.isr_vector)
 .isr_vector    0x08000000      0x238 mbed-os/CMakeFiles/mbed-os.dir/targets/TARGET_STM/TARGET_STM32U5/TARGET_STM32U575xG/TOOLCHAIN_GCC_ARM/startup_stm32u575xx.S.obj
                0x08000000                g_pfnVectors
 *(.text*)
 .text.deregister_tm_clones
                0x08000238       0x1c /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/thumb/v8-m.main+fp/softfp/crtbegin.o
 .text.register_tm_clones
                0x08000254       0x28 /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/thumb/v8-m.main+fp/softfp/crtbegin.o
 .text.__do_global_dtors_aux
                0x0800027c       0x28 /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/thumb/v8-m.main+fp/softfp/crtbegin.o
 .text.frame_dummy
                0x080002a4       0x24 /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/thumb/v8-m.main+fp/softfp/crtbegin.o
 .text          0x080002c8       0x7c /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/crt0.o
                0x080002c8                _stack_init
```

Not starting right. I gave to go now, I will check further later in the evening.

JohnK1987, 2024-12-10:
Can you push latest change to your repo?

lefebvresam, 2024-12-10:
But will do it not before 23h Brussels time, have to go now. Thanks in advance.

lefebvresam, 2024-12-11:
I just populated a new repo based on the previous one but with adaptations to behave as an app:

`git clone git@bitbucket.org:saleconix/mcuboot_app.git`

When you perform these commands:

```
git submodule init
git submodule update
sh build.sh -d
```

It compiles but with the same issue:

```
...
Warning: Symbol <initializer for .data.m_trace> (at address 0x800ee58, size 64) is not inside a defined memory bank for this target.
Warning: Symbol <initializer for .data._ZZ5_sbrkE4heap> (at address 0x800ee98, size 4) is not inside a defined memory bank for this target.
Warning: Symbol <initializer for .data._ZL11filehandles> (at address 0x800ee9c, size 256) is not inside a defined memory bank for this target.
Warning: Symbol <initializer for .data._ZL12flash_map_bd> (at address 0x800ef9c, size 12) is not inside a defined memory bank for this target.
...
```

JohnK1987, 2024-12-11:
Ok, what do you see on bottom of `...build/mbed_config.cmake` file under comment  `# Memory bank macros`?

lefebvresam, 2024-12-11:
```
    # Memory bank macros

    MBED_CONFIGURED_RAM1_SIZE=0x80000
    MBED_CONFIGURED_RAM1_START=0x20040000
    MBED_CONFIGURED_RAM_BANK_SRAM1_2_SIZE=0x40000
    MBED_CONFIGURED_RAM_BANK_SRAM1_2_START=0x20000000
    MBED_CONFIGURED_RAM_BANK_SRAM3_SIZE=0x80000
    MBED_CONFIGURED_RAM_BANK_SRAM3_START=0x20040000
    MBED_CONFIGURED_RAM_SIZE=0x40000
    MBED_CONFIGURED_RAM_START=0x20000000
    MBED_CONFIGURED_ROM_BANK_Flash_SIZE=0xe0000
    MBED_CONFIGURED_ROM_BANK_Flash_START=0x8020000
    MBED_CONFIGURED_ROM_SIZE=0xe0000
    MBED_CONFIGURED_ROM_START=0x8020000
    MBED_RAM1_SIZE=0x80000
    MBED_RAM1_START=0x20040000
    MBED_RAM_BANK_SRAM1_2_SIZE=0x40000
    MBED_RAM_BANK_SRAM1_2_START=0x20000000
    MBED_RAM_BANK_SRAM3_SIZE=0x80000
    MBED_RAM_BANK_SRAM3_START=0x20040000
    MBED_RAM_SIZE=0x40000
    MBED_RAM_START=0x20000000
    MBED_ROM_BANK_Flash_SIZE=0x100000
    MBED_ROM_BANK_Flash_START=0x8000000
    MBED_ROM_SIZE=0x100000
    MBED_ROM_START=0x8000000
```

JohnK1987, 2024-12-11:
Ok, I got it. I it is my fault
1. The name of flash memory is the issue. In linker we have `_FLASH_` https://github.com/mbed-ce/mbed-os/blob/00d6b9f222f05b2684f733f10e8fdeb4b1bd9546/targets/TARGET_STM/TARGET_STM32U5/TARGET_STM32U575xG/TOOLCHAIN_GCC_ARM/stm32u575xg.ld#L32-L36
  but in `custom_targets.json5` we have `Flash` https://bitbucket.org/saleconix/mcuboot_app/src/a312d7faa9243a6e5e7e621dbc3483d0df54f66a/custom_targets.json5#lines-38
  It must be same in both, so rename the one in `custom_targets.json5` to `FLASH`. Then it will be OK, I think.

2.  delete this part https://github.com/mbed-ce/mbed-os/blob/00d6b9f222f05b2684f733f10e8fdeb4b1bd9546/targets/TARGET_STM/TARGET_STM32U5/TARGET_STM32U575xG/cmsis_nvic.h#L20-L38

lefebvresam, 2024-12-11:
There was another typo, linker script must define MBED_RAM_BANK_SRAM3_SIZE and not MBED_RAM_BANK_SRAM_3_SIZE

multiplemonomials, 2024-12-11:
That's actually a really good question.  My thoughts on this are that on single core devices, there really isn't any reason for the running application to not use all the available RAM, so there's no reason to make it configurable.  On multicore devices, that changes.

lefebvresam, 2024-12-11:
I'm trying the whole demo at [mbed-mcuboot-demo](https://github.com/AGlass0fMilk/mbed-mcuboot-demo/blob/master/README.md) but blocking here:
```
sam@saleconix:~/git/mcuboot$ hexmerge.py -o merged.hex --no-start-addr build/mbed-mcuboot-bootloader.hex signed.hex signed-update.hex
Merging: signed-update.hex
Starting addresses are different
```
and merge won't be created

I have:

```
sam@saleconix:~/git/mcuboot$ hexinfo.py signed.hex
- file: 'signed.hex'
  data:
  - { first: 0x0801F000, last: 0x0802F1CB, length: 0x000101CC }
  - { first: 0x080DEFF0, last: 0x080DEFFF, length: 0x00000010 }

sam@saleconix:~/git/mcuboot$ hexinfo.py signed-update.hex
- file: 'signed-update.hex'
  entry: 0x40000000
  data:
  - { first: 0x0805F000, last: 0x0806F1CB, length: 0x000101CC }
```

lefebvresam, 2024-12-11:
One thing was already wrong. The memory config is without header and trailer:

```
            "target.memory_bank_config": {
                "FLASH": {
                    "size": 0xBE000,
                    "start": 0x8021000
                }
            }
```

instead of

```
            "target.memory_bank_config": {
                "FLASH": {
                    "size": 0xE0000,
                    "start": 0x8020000
                }
            }
```

lefebvresam, 2024-12-11:
By doing this:

`arm-none-eabi-objcopy --change-addresses 0x40000 signed-update.hex signed-update.hex`

There comes `entry: 0x40000000` in signed-update.hex
And it clashes with `entry: 0x08000345` in mbed-mcuboot-bootloader.hex unless you do --overlap=ignore.
No idea how to do it correctly.

lefebvresam, 2024-12-11:
It's a trick:

```
sam@saleconix:~/git/mcuboot$ arm-none-eabi-objcopy --change-addresses 0x40000 --set-start 0x7FC0345 signed-update.hex signed-update.hex
sam@saleconix:~/git/mcuboot$ hexinfo.py signed-update.hex
- file: 'signed-update.hex'
  entry: 0x08000345
  data:
  - { first: 0x08060000, last: 0x080701CB, length: 0x000101CC }

sam@saleconix:~/git/mcuboot$ hexinfo.py build/mbed-mcuboot-bootloader.hex
- file: 'build/mbed-mcuboot-bootloader.hex'
  entry: 0x08000345
  data:
  - { first: 0x08000000, last: 0x08011A53, length: 0x00011A54 }

sam@saleconix:~/git/mcuboot$ hexmerge.py -o merged.hex --no-start-addr build/mbed-mcuboot-bootloader.hex signed.hex signed-update.hex
sam@saleconix:~/git/mcuboot$ hexinfo.py merged.hex
- file: 'merged.hex'
  data:
  - { first: 0x08000000, last: 0x08011A53, length: 0x00011A54 }
  - { first: 0x08020000, last: 0x080301CB, length: 0x000101CC }
  - { first: 0x08060000, last: 0x080701CB, length: 0x000101CC }
  - { first: 0x080DFFF0, last: 0x080DFFFF, length: 0x00000010 }

sam@saleconix:~/git/mcuboot$ 
```

lefebvresam, 2024-12-12:
After programming, there is nothing in the serial console. I think I have to add stuff to the 

```
target_link_libraries(${APP_TARGET}
    mbed-baremetal
    mbed-storage
    mbed-mcuboot
)
```

in replacement of 

`"requires": ["bare-metal", "mbedtls", "mcuboot", "flashiap-block-device", "spif-driver", "qspif", "mbed-trace", "sd"],`

Something I miss here?

lefebvresam, 2024-12-12:
I discovered two issues:

1/ Flashing is not with .hex but with .bin, I had to do the conversion first
2/ I have to add `add_compile_options(-DMCUBOOT_HAVE_LOGGING=1)` to CMakeLists.txt in order to see the mbed trace info  

It is recommended to test and program the bootloader only with 'wide flash' settings before doing the merge.

lefebvresam, 2024-12-12:
Currently:

`assertion "rc == 0" failed: file "/home/sam/git/mcuboot/mcuboot/boot/bootutil/src/swap_misc.c", line 105, function: swap_status_init`

lefebvresam, 2024-12-14:
When I add in bootutil_public.c at the point something is written in the image trailer:

```
    align = flash_area_align(fap);
    BOOT_LOG_DBG("inlen %d", inlen);
    BOOT_LOG_DBG("align %d", align);
    align = ALIGN_UP(inlen, align);
    BOOT_LOG_DBG("align %d", align);
    BOOT_LOG_DBG("boot_max_align %d", BOOT_MAX_ALIGN);
    if (align > BOOT_MAX_ALIGN) {
        return -1;
    }
```

I get:

```
[DBG ][MCUb]: writing swap_info; fa_id=2 off=0x1ffd8 (0x80fffd8), swap_type=0x2 image_num=0x0
[DBG ][MCUb]: inlen 1
[DBG ][MCUb]: align 16
[DBG ][MCUb]: align 16
[DBG ][MCUb]: boot_max_align 8
[DBG ][MCUb]: boot_write_swap_info returns -1
assertion "rc == 0" failed: file "/home/sam/git/mcuboot/mcuboot/boot/bootutil/src/swap_misc.c", line 106, function: swap_status_init

```

The solution is to add a config at both in mcuboot as in the app:

```
    "macros": [
        "MCUBOOT_BOOT_MAX_ALIGN=16"
    ],
```

I see in the case of other OS'es there are a lot of new configs which are currently no longer supported in the mbed variant. They won't update it it because of the obsoleteliness of mbed.

multiplemonomials, 2024-12-14:
Just wanted to let you know, I have been working this week on integrating more of mcuboot's build workflow into CMake.  I have this demo project set up [here](https://github.com/mbed-ce/mbed-mcuboot-demo-app) that generates both types of images using a CMake command: https://github.com/mbed-ce/mbed-mcuboot-demo-app/blob/4111457d449937f9f64a7c4af9361a695a89dc4f/CMakeLists.txt#L34

You currently have to use my fork of mcuboot and my branch of Mbed (dev/tweak-upload-method-deps) to make it work, but it should be very convenient!

multiplemonomials, 2024-12-14:
Though, I haven't actually gotten it to work yet, I am getting
```
[INFO][UpdaterApp]: Secondary BlockDevice inited
[INFO][UpdaterApp]: > Press button to erase secondary slot
[INFO][UpdaterApp]: Erasing secondary BlockDevice...
[INFO][UpdaterApp]: Secondary BlockDevice erased
[INFO][UpdaterApp]: > Press button to copy update image to secondary BlockDevice
[INFO][UpdaterApp]: > Image copied to secondary BlockDevice, press button to activate
[DBG ][MCUb]: flash area 1 open count: 1 (+)
[DBG ][MCUb]: initializing flash area 1...
[ERR ][MCUb]: Invalid read: fa_id 1 offset 0xec57fff0 len 0x200
[DBG ][MCUb]: flash area 1 open count: 0 (-)
[DBG ][MCUb]: deinitializing flash area block device 1...
[ERR ][UpdaterApp]: Failed to set secondary image pending: 1
```
but I haven't really actually started debugging it yet.

multiplemonomials, 2024-12-14:
Also, @zhiyong-ft and I are going to ask the mcuboot devs to officially support mbed-ce!  Hopefully we can convince them not to drop support

lefebvresam, 2024-12-14:
[image]

I did a lot of fixes. But have to go now. I come back on it tomorrow.

zhiyong-ft, 2024-12-14:
@lefebvresam did you get mcuboot with mbed-os to work? I haven't tested it with mbed-ce yet, but I was able to get mcuboot v1.7 & v1.8 to work with mbed-os v6.17 & v5.15 to work, to the extent that bootloader can upgrade firmware saved into secondary slot. Newer versions of mcuboot v1.9, v1.10 and v2.0, v2.1 didn't work when I tested it on STM32F767 and F756NG. Bootloader will just ignore images saved into 2ndary slot.

Also I was able to compile with encryption enabled after some tweaking, but bootloader ignores image saved in 2ndary slot as well.

I will try to talk to mcuboot team to keep support for mbed-os and mbed-ce later.

multiplemonomials, 2024-12-16:
All right, I got it working! First off, I was running into two different bugs related to using a MicroSD card as the secondary block device, which I pushed fixes for.  But, then I was getting the dreaded "doesn't boot secondary application" error.

Then, I noticed that Sam pushed a commit to the bootloader repo to correctly turn on logging, and that turned out to be key!  Because then I noticed I was getting the error
```
Cannot upgrade: slots have non-compatible sectors
```

Digging into it some more, this was caused by an incorrect setting for the `mcuboot.max-img-sectors` option!  If this is set too small, the mcuboot sector checking algorithm starts indexing into undefined memory, causing it to read garbage for the sector sizes and produce the above warning.  I had set max-img-sectors based on `mcuboot.slot-size` divided by the flash sector size (0x1000). However, you actually need to set this based on the secondary BD sector size if it has smaller sectors!  So, since I was using a MicroSD card I had to set it to `mcuboot.slot-size`/512!

Once I made that change (and also tweaked the mcuboot source to add a better error message for this issue), my upgrades worked fine!

@zhiyong-ft could you try building with the mbed-ce mcuboot fork and see if it's able to upgrade OK on your target?  If not, what does it print when it boots?

zhiyong-ft, 2024-12-16:
@multiplemonomials do you mean you got encrypted images to work? or the latest version of mcuboot(without encrypted image)?

The `imgtool.py` doesn't seem to support extract public key into pem format. We will have to use openssl to do it like this:
```
openssl rsa -in signing-keys.pem -pubout >> signing-key-pub.pem
```
The exported signing-key-pub.pem will be used for image encryption, at least that is my understanding.

When I did this, bootloader will stuck when trying to upgrade from image saved in secondary slot.

multiplemonomials, 2024-12-16:
No encryption but latest mcuboot version!

zhiyong-ft, 2024-12-16:
I cloned the bootloader project as of 26f8a76c02ec2e7746ec1989169501ac673210bd, added `cmake-variants.yaml`, and copied DISCO_F746NG related settings into mbed_app.json5. There seems to be something related to signing key path, similar to previous observation.
```
[cmake] -- Checking for Python package imgtool.main -- found
[cmake] CMake Error at mcuboot/boot/mbed/mcuboot_imgtool.cmake:33 (message):
[cmake]   Must specify path to valid image signing key via MCUBOOT_SIGNING_KEY CMake
[cmake]   option in order to build this project.
[cmake] Call Stack (most recent call first):
[cmake]   mcuboot/boot/mbed/CMakeLists.txt:5 (include)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[proc] The command: "C:\Program Files\CMake\bin\cmake.EXE" -DCMAKE_BUILD_TYPE:STRING=Develop -DMBED_TARGET:STRING=DISCO_F746NG -DUPLOAD_METHOD:STRING=STM32CUBE -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE "-DCMAKE_C_COMPILER:FILEPATH=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-gcc.exe" "-DCMAKE_CXX_COMPILER:FILEPATH=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-g++.exe" --no-warn-unused-cli -SD:/Temp/mbed-mcuboot-bootloader -Bd:/Temp/mbed-mcuboot-bootloader/build -G Ninja exited with code: 1
```

multiplemonomials, 2024-12-16:
Oh well, just pass that argument to CMake: `-DMCUBOOT_SIGNING_KEY=/path/to/signing-key.pem`.  I updated the build system to run all of the needed commands automatically.

zhiyong-ft, 2024-12-16:
Like this?
```
'C:\Program Files\CMake\bin\cmake.exe'  -DMCUBOOT_SIGNING_KEY=sygining-keys.pem  
```
Got error message:
```
At line:1 char:41
+ 'C:\Program Files\CMake\bin\cmake.exe'  -DMCUBOOT_SIGNING_KEY=syginin ...
+                                         ~~~~~~~~~
Unexpected token '-DMCUBOOT' in expression or statement.
At line:1 char:50
+ ... m Files\CMake\bin\cmake.exe'  -DMCUBOOT_SIGNING_KEY=sygining-keys.pem
+                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Unexpected token '_SIGNING_KEY=sygining-keys.pem' in expression or statement.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : UnexpectedToken
```

[75 more replies]