mbed-ce/mbed-os #405: undefined reference to `swprintf'

lefebvresam, 2024-12-18:
Using C++20 option std::format:

```
uint8_t Display::getVersion(string& version) {
    struct image_version iver;
    if (boot_get_current_version(&iver)) {
        ERROR("Cannot read version number");
        return 1;
    }
    version = std::format("%u.%u.%u+%lu", iver.iv_major, iver.iv_minor, +iver.iv_revision, +iver.iv_build_num);
    return 0;
}
```

gives:

```
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/libc_nano.a(libc_a-wcsftime.o): in function `__strftime.isra.0':
wcsftime.c:(.text.__strftime.isra.0+0x2fa): undefined reference to `swprintf'
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: wcsftime.c:(.text.__strftime.isra.0+0x32c): undefined reference to `swprintf'
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: wcsftime.c:(.text.__strftime.isra.0+0x390): undefined reference to `swprintf'
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: wcsftime.c:(.text.__strftime.isra.0+0x608): undefined reference to `swprintf'
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: wcsftime.c:(.text.__strftime.isra.0+0x700): undefined reference to `swprintf'
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: /usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/lib/thumb/v8-m.main+fp/softfp/libc_nano.a(libc_a-wcsftime.o):wcsftime.c:(.text.__strftime.isra.0+0x834): more undefined references to `swprintf' follow
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

JohnK1987, 2024-12-18:
1. you are using c++20 what is not officially supported in Mbed as i remember.
2. I remember you had set `"target.c_lib": "small"`. Have you also tried `"target.c_lib": "std"`?

lefebvresam, 2024-12-18:
Then I have:

```
l/build/develop/memory_banks.json
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: source/hmc20.elf section `.text' will not fit in region `FLASH'
/usr/local/gcc-arm/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 71436 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Building failed
```

Alternative is old school but ugly:

```
uint8_t Display::appSelected(string& appname) {
    appname = "none";
    if (!_loadingenabled) {
        appname = "STATIC";
        return 0;
    }
    DMAFlash::datinfo_s datinfo;
    uint8_t err = _dma.getDataInfo(APPSELECT_KEY, datinfo);
    if (err == 2) return 2; // not found
    if (err) return 1; // other error
    appname.resize(datinfo.datsize);
    if (_dma.readData(APPSELECT_KEY, (void*)appname.c_str())) return 1; // TODO: is that legal?
    return 0;
}
```

JohnK1987, 2024-12-18:
The new error is probably related to the change from `small` to `std` because the std occupied more flash memory.

lefebvresam, 2024-12-18:
I try to avoid advanced libs like streaming operator << etc, it will dramatically increase memory use.

multiplemonomials, 2025-01-18:
Going to close this as it does not seem to be an Mbed specific issue.

closed 2025-01-18 by multiplemonomials