mbed-ce/mbed-os #407: MBED_CONF_RTOS_PRESENT incorrectly pass to in-tree static library or conversely

ccli8, 2024-12-27:
`MBED_CONF_RTOS_PRESENT` is defined in [mbed-rtos-flags](https://github.com/mbed-ce/mbed-os/blob/2f3a3c330210c8b270ba6d9c3c265297eefad750/cmsis/device/rtos/CMakeLists.txt#L32-L35). For libraries which need to distinguish between `mbed-os` or `mbed-bardmetal` build, e.g. [ESP8266Interface.h](https://github.com/mbed-ce/mbed-os/blob/2f3a3c330210c8b270ba6d9c3c265297eefad750/connectivity/drivers/wifi/COMPONENT_ESPRESSIF_ESP8266/ESP8266Interface.h#L35-L37), they need to determine to link `mbed-rtos-flags` to get in `MBED_CONF_RTOS_PRESENT` (for linking `mbed-os`) or not (for linking `mbed-baremetal`). But only application-level CMakeLists.txt knows which of `mbed-os` and `mbed-baremetal` is linked, and this information isn't passed to these libraries which rely on `MBED_CONF_RTOS_PRESENT`. This will generate such error: application linking `mbed-baremetal` but libraries having `MBED_CONF_RTOS_PRESENT` definedㄝor conversely.

One rough fix is these libraries all link `mbed-core-flags` and let application-level CMakeLists.txt determine to link `mbed-rtos-flags` additionally for these libraries. But this fix isn't compatible and isn't easy to apply.

multiplemonomials, 2024-12-27:
> One rough fix is these libraries all link mbed-core-flags and let application-level CMakeLists.txt determine to link mbed-rtos-flags additionally for these libraries. But this fix isn't compatible and isn't easy to apply.

Yeah, that's basically what I have done until now.  But honestly, it's an ugly way to do it, and it can cause a lot of unexpected issues.

I've been seriously considering just making it so a given project is set to "baremetal mode" or "OS mode" and then that globally controls whether mbed-core-flags and mbed-rtos-flags are linked together or not.  That would remove the ability to have some apps build with baremetal and others build with OS within a single project.  But honestly, I am not sure anyone is actually using that feature.

So basically, I'd accept a PR to fix this mess by only having one of mbed-baremetal or mbed-os active at a time, set by a JSON option.

ccli8, 2024-12-30:
>  That would remove the ability to have some apps build with baremetal and others build with OS within a single project. But honestly, I am not sure anyone is actually using that feature.

This feature needs two build modes at the same time for single static library, like `mbed-os` and `mbed-baremetal`. But currently these static libraries only have single build mode: either `mbed-core-flags` or `mbed-core-flags`/`mbed-rtos-flags` exclusively. So I think this feature can be not taken into account, except these static libraries also support two build modes simultanelusly.

multiplemonomials, 2025-03-28:
I think this is fixed in #410 right?

ccli8, 2025-03-31:
Fixed via #410

closed 2025-03-31 by ccli8