mbed-ce/mbed-os #427: STM SPI DMA utils, malloc in ISR

chrissnow, 2025-01-22:
https://github.com/mbed-ce/mbed-os/blob/e7122303cf52de69bc29d4ec97afe8dc6b2af4f1/targets/TARGET_STM/stm_dma_utils.c#L634
When using SPI with DMA and "transfer" in an ISR  the first time it's called malloc will be called and fail as it's not allowed.

You can workaround it by ensuring you have used transfer before even if it's a dummy transaction.
Perhaps we should link during enabling DMA or SPI object creation?

multiplemonomials, 2025-02-11:
Hmm yeah that is an issue, but I'm not sure if there's an easy way around it other than doing what you said.

chrissnow, 2025-02-11:
DMA in an ISR is incredibly useful when dealing with fast continual conversion ADC's etc.
Should we just link it when enabling DMA? it's unlikely you would want to do that in an ISR, at least we would not lose functionality.

multiplemonomials, 2025-02-11:
Maybe the way to go is to add a configuration option like `target.spi-always-allocate-dma` (or something). If set, it would allocate DMA channels when a SPI is initially constructed, so that it is guaranteed to work from an ISR, at the cost of more RAM usage when constructing each SPI.