mbed-ce/mbed-os #437: USB HID Tests fails

JojoS62, 2025-02-14:
The usb hid test always fails.

Reason: python module ‘hidapi’ is missing.
Manual installation fixes it, but needs to be added to requirements.txt.

JohnK1987, 2025-02-15:
That is strange because it was working before end of year 2024.
However I faced similar issue with another python module (pyjson5) and solution was simple. I deleted venv folder and re-run cmake solved that. It seems like some steps of venv creation can silently fail.

JojoS62, 2025-02-15:
Deleting venv helps when python is upgraded on the pc. But for usb-hid it’s the missing module. 
Maybe the test could be rewritten to use pyusb, but that is very different, installing the missing module is much simpler. Unless it is kicked out, the python version hell is no fun.
I think the version that Mbed uses should be frozen and a particular python version should be required, e.g. 3.12. 

I’ve started also to find the problem in usb-serial, which crashes also in python test code. But I’m stuck in starting the module under the debugger in VSC.

JohnK1987, 2025-02-16:
And USB tests or let say tests in general are using a different venv folder then application?

[image]

JojoS62, 2025-02-16:
had just a quick check: there are several requirements*.txt. The main for the building system was improved by Jamie, it is observed by cmake and installs in the venv. I remember that was recently improved, also after json5 was not found. Now if this requirements.txt changes, missing modules would be installed.
But the hidapi is in requirements-ci-tests.txt, and this is not yet installing in the venv. There may be also different behaviour of pip now, it refuses to install globally. Its also possible that it is controlled by an environment variable and can behave different depending on the python installation. But the correct way will be to fix other requirements*.txt to install also in the venv.
Or simplify it and use only one requirements for all, now with forced using venv it should be no problem, before that could be a problem with global installation of different versions.

JohnK1987, 2025-09-07:
It seems I solve it without any special note, check the PR!

closed 2026-02-22 by JohnK1987